mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #107: Extending the library to support multiplexing

kustom666, 2021-03-29:
Hello,

So this is a follow up to #105 where I discovered that some panels coming from specific manufacturers use FM6124 LED drivers, 74hc245c chips to drive the drivers, but have some multiplexing mechanism that makes them appear to be at a different scan rate than they actually are. 

Without modifications to the library, the display will appear to display only half the pixels, so far I've only seen displays that appear to only display every other 8 rows. They're marketed as *outdoors* panels. Their PCB is coated with conformal coating (which makes identifying the chips a real pain, but makes them IP resistant) and they have too few row select pins (ABC only for a 64x32, I suspect smaller panels would only have AB).

I've managed to make them work using rpi-rgb-led-matrix, with the --led-multiplexing=1 option, which activates their [StripeMultiplexMapper](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L95). 
There's a couple things those seem to do internally: create a virtual area that has twice as many columns and half as many rows as the physical panel [see here](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L32) (the stretch factor is 2 in the case of [striped multiplexing](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L97)

The second thing this seems to do is to remap x and y coords that are passed using [that logic](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L111).

Perhaps we can create a virtual display that also remaps the coordinates, like in the 1/4 scan case?

mrfaptastic, 2021-03-29:
Hi Kustom666,

Thanks for raising this issue and also doing some investigation as to the likely root cause. **That helps greatly**. Can you post some pictures as well? 

Would be interesting to see the output of a sketch that draws to each pixel slowely starting top left to bottom right.

Looks like the solution could be pretty simple if what you said is on the mark?

In the core DMA class:
- set MATRIX_WIDTH = 2x what the actual pixel dimensions of the panel are
- set MATRIX_HEIGHT = 0.5 what the actual pixel dimension of the panel are

Then in some hacked-up virtual display class:

```cpp

inline VirtualCoords VirtualMatrixPanel::getCoords(int16_t x, int16_t y) {

  coords.x = coords.y = 0;

    // Untested, copy-paste and replace of https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L95
    const bool is_top_stripe = (y % (MATRIX_HEIGHT/2)) < MATRIX_HEIGHT/4;
    coords.x  = is_top_stripe ? x + MATRIX_WIDTH: x;
    coords.y  = ((y / (MATRIX_HEIGHT/2)) * (MATRIX_HEIGHT/4)  + y % (MATRIX_HEIGHT/4));

  return coords;  

}

```

You experimentation required.

mrfaptastic, 2021-03-29:
Actually... thinking about it a bit more based on the information you have provided... MATRIX_WIDTH and MATRIX_HEIGHT probably do need to be what you said (2x and 0.5 real)

But in the virtual display, it needs to be as if these were normal ABCDE addressable pannels, so: 

```cpp

// Have these somewhere
#define REAL_MATRIX_WIDTH
#define REAL_MATRIX_HEIGHT

inline VirtualCoords VirtualMatrixPanel::getCoords(int16_t x, int16_t y) {

  coords.x = coords.y = 0;

    // Untested, copy-paste and replace of https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L95
    const bool is_top_stripe = (y % (REAL_MATRIX_HEIGHT/2)) < REAL_MATRIX_HEIGHT/4;
    coords.x  = is_top_stripe ? x + REAL_MATRIX_WIDTH: x;
    coords.y  = ((y / (REAL_MATRIX_HEIGHT/2)) * (REAL_MATRIX_HEIGHT/4)  + y % (REAL_MATRIX_HEIGHT/4));

  return coords;  

}

```

Have fun. Feel free to get me one of these intriguing panels :-)

mrfaptastic, 2021-03-29:
Actually... thinking about it a bit more based on the information you have provided... MATRIX_WIDTH and MATRIX_HEIGHT probably do need to be what you said (2x and 0.5 real)

But in the virtual display, it needs to be as if these were normal ABCDE addressable pannels, so: 

```cpp

// Have these somewhere, example based on assumed 64x32 panel
#define REAL_MATRIX_WIDTH 64
#define REAL_MATRIX_HEIGHT 32

inline VirtualCoords VirtualMatrixPanel::getCoords(int16_t x, int16_t y) {

  coords.x = coords.y = 0;

    // Untested, copy-paste and replace of https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc#L95
    const bool is_top_stripe = (y % (REAL_MATRIX_HEIGHT/2)) < REAL_MATRIX_HEIGHT/4;
    coords.x  = is_top_stripe ? x + REAL_MATRIX_WIDTH: x;
    coords.y  = ((y / (REAL_MATRIX_HEIGHT/2)) * (REAL_MATRIX_HEIGHT/4)  + y % (REAL_MATRIX_HEIGHT/4));

  return coords;  

}

```

kustom666, 2021-03-30:
Alright, I'll get started soon with the MATRIX_WIDTH/HEIGHT set to 64x32 in the virtual, but 128x16 in physical with virtual coords.

I'll upload a video soon of the default behavior with text and per pixel lighting

Paulie92, 2021-04-15:
Hello,

this feature will be great. I also would need to use the "outdoor" panels, which are not avaiable in 1/16 scan variant. @kustom666 , did you do any progress in your research?

Thank for your feedback.

kustom666, 2021-04-18:
Haven't had much time to work on it yet, but I should soon

kustom666, 2021-05-10:
Ok, so a little update: I've made a custom breakout for the esp32s to wire to the panel as testing with a breadboard and free floating wires led to really annoying issues with the display. I'll get the multiplexing working when I receive the boards

DarrylStrong, 2021-05-27:
Interesting, I too would like to be able to drive 8:1 panels using this library. Taking a standard 8:1 32x32 panel as an example the top RGB bits drive the top left 16 columns of 8 rows and then the data goes to the next 8 rows down starting from the left side. This is then repeated starting at the 17th column from the left.
[image]

mrfaptastic, 2021-05-27:
This is a simple pixel co-ordinate re-mapping exercise then. There's a 1/4 scan panel that you can modify to suit how 1/8 panels work. Please feel free to contribute it back to this library to help others.

kustom666, 2021-06-02:
Ok I received my final PCBs (SMDs can sometimes be a pain in the rear to solder though) and I've tested with the plasma example, can confirm it looks like the striping follows what darryl said. I'll send a video tomorrow and I'll try making the virtual display work

mrfaptastic, 2021-12-13:
Closing this as there is now a 1/8 scan panel example.

closed 2021-12-13 by mrfaptastic