mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #111: display rotation?

matphillips, 2021-04-05:
I have three 64x32 panels arranged in a stack, making the resolution 96 high by 64 wide. I'd like to use this rotated 90 degrees to give a 9:6 ratio 96x64 resolution.
Is there a way to tell the virtual matrix to rotate?

mrfaptastic, 2021-04-05:
Try with the git. I've just added a Virtualdisplay.setRotate(true/false) function.

Completely untested. Reply with a photo of the result if it doesn't work, close this issue if it does work please.

matphillips, 2021-04-05:
Hi, thanks!
I updated to the latest files from the git and set the rotate flag and the display is now rendered in 'portrait' orientation.
However, the library still thinks the width of the display is 64 and the height is 96, so nothing is shown on the right-hand third of the screen, and text wraps to a new line at the 64 pixel point.

I'll try to add a photo...

```
  Serial.print("MATRIX_WIDTH       : " + String(PANEL_RES_X * PANEL_CHAIN));
  Serial.println("  MATRIX_HEIGHT       : " + String(PANEL_RES_Y));
  Serial.print("VIRTUAL PANEL WIDTH: " + String(VPANEL_W));
  Serial.println("   VIRTUAL PANEL HEIGHT: " + String(VPANEL_H));

MATRIX_WIDTH       : 192  MATRIX_HEIGHT       : 32
VIRTUAL PANEL WIDTH: 64   VIRTUAL PANEL HEIGHT: 96
```

matphillips, 2021-04-05:
[image]

https://imgur.com/a/YshMHs1

mrfaptastic, 2021-04-06:
Try again now.

matphillips, 2021-04-06:
That did make a change, it seems to know the display is wider now, but still won't light any pixels in the right third (except one, for some reason). I turned off rotation and ran the same code, just to check everything is ok. Photo included to show both results. (note the text being cropped at the 66% point, but not flowing to next line immediately as characters seem to be "invisible" for a few letters, and then it moves to next line)

If this seems like it's gonna be tough to implement, don't bother, I suppose most people would have just bought 4 panels and use them in a 2x2 matrix to get the same result! Somehow, I ended up with three :)

https://imgur.com/a/WsQPFa6

mrfaptastic, 2021-04-07:
Hah! All good, as they say, "third time lucky". Try again now :-)

matphillips, 2021-04-07:
Ah, that got it! Working fine now.
Thanks for taking a look at this, hopefully a rotation ability may be useful to someone else too.
I'll close this issue. Thanks again.

closed 2021-04-07 by matphillips