mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #112: Is there some internal faults with the library?

MythicalForce, 2021-04-11:
# Video
video showing the error (notice on the Right side they follow the circle line but on the left they move 1 pixel IN)
[https://streamable.com/lh7klw](https://streamable.com/lh7klw)

# Setup
her is my setup
```
#pragma once

#include <Arduino.h>

#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>
#include <FastLED.h> 

#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls see ChainedPanels example
#define SERPENT false // Should be false
#define TOPDOWN true // Should be true

// Virtual Panl dimensions - our combined panel would be a square 4x4 modules with a combined resolution of 128x128 pixels
#define VPANEL_W PANEL_RES_X*NUM_COLS // Kosso: All Pattern files have had the MATRIX_WIDTH and MATRIX_HEIGHT replaced by these.
#define VPANEL_H PANEL_RES_Y*NUM_ROWS //

// placeholder for the matrix object
MatrixPanel_I2S_DMA *matrix = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel  *virtualDisp = nullptr;

namespace MatrixConfig
{
    void setup()
    {
        HUB75_I2S_CFG mxconfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);

        matrix = new MatrixPanel_I2S_DMA(mxconfig);

        matrix->setBrightness8(96);

        if( not matrix->begin() )
        {
            Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
        }

        virtualDisp = new VirtualMatrixPanel((*matrix), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
    }
};
```

# The Circle (planet)
The circle is centered x = 64, y 64 with a radius of 32
`Planet planet(64, 64, 32, GREEN);`

this is the function that moves the rectangles around
```
    void attachObj2()
    {
        for (int i = 0; i < _Shapes.size(); i++)
        {
            _Shapes.at(i)->draw(
                (_Radius + (16/2))*cos(_rotSpeed + i*_ObjSpacing*pi/_Shapes.size()) + _xPos, 
                (_Radius + (16/2))*sin(_rotSpeed + i*_ObjSpacing*pi/_Shapes.size()) + _yPos
                               );

            _Shapes.at(i)->setAngle(_rotSpeed + i*_ObjSpacing*pi/_Shapes.size());
        }
    }
    ```
    
    and no matter what size i give the circle "planet" or the rectangles they still go in 1 pixel on the left side for some reason :S

mrfaptastic, 2021-04-11:
You'll need to test using the 'virtualDisp ->drawPixel(x,y,color)' function and test with a bunch of random x,y co-ords, and then count the leds on the display to confirm the right one is indeed being 'lit'.

Not sure if it's an issue with this library as this library doesn't do any graphical shape drawing stuff - that's all provided from Adafruit_GFX library. However, there could possibly be some bug in the VirtualDisplay pixel co-ord re-maping.

mrfaptastic, 2021-04-11:
You'll need to test if it's a bug by using the 'virtualDisp ->drawPixel(x,y,color)' function and test with a bunch of random individual x,y co-ords, and then count the leds on the display to confirm the right one is indeed being 'lit'.

i.e. drawPixel(12, 34, color565.(255,255,255)) will light pixel 13 when you count from the left (starting at number 1), and pixel 34 when counting down from the top (assuming you have the physical panls the right way around....).

If there's some bug, you'd be able to see if this is a case pretty easily doing this test.

MythicalForce, 2021-04-11:
> You'll need to test if it's a bug by using the 'virtualDisp ->drawPixel(x,y,color)' function and test with a bunch of random individual x,y co-ords, and then count the leds on the display to confirm the right one is indeed being 'lit'.
> 
> i.e. drawPixel(12, 34, color565.(255,255,255)) will light pixel 13 when you count from the left (starting at number 1), and pixel 34 when counting down from the top (assuming you have the physical panls the right way around....).
> 
> If there's some bug, you'd be able to see if this is a case pretty easily doing this test. i.e. If you're saying the left half is shifted to the right then.
> 
> drawPixel(62, 64, colour565(255,0,0)); drawPixel(63, 64, colour565(0,255,0)); drawPixel(64, 64, colour565(0,0,255)); would only show two dots right ?
> 
> I don't have these panels and setup so I can't test.

i'll do some testing when i get home and report back to you, no problem :)

MythicalForce, 2021-04-11:
> drawPixel(62, 64, colour565(255,0,0)); drawPixel(63, 64, colour565(0,255,0)); drawPixel(64, 64, colour565(0,0,255)); would only show two dots right ?

so you said `would only show two dots right ?` no it shows 3 dots

MythicalForce, 2021-04-11:
# Code
```
    virtualDisp->drawFastHLine(0, 34, 128, virtualDisp->color565(255,0,0));
    virtualDisp->drawFastVLine(12, 0, 128, virtualDisp->color565(0,255,0));

    virtualDisp->drawFastHLine(0, 94, 128, virtualDisp->color565(255,0,0));
    virtualDisp->drawFastVLine(116, 0, 128, virtualDisp->color565(0,255,0));
```
# Results
[https://cdn.discordapp.com/attachments/768907118589247608/830835776975536128/172085256_521047455923284_1992432367801028052_n.jpg](https://cdn.discordapp.com/attachments/768907118589247608/830835776975536128/172085256_521047455923284_1992432367801028052_n.jpg)

# Description
on the right side it counts 12 and the 13 line is meeting the green line (using a white dot that would make the 13 pixel white)
but.... on the left side i count 11 and on the 12 line it meets the green line (using a white dot that would make the 12 pixel white)

mrfaptastic, 2021-04-12:
Hi @MythicalForce. I just counted every pixel individually and I notice that the following is going to the right 'column' except for the fact that it seems the x-coord should start at 0, which it doesn't.

```
   // Lights-up the coorrect x-row of LEDS except for the fact it starts at 1!
    virtualDisp->drawFastVLine(12, 0, 128, virtualDisp->color565(0,255,0));
    virtualDisp->drawFastVLine(116, 0, 128, virtualDisp->color565(0,255,255));
``
![image](https://user-images.githubusercontent.com/12006953/114367175-1671cb00-9b74-11eb-85b4-48ee2b7a17a9.png)

Can you do this test for me?

``
    virtualDisp->drawFastVLine(116, 0, 128, virtualDisp->color565(0,255,0));
    virtualDisp->drawFastVLine(116, 0, 128, virtualDisp->color565(0,255,0));
``
We should only see one vertical line.

mrfaptastic, 2021-04-12:
Hi @MythicalForce. I just counted every pixel individually and I notice that if I start counting from one/1 it is the correct x-coord, except it should start at 0, which it doesn't.

```cpp
   // Lights-up the coorrect x-row of LEDS except for the fact it starts at 1!
    virtualDisp->drawFastVLine(12, 0, 128, virtualDisp->color565(0,255,0));
    virtualDisp->drawFastVLine(116, 0, 128, virtualDisp->color565(0,255,255));
```
[image]

Can you do this test for me?

```cpp
    virtualDisp->drawFastVLine(127, 0, 128, virtualDisp->color565(0,255,0));
    virtualDisp->drawFastVLine(128, 0, 128, virtualDisp->color565(0,255,255));
    virtualDisp->drawFastVLine(0, 0, 128, virtualDisp->color565(255,255,255));
```
We should only see one vertical line?

MythicalForce, 2021-04-12:
> We should only see white on left, green on right.

and that is what i see hmm must be something wrong with my code then...

mrfaptastic, 2021-04-12:
Yeah I'm not sure I'm afraid. If there's an issue with the co-ord mapping, it would impact the entire virtualpanel as one display (i.e. everything's shifted to the right, left, up down etc.), not just say, particular drawing of shapes around a circle in a portion of the virtual panel.

Triple check your code for weird math rounding issues on the x-coord.

mrfaptastic, 2021-04-12:
Yeah I'm not sure I'm afraid. If there's an issue with the virtualdisplay's co-ord mapping logic, it would impact the entire virtualpanel as one display (i.e. everything's shifted to the right, left, up down etc.), not just say, particular drawing of shapes around a circle in a portion of the virtual panel.

Triple check your animation code for weird math rounding issues on the x-coord.

closed 2021-05-11 by mrfaptastic