mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #113: 'class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565'

FrancesDK, 2021-04-11:
Hello,
I'm using ESP32 and 64x64LedPanel

In the AnimatedGIFPanel exemple I'm getting this error while compiling
All other examples works well.

Can anyone explain to me what am I going wrong?
Thank you

C:\Users\wwwru\AppData\Local\Temp\arduino_modified_sketch_717407\AnimatedGIFPanel.ino: In function 'void GIFDraw(GIFDRAW*)':
AnimatedGIFPanel:98:14: error: 'class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565'
  dma_display.drawPixelRGB565(x + xOffset, y, usTemp[xOffset]);
              ^
AnimatedGIFPanel:126:14: error: 'class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565'
  dma_display.drawPixelRGB565(x, y, usPalette[*s++])
exit status 1
'class MatrixPanel_I2S_DMA' has no member named 'drawPixelRGB565'
[image]

mrfaptastic, 2021-04-11:
Change the name of the function called to 'drawPixel'.

I'll have to update the example.

FrancesDK, 2021-04-11:
> 
> 
> Change the name of the function called to 'drawPixel'.
> 
> I'll have to update the example.

Thank you or your quick reply and great work Sir,
It's working now

closed 2021-04-11 by FrancesDK