mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #115: Method for fast full screen updates from a buffer?

InstantArcade, 2021-05-11:
Loving this library, so far. I'm using a 64x64 matrix, but updating the entire screen using one of the drawPixel() variants 4096 times is pretty slow. I noticed that clear() is really fast because it doesn't care about preserving other colors.

My request it for a blit() function that updates the matrix from a buffer all at once, but in a faster way (not calling drawPixel() under the covers) if that's at all possible. I did take a look myself, but the streaming was a bit over my head.

Cheers, and great work!

vortigont, 2021-05-11:
this feature is in my TODO list actually :) I was thinking about implementing the same along with drawFast* methods but this would be out of scope of Adafruit compatible methods. So it still needs some planning, besides it would introduce lot's of questions for combined panels considering all possible orientations and matrix configuration. How should we consider a buffer? Just a plain 1 dimension vector with a suppliment geomatry or 2 dimension plain X,Y buffer? If the later than who will do the transformation - user or library? Some transformations could ruin all benefits of pipelined DMA buffer update.
Would be nice to discuss this feature.

InstantArcade, 2021-05-12:
I thought I'd ask in case there was something I had missed. 
I was looking at the difference between drawPixel() and clear(), and tried to make my own version specifically for a 64x64 buffer to see if I could get it to go quicker, but I wasn't able to improve on the speed.

One other thing I was thinking about was doing some of the processing on the ESP32s other core, but that's a whole other kettle of fish.

I can get ~45 FPS changing every pixel if I'm not doing anything else, so that might be the limit of the ESP32.

Thanks for the response, I might try a RasPi - I wonder if the Pico is fast enough.

closed 2021-05-12 by InstantArcade

vortigont, 2021-05-12:
Let's continue discussion in #117 if someone is interested in ideas and concepts :)