mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #118: Problem with chaining panels 2x3

fajharelektronik4, 2021-05-14:
Hello, sir ..
I am very happy to use this library, very good. but when I tried to assemble with 6 panels 64 * 32, but I encountered a problem like in the video.

Thank you very much

https://www.youtube.com/watch?v=7Rw4k1dFBvY

#include <Arduino.h>

#include <Adafruit_GFX.h>
#include "Fonts/Font5x7Fixed.h"
#include "Fonts/FreeSansBold9pt7b.h"

/* -------------------------- Display Config Initialisation -------------------- */
//    Assume we have four 64x32 panels daizy-chained and ESP32 attached to the bottom right corner
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 3 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls see ChainedPanels example
#define SERPENT true
#define TOPDOWN false

#define VPANEL_W PANEL_RES_X*NUM_COLS // Kosso: All Pattern files have had the MATRIX_WIDTH and MATRIX_HEIGHT replaced by these.
#define VPANEL_H PANEL_RES_Y*NUM_ROWS //

/* -------------------------- Class Initialisation -------------------------- */
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// placeholder for the matrix object
MatrixPanel_I2S_DMA *matrix = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel  *virtualDisp = nullptr;

void setup()
{
  // Setup serial interface
  Serial.begin(115200);
  delay(250);

  // Configure your matrix setup here
  HUB75_I2S_CFG mxconfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);

  // in case that we use panels based on FM6126A chip, we can change that
  mxconfig.driver = HUB75_I2S_CFG::FM6126A;

  // FM6126A panels could be cloked at 20MHz with no visual artefacts
 // mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

  // OK, now we can create our matrix object
  matrix = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  matrix->setBrightness8(96);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if ( not matrix->begin() )
    Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

  virtualDisp = new VirtualMatrixPanel((*matrix), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
  virtualDisp->setTextWrap(false);
}

void loop()
{
  virtualDisp->fillScreen(0);
  scroll_text(60, 50);
  // virtualDisp->clearScreen();
}

void scroll_text(int y, int speed) {

  String pesan = "qwertyuiop asdfghjkl zxcvbnm";
  static uint32_t pM1;
  static uint32_t x1;
  uint32_t cmb = millis ();
  uint16_t w = PANEL_RES_X * NUM_COLS ;

  int R = ( pesan.length() * 10) + w ;

  if (cmb - pM1 > speed ) {
    pM1 = cmb;
    if (x1 < R) {
      ++x1;
    } else {

      x1 = 0;
    }
  }
  virtualDisp->setFont(&FreeSansBold9pt7b);
  virtualDisp->setTextColor(virtualDisp->color333(7, 0, 4));
  virtualDisp->setCursor(w - x1 , y);
  virtualDisp->print(pesan) ;

}

Orel-A, 2021-05-15:
Seems like a problem with POWER wires. Connect a multimeter to power wires, select AC mode and see what is the voltage fluctuation. ESP32 GND and power GND are tied ?

fajharelektronik4, 2021-05-15:
> Seems like a problem with POWER wires. Connect a multimeter to power wires, select AC mode and see what is the voltage fluctuation. ESP32 GND and power GND are tied ?

it seems there is no problem with my power supply, i use 5v 40A and gnd esp32 i have connected it to gnd power supply.

 I try to connect the multimeter with my power supply, there is no voltage fluctuation (0 VAC).

is there something wrong with my code?

Thank you very much

mrfaptastic, 2021-05-16:
It's not an issue with this library, its an issue with how this sketch is coded to use the Adafruit GFX print() function as `virtualDisp->setCursor(w - x1 , y);` ends up wasting CPU/draw time trying to 'print' to co-ordinates that are off-screen the more the string moves off the screen....

You will need to rewrite the logic to some how only change pixels for characters that are on screen. I have no idea how this would be done.

fajharelektronik4, 2021-05-20:
> Ini bukan masalah dengan pustaka ini, ini masalah dengan bagaimana sketsa ini dikodekan untuk menggunakan fungsi Adafruit GFX print () karena `virtualDisp->setCursor(w - x1 , y);`akhirnya membuang-buang waktu CPU / menggambar untuk mencoba 'mencetak' ke koordinat yang tidak ada yang mati- layar semakin banyak string bergerak dari layar ke kiri ....
> 
> Anda perlu menulis ulang logika ke beberapa cara menggambar hanya ke piksel / koordinat untuk karakter yang ada di layar. Saya tidak tahu bagaimana ini akan dilakukan.

thanks for the response from you .. I really like using this library. very good . Now I have succeeded without blinking with a 2x3 configuration  https://youtu.be/eMOoA-z88PE

fajharelektronik4, 2021-05-20:
Actually, I really don't know the actual function but it works.
void loop (){
matrix->flipDMABuffer();
virtualDisp->fillScreen(0);
scroll_text(30,50);
matrix->showDMABuffer();
}

closed 2021-05-20 by fajharelektronik4

Orel-A, 2021-05-20:
Yes I also checked this, the problem is that the function clearScreen(); makes the matrix flicker. Just change to fillScreen(0); instead.

mrfaptastic, 2021-05-20:
@fajharelektronik4 Thank you for your kind words :-). Glad the library is of use.

@vortigont - Any ideas why  clearscreen()/clearFrameBuffer would cause more flicker vs. fillScreen(0) ?

fajharelektronik4, 2021-05-20:
> @ fajharelektronik4 Terima kasih atas perhatiannya :-). Senang perpustakaan ini berguna. Video yang bagus!
> 
> @vortigont - Adakah ide mengapa clearscreen () / clearFrameBuffer akan menyebabkan lebih banyak flicker vs. fillScreen (0)?

previously I have tried clearscreen () and fillScreen (0) but in my opinion there is no difference .. after I added flipDMABuffer and showDMABuffer it doesn't fliker anymore. perfect for displaying running text

Orel-A, 2021-05-20:
Here is an example for the problem. Example code shows flickering. Replacing to fillScreen(0); resolves the problem.
```
#include <WiFi.h>
#include <ArduinoOTA.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <TetrisMatrixDraw.h>
#include <ezTime.h>

char ssid[] = "ssid";
char password[] = "password";
#define MYTIMEZONE "Asia/Jerusalem"

MatrixPanel_I2S_DMA* matrix = nullptr;
TetrisMatrixDraw* tetris = nullptr;
Timezone myTimeZone;

void setup() {
  Serial.begin(115200);
  WiFi.begin(ssid, password);
  waitForSync();
  myTimeZone.setLocation(F(MYTIMEZONE));
  ArduinoOTA.setHostname("Tetris Clock");
  ArduinoOTA.begin();
  HUB75_I2S_CFG mxconfig;
  mxconfig.double_buff = true;
  matrix = new MatrixPanel_I2S_DMA(mxconfig);
  matrix->begin();
  matrix->setPanelBrightness(16);//setBrightness8(64)
  tetris = new TetrisMatrixDraw(*matrix);
  tetris->scale = 2;
}

void loop() {
  ArduinoOTA.handle();
  static bool bShowColon = false;
  static int iColonCounter = 0;
  if (iColonCounter++ == 20) {
    iColonCounter = 0;
    bShowColon = !bShowColon;
  }
  matrix->flipDMABuffer();
  matrix->clearScreen();//matrix->fillScreen(0);
  tetris->setTime(myTimeZone.dateTime("H:i"), false);
  tetris->drawNumbers(2, 26, bShowColon);
  matrix->showDMABuffer();
  delay(50);
}
```

vortigont, 2021-05-20:
Current implementation of clearScreen() resets BOTH DMA buffers no matter if double buffering is used or not. This is done to make sure both buffers has correct brightness/control data set. Not sure if this is intended/expected, as I remember I've tried to keep it compatible to the previous behavior. Maybe it should be changed (or overloaded and documented)

mrfaptastic, 2021-05-20:
@vortigont - Always love your work lol.

Couldn't we also change it so that 'clearScreen' actually calls fillScreen(0)? Then use your 'hardware' clearScreen when it's actually needed - like when brightness is changed or on library setup only?

vortigont, 2021-05-20:
yes, maybe you right, 'clearScreen' name is a little bit misleading here. Doubt if there could be any real use cases when one needs to clear both buffers at the same time except during initialization.
OK, I'll make a PR then )

csloz, 2021-05-20:
Suggest to make it backwards compat and do something along the lines of

clearScreen ()   - no parameters= assume both buffers
clearScreen (1) - pass a parameter = clear buffer,  [Buffer1, 2]

Orel-A, 2021-05-20:
Then that would be resetBuffers()

vortigont, 2021-05-21:
> clearScreen () - no parameters= assume both buffers clearScreen (1) - pass a parameter = clear buffer, [Buffer1, 2]

I guess it will be even more misleading then... i.e. I want to just clear current (back) buffer - what should I put there, 1 or 2? Besides there is already a method (although it is protected) doing exactly the same - clearFrameBuffer(bool _buff_id).

So there are two main questions as I see it:
1) Do one really needs a public method to selectively reset DMA buffer(s)
2) The concept of clearScreen() - we just fill screen with back or we do mean full reset for single(both) buffers

IMHO:
1) NO. I can't imagine a case when DMA gets trashed and might require reset via user request. For all other cases a child class could access protected methods for fine control if required.
2) The idea of simple clearing active buffer (the one that is affected by all other methods like drawPixel()) 
is more intuitive and expected.

mrfaptastic, 2021-05-21:
>     1. NO. I can't imagine a case when DMA gets trashed and might require reset via user request. For all other cases a child class could access protected methods for fine control if required.

Agreed. That's my point. Your 'buffer clear function' (which is really a HUB 75 reset function) shouldn't be visable to the public in any way. It's only used at library boot up and change of brightness to get the bits in the right place.

mrfaptastic, 2021-05-21:
We're getting pedantic here, but...

MatrixPanel_I2S_DMA::clearFrameBuffer could be named something like MatrixPanel_I2S_DMA::resetMatrixDMABuffer
clearScreen = updateMatrixDMABuffer(0, 0, 0)
fillScreen = Stays as is.

:-D

Orel-A, 2021-05-21:
Lets be more specific. The intended action of clearScreen() function is to initialize the DMA buffers to 0 which causes the matrix to show black. Whereas the expected action derived from its name is to simply clear the matrix to black. Furthermore, 'Screen' should be called as the whole chain of matrixes.
Now that we finally understand its purpose and the misconception, we can proceed to deal with the issue.

vortigont, 2021-05-21:
OK, here is a possible option
clearScreen() now wipes only color data in a current back buffer.
What was clearScreen() before now is called resetbuffers() and is a protected method that could be used only internally or via child class inheritance.
Tests/feedbacks are welcome )