mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #121: Panel Connection 128x64px

reto88, 2021-05-17:
I have a panel with 128x64px and without the D pin (see picture). How do I have to connect it, or isn't it supported by the library?
[image]

mrfaptastic, 2021-05-25:
What happens when you connect? It's probably a 1/8 scan panel which this library doesn't support support (there is an example that works around this though).

Have you tried a test sketch anyway?

reto88, 2021-05-26:
The panel I have is from Aliexpress [https://de.aliexpress.com/item/4000126043085.html?spm=a2g0s.9042311.0.0.1b6e4c4dtnREMr](https://de.aliexpress.com/item/4000126043085.html?spm=a2g0s.9042311.0.0.1b6e4c4dtnREMr ) and a 1/32.
When I connect the panel some pixels are flickering.

https://user-images.githubusercontent.com/25948008/119606094-3393ed80-bdf2-11eb-8213-8383c57d72e2.mp4

mrfaptastic, 2021-05-26:
Can you try with PxMatrix library and see what configuration you need to make it work?

ertdfgcvb, 2021-06-15:
> Can you try with PxMatrix library and see what configuration you need to make it work?

Any news about this? I also have a batch of these 128x64 P2 panels (with the ICN2038S <- not the 'S') chip and it won't work. I tried it with the SmartMatrix library but apparently it’s not compatible with this chipset. 
Thank you for any info.

mrfaptastic, 2021-06-17:
I take it you have set the ```mxconfig.driver = HUB75_I2S_CFG::ICN2038S;   ```  in your setup ?

mrfaptastic, 2021-06-17:
> The panel I have is from Aliexpress https://de.aliexpress.com/item/4000126043085.html?spm=a2g0s.9042311.0.0.1b6e4c4dtnREMr and a 1/32.
> When I connect the panel some pixels are flickering.
> WhatsApp.Video.2021-05-26.at.07.15.11.mp4

Have you set the mxconfig.driver = HUB75_I2S_CFG::FM6124; in your setup ? Your photo seems to indicate it's a FM6124 chipset.

mrfaptastic, 2021-07-22:
Did that fix the issue @reto88 ?

reto88, 2021-08-02:
sorry i didn't had time until now, I will check it

closed 2021-09-06 by mrfaptastic

zombodotcom, 2023-01-09:
I had this same issue, GND on the board is D. 

just solved my panel problem

wilson3682, 2023-07-14:
@reto88 @zombodotcom Hi Guys, I would like to know how did you solve the problem with this type of Panels, I just bought one, 128x64 with the FM6124 chipset and no letter D instead is ground, Do you guys have a sketch or some advice so I could make this panel to work? Thanks in advance. I used the mxconfig.driver = HUB75_I2S_CFG::FM6124; chipset but did not work.

This is the SimpleTestShapes sketch I'm using:
`
// Example sketch which shows how to display some patterns
// on a 64x32 LED matrix
//

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 128      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another
 
//MatrixPanel_I2S_DMA dma_display;
MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
// From: https://gist.github.com/davidegironi/3144efdc6d67e5df55438cc3cba613c8
uint16_t colorWheel(uint8_t pos) {
  if(pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if(pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{
  
  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(5, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA";
  for (w=0; w<strlen(str); w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print(str[w]);
  }

  dma_display->println();
  dma_display->print(" ");
  for (w=9; w<18; w++) {
    dma_display->setTextColor(colorWheel((w*32)+colorWheelOffset));
    dma_display->print("*");
  }
  
  dma_display->println();

  dma_display->setTextColor(dma_display->color444(15,15,15));
  dma_display->println("LED MATRIX!");

  // print each letter with a fixed rainbow color
  dma_display->setTextColor(dma_display->color444(0,8,15));
  dma_display->print('3');
  dma_display->setTextColor(dma_display->color444(15,4,0));
  dma_display->print('2');
  dma_display->setTextColor(dma_display->color444(15,15,0));
  dma_display->print('x');
  dma_display->setTextColor(dma_display->color444(8,15,0));
  dma_display->print('6');
  dma_display->setTextColor(dma_display->color444(8,0,15));
  dma_display->print('4');

  // Jump a half character
  dma_display->setCursor(34, 24);
  dma_display->setTextColor(dma_display->color444(0,15,15));
  dma_display->print("*");
  dma_display->setTextColor(dma_display->color444(15,0,0));
  dma_display->print('R');
  dma_display->setTextColor(dma_display->color444(0,15,0));
  dma_display->print('G');
  dma_display->setTextColor(dma_display->color444(0,0,15));
  dma_display->print("B");
  dma_display->setTextColor(dma_display->color444(15,0,8));
  dma_display->println("*");

}

void setup() {

  // Module configuration
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN    // Chain length
  );

  //mxconfig.gpio.e = 18;
  //mxconfig.clkphase = false;
  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;
  mxconfig.driver = HUB75_I2S_CFG::FM6124;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(90); //0-255
  dma_display->clearScreen();
  dma_display->fillScreen(myWHITE);
  
  // fix the screen with green
  dma_display->fillRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(0, 15, 0));
  delay(500);

  // draw a box in yellow
  dma_display->drawRect(0, 0, dma_display->width(), dma_display->height(), dma_display->color444(15, 15, 0));
  delay(500);

  // draw an 'X' in red
  dma_display->drawLine(0, 0, dma_display->width()-1, dma_display->height()-1, dma_display->color444(15, 0, 0));
  dma_display->drawLine(dma_display->width()-1, 0, 0, dma_display->height()-1, dma_display->color444(15, 0, 0));
  delay(500);

  // draw a blue circle
  dma_display->drawCircle(10, 10, 10, dma_display->color444(0, 0, 15));
  delay(500);

  // fill a violet circle
  dma_display->fillCircle(40, 21, 10, dma_display->color444(15, 0, 15));
  delay(500);

  // fill the screen with 'black'
  dma_display->fillScreen(dma_display->color444(0, 0, 0));

  //drawText(0);

}

uint8_t wheelval = 0;
void loop() {

    // animate by going through the colour wheel for the first two lines
    drawText(wheelval);
    wheelval +=1;

    delay(20); 
/*
  drawText(0);
  delay(2000);
  dma_display->clearScreen();
  dma_display->fillScreen(myBLACK);
  delay(2000);
  dma_display->fillScreen(myBLUE);
  delay(2000);
  dma_display->fillScreen(myRED);
  delay(2000);
  dma_display->fillScreen(myGREEN);
  delay(2000);
  dma_display->fillScreen(myWHITE);
  dma_display->clearScreen();
  */  
}`
[image]
[image]
[image]
[image]

zombodotcom, 2023-07-14:
I think when I did it I said screw the markings and made the 1st ground D. 
Booted right up.  The PCB markings are wrong. 

I'm not at my computer and on vacation right now. Let me know if that helps.

zombodotcom, 2023-07-14:
Also how are you interfacing with the hub? Direct to the esp32 ? Or are you using a premade hub75 adapter PCB ?

wilson3682, 2023-07-14:
@

> Also how are you interfacing with the hub? Direct to the esp32 ? Or are you using a premade hub75 adapter PCB ?

@zombodotcom Thanks for replying so fast even dough you are on vacation. This is the interface board I'm using, and is compatible with the library's default pins:
https://www.electrodragon.com/product/rgb-matrix-panel-drive-interface-board-for-esp32-dma/

Thanks again.

zombodotcom, 2023-07-14:
I used that same board  but the v2 I think, the red one. 

Make note of what it says about the pinout: 
 v2 board
https://w2.electrodragon.com/Board/IDD/IDD1013-DAT/IDD1013-DAT.md#devkitc-board-pin-definitions
https://w.electrodragon.com/w/ESP32_DMA_RMP

https://w.electrodragon.com/w/ESP32_DMA_RMP#Software_ESP32_DMA_Demo_Code_Setup more pin definitions but notice something here as well

[image]

I don't remember if I ended up not having to solder those or not. this was a few months ago. 

I think I used this sketch for the electrodragon but I cant be sure

https://github.com/jnthas/clockwise/blob/3aeff99fef23c979cd743959fd41b296d8cc89a3/firmware/src/main.cpp

Notice where I setup the pins here to make it bypass the defaults.

https://github.com/jnthas/clockwise/blob/3aeff99fef23c979cd743959fd41b296d8cc89a3/firmware/src/main.cpp#L30

Sorry I can't be more help right now, I don't have the devices in front of me and I wont be back for a while to them.

zombodotcom, 2023-07-14:
Also forgot to mention that some of those boards had bad solder or something and was causing weird issues, so I added more solder the the hub75 adapter part and it fixed the flickering.

wilson3682, 2023-07-14:
Thanks @zombodotcom for all the info, the board I'm using right now I know for sure is working fine, I've been using it with other panels and have not have any issues, and yes I'm aware of the pin configuration on the board.
I will take a look and try the clockwise sketch you mentioned and let you know how it went. Thanks again. Enjoy your vacation.

zombodotcom, 2023-07-14:
@wilson3682 also noticed this in the readme for esp32 dma library. seems like this person got it working for 128x64 with fm6124

https://github.com/bogd/esp32-morphing-clock/blob/main/code/src/rgb_display.cpp

board707, 2023-07-16:
Yes, he got it worked, but his panel had a full set of ABCDE pins.
I am wondered how manage to work 32s panel wirh only A B C pins. The o ly way I know is use a 595 type multiplexor. But this panel seems doesn't have it.

wilson3682, 2023-07-16:
Hi @board707 I took off the frame and under it I can see there are 8 RT5957DSP multiplexers I found the datasheet here: http://www.raffar.com.tw/upload_file/2019071017305286848.pdf
Also there are 3 74HC245C Octal Bus Transceivers datasheet: https://www.mouser.com/datasheet/2/308/74HC245.REV1-69988.pdf
The rest of the chips are FM6124 shift registers 48 in total. datasheet: https://www.docdroid.net/Oy5GKYW/fm6124-pdf
I'm only referencing the datasheets for Information purposes.

Here is the picture of the board:
 
[image]

I also see that the PxMatrix library may support this type of boards. https://github.com/2dom/PxMatrix/blob/master/README.md#shiftreg_abc---for-rt5957-panels

Thanks!

board707, 2023-07-16:
Now everything has become clearer. The 5957 chip is a shift register type multiplexer. To control it, three pins are enough, this explains why there are only pins A B C on the panel.
Overall, however, this panel is quite a challenge, as it is a combination of a 6124 driver with a shift multiplexer. On any library, you will have to work to run it.
I do not know if it is possible to run it on this library. This is a question for the author, @mrfaptastic 
I can offer to try my library, but it only works on stm32 and rp2040 boards, esp32 is not supported

mrfaptastic, 2023-07-16:
The hardware isn't really relevant here. Should be able to get this to work with any micro.

What existing code works with these panels? Can read it to see if any of it can be ported.

board707, 2023-07-16:
Sorry, I confused 6124 with 6126A, FM6124 is a completely normal driver that should works fine with your library.
As for the shift register multiplexer, you already asked me and I gave you a link to the code for working with it. As far as I can see you have code to support 5266 multiplexer - the 5957 should works exactly the same.