mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #123: HUB75E Green only half display other collors ok

donnersm, 2021-05-24:
I tried several sceens running a simple sketh to light up all pixels in Green then Red, then Blue
I cant get the green to fill the whole screen, red and blue work fine. Also, i can light up any pixel 64x64 but on the lower half of the screen all green is off. i also tried using different pins on the ESP32 but no success.
The display is a HUB75E  chip type RUC7258

Anyone encountered this before of maybe someone knows what is wrong?
I triple checked wiring

```
//#include <FastLED_NeoMatrix.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
//#include <FastLED.h>

//panel
#define A_PIN     23
#define B_PIN     19
#define C_PIN     5
#define D_PIN     17
#define E_PIN    2
#define LAT_PIN      4
#define OE_PIN       15
#define CLK_PIN      16
#define R1_PIN      25
#define B2_PIN      13
#define R2_PIN       14
#define G2_PIN       21
#define G1_PIN       26
#define B1_PIN      27

#define PANEL_WIDTH 64
#define PANEL_HEIGHT 64    // Panel height of 64 will required PIN_E to be defined.
#define PANELS_NUMBER 1   // Number of chained panels, if just a single panel, obviously set to 1
#define PANE_WIDTH PANEL_WIDTH * PANELS_NUMBER
#define PANE_HEIGHT PANEL_HEIGHT

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {

  //panel
  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
  
  HUB75_I2S_CFG mxconfig;
  mxconfig.mx_height = PANEL_HEIGHT;      // we have 64 pix heigh panels
  mxconfig.mx_width = PANEL_WIDTH;
  mxconfig.chain_length = PANELS_NUMBER;  // we have 2 panels chained
  mxconfig.gpio.e = E_PIN;                // we MUST assign pin e to some free pin on a board to drive 64 pix height panels with 1/32 scan
  //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can change that

    Serial.begin(115200);

// OK, now we can create our matrix object
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  // let's adjust default brightness to about 75%
  dma_display->setBrightness8(50);    // range is 0-255, 0 - 0%, 255 - 100%

  // Allocate memory and start DMA display
  if( not dma_display->begin() )
      Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
  
}

void loop() {
  dma_display->fillScreenRGB888(0, 255, 0);
  delay(1000);
  dma_display->fillScreenRGB888(255, 0, 0);
  delay(1000);
  dma_display->fillScreenRGB888(0, 0, 255);
  delay(1000);
  dma_display->fillScreenRGB888(64, 64, 64);
  delay(1000);
  dma_display->fillScreenRGB888(0, 0, 0);
  delay(1000);
}

```
Attached are some photo's of the result
[image]
[image]
[image]

donnersm, 2021-05-25:
Problem has been found. After seeing this on two panels, today I tried a third....o man....talk about bad luck..turned out that the first two panels had a problem. Both panels have the same defect, a driver chip defective...
So in the end no issue... problem solved. 
Btw for those wondering, it was related to the G2 pin

closed 2021-05-25 by donnersm

mrfaptastic, 2021-05-25:
Thanks for the update. Somebody else posted a similar G pin issue a while back. Seems like there are some defect chinese chips floating about.

If it were an issue with this library, impact every panel, not just some.