mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #124: Ability to flip x or y axe

donnersm, 2021-05-27:
So far I am missing the option to flip an axe. depending on how i mount a panel ( or series of panels) I want to tell the driver what is pixel 1, for example right top, left top, bottom etc. Simular to what I can do with Fastled. That way it is easy to flip left and right top and bottom.
Maybe it is implemented in the driver but I don't see it. So if it's there please someone tell me how and if it's not, it would be  a nice enhancement to the driver.

mrfaptastic, 2021-05-27:
It's in the virtual display example

closed 2021-05-27 by donnersm

lehmartin, 2021-05-29:
Maybe a stupid question, but do I understand this flipping function correctly - that everything will be displayed in reverse, like text is written from right to left with mirrored characters? That's exactly what I want to do and I am not able to find a way to achieve this...

mrfaptastic, 2021-05-30:
@lehmartin - Basically you're looking to mirror the display? There is no functionality in the library to do this as this library is not really a 'graphics effects' library, but a hardware interface from the ESP32 to the panel.

If you want to implement pixel co-ordinate flipping, then you'll need to write custom code. Your best bet will be to implement such functionality in the VirtualDisplay class. Code would be as simple the following I think:
```
x = WIDTH-x
y = HEIGHT-y
```

Hack the getCo-ordinates function or whatever to do the job. Or write your own class that inherets the IS2DMA library and overrides the drawPixel() function.

lehmartin, 2021-05-30:
@mrfaptastic, Yes, that's exactly the feature I was looking for. Guess I'll have to dig deeper into your library. Thanks for your hints!