mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #126: How to use lvgl font?

xuniuer, 2021-06-03:
HI, there.

I want to use lvgl font in this project, which provide a font_conv (```https://lvgl.io/tools/fontconverter```) tool to generate ttf font indexed by Unicode ord. Because I need to display some Chinese character.
```
/*-----------------
 *    BITMAPS
 *----------------*/

/*Store the image of the glyphs*/
static LV_ATTRIBUTE_LARGE_CONST const uint8_t glyph_bitmap[] = {
    /* U+0020 " " */

    /* U+0021 "!" */
    0xe, 0xa0, 0xd9, 0xd, 0x90, 0xc8, 0xc, 0x80,
    0xb7, 0xa, 0x60, 0x11, 0xb, 0x80, 0xd9,

    /* U+0022 "\"" */
    0x1f, 0x9, 0x91, 0xf0, 0x88, 0x1f, 0x8, 0x80,
    0xf0, 0x88, 0x0, 0x0, 0x0,

    /* U+0023 "#" */
    0x0, 0xd, 0x20, 0x3c, 0x0, 0x0, 0xf, 0x0,
    0x69, 0x0, 0x2f, 0xff, 0xff, 0xff, 0xf7, 0x1,
    0x5c, 0x11, 0xa6, 0x10, 0x0, 0x69, 0x0, 0xc3,
    0x0, 0x0, 0x88, 0x0, 0xd2, 0x0, 0x9f, 0xff,
    0xff, 0xff, 0xf0, 0x12, 0xc5, 0x23, 0xe2, 0x20,
    0x0, 0xd2, 0x3, 0xc0, 0x0, 0x0, 0xf0, 0x4,
    0xb0, 0x0,
   
  /* so and on - I want to draw M on display*/
    /* U+004D "M" */
    0x8e, 0x0, 0x0, 0x0, 0x8, 0xe8, 0xf7, 0x0,
    0x0, 0x2, 0xfe, 0x8f, 0xf1, 0x0, 0x0, 0xbf,
    0xe8, 0xdc, 0xa0, 0x0, 0x4f, 0x9e, 0x8d, 0x2f,
    0x30, 0xd, 0x87, 0xe8, 0xd0, 0x9c, 0x6, 0xe0,
    0x7e, 0x8d, 0x1, 0xe7, 0xe5, 0x7, 0xe8, 0xd0,
    0x6, 0xfc, 0x0, 0x7e, 0x8d, 0x0, 0xa, 0x20,
    0x7, 0xe8, 0xd0, 0x0, 0x0, 0x0, 0x7e,
```

I think if I could get font's glyph_bitmap, then to call ```dma_display.drawBitmap(x, y, (uint8_t*)bitmap, w, h, color)``` the font should be displayed(```w, h``` could be obtained from lvgl library functions). But I got a unrecognized show when passed the bitmap of ```**M**```.
[image]

BTW: the display is tested to be OK by running ```SimpleTestShapes```

So wonder to know how to use lvgl font in ESP32-HUB75-MatrixPanel-I2S-DMA? 
Any advice appreciated!

BR~

mrfaptastic, 2021-06-03:
Hi, this library is based on AdaFruitGFX. LvGL is a totally different graphics library.

You will either need to hack this library to take LvGL bitmap data array as the input (you'll need to look at the internals of that library and see how it works).

Or perhaps use this library that support unicode: https://github.com/joeycastillo/Adafruit-GFX-Library

or create the Chinese characters fro AdaFruitGFX using:
https://learn.adafruit.com/adafruit-gfx-graphics-library/graphics-primitives#bitmaps-2002806-32

xuniuer, 2021-06-04:
Thanks @mrfaptastic. I tested a font pixel matrix generated by a pctolcd tool, it woks by calling ```drawBitmap```. 

[image]

By comparing display with pixel matrix, the result is plain. 1 byte to control 8 led lights (pixels), one bit - 0 or 1 - to turn on or off light. So I think I could use python PIL library with freetype library support to generate bitmap matrixs with any size for any font I want.

Plus, I will follow the links to find more new or simple method.

BR~

closed 2021-06-04 by xuniuer

mrfaptastic, 2021-06-04:
That's great to hear. Do you mind sharing the link to that pctolcd tool?

xuniuer, 2021-06-04:
No problem. It's a Windows tool with Chinese. According to Help info, it's a 'graduation article' from a Chinese college student. I don't know if you could use or not.

The link is from https://xiazai.zol.com.cn/detail/48/470519.shtml
Because it's a free tool, I attach it here.
[PCtoLCD2002.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/6596238/PCtoLCD2002.zip)