mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #127: Conflicted with third library while double_buffer enabled?

xuniuer, 2021-06-07:
Hello there,

When to enable double_buffer feature, library `ESPAsyncTCP-esphome` will report error at ```begin()``` as following:
 
```[E][AsyncTCP.cpp:1291] begin(): failed to start task```

Look into its begin() function, 
```
void AsyncServer::begin(){
    if(_pcb) {
        return;
    }

    if(!_start_async_task()){
        log_e("failed to start task");
        return;
    }
    int8_t err;
    _pcb = tcp_new_ip_type(IPADDR_TYPE_V4);
    if (!_pcb){
        log_e("_pcb == NULL");
        return;
    }
 ...
```
Where ```_start_async_task``` function is as follows:
```
static bool _start_async_task(){
    if(!_init_async_event_queue()){
        return false;
    }
    if(!_async_service_task_handle){
        customTaskCreateUniversal(_async_service_task, "async_tcp", 8192 * 2, NULL, 3, &_async_service_task_handle, CONFIG_ASYNC_TCP_RUNNING_CORE);
        if(!_async_service_task_handle){
            return false;
        }
    }
    return true;
}
```
Stack for 'async_tcp' is assigned to 16K.  I cannot fix, just report here.

BR~

mrfaptastic, 2021-06-07:
More detail required please

xuniuer, 2021-06-07:
@mrfaptastic Let's describe it in details.

I use WeMOS mini D1 to drive 4 panels (64*32) to display something. To support web service by which user can control what to display, I add a webserver. Firstly using ```ESPAsyncWebServer-esphome by Hristo Gochkov``` which is adopted by many users (because its downloaded number is high). Of course, to prevent flickering, I enable double_buffer feature, but the library report error above mentioned. In result, webserver is unavailable although wifi connection is OK(got localIP). Just described above, the async webserver library is sort of heavy to mini D1, I guess, then I change to standard webserver ```#include <WebServer.h>``` provided by Arduino. Now everything is OK. So, simply, the ```ESPAsyncWebServer-esphome``` is not suitable to my case using your **great** library. And th report is just a hint to others when choosing webserver library.

Although it's ugly till now, I share it. There are some Chinese voice in it, don't mind please.

https://user-images.githubusercontent.com/1916257/120992103-db2afb80-c7b4-11eb-91aa-c5373ad16371.mp4

mrfaptastic, 2021-06-09:
Hi @xuniuer. Double buffering uses double the memory and I think the Async library also uses more memory than the non-async library, so the combination of Async+double buffering = crash.

Using the NON-async library must provide just enough spare memory such that all works well.

Thank you for your kind words. Glad the library is of use to you.

xuniuer, 2021-06-09:
@mrfaptastic It's right. Thanks for your attention and explanation. 
The formula of memory usage for double buffering is useful.

Thanks again!

closed 2021-06-09 by xuniuer