mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #128: Single 128 x 128 matrix

jantenne, 2021-06-07:
Hi there. I have some 128x128 panels. Like these: https://m.german.alibaba.com/amp/p-detail/60790902389.html

they seem to be a bit different. They have two hub75 connections NOT labeled in and out. There are A B and C and no D or E but a N and some grounds. 

When I plug my driver based on this library and esp32 I can lit the upper OR lower part of it but the image is crapy. 

Does anyone know the confit. I think I have to chain the connections, but I don’t know Settings for the library and the pin connections. 

Thank you in advance !!!

jantenne, 2021-06-08:
I figured out the the two HUB75 connectors are not a chain but two inputs. so each of then will drive one half of the display. more or less like a chain of two 128x64 panels. I don't know if that helps with this library, but maybe.

I would appreciate any help or tips and do apologize for maybe bothering you all.

xuniuer, 2021-06-09:
This library just has one Hub75 output. According to readme, the type of panel is not supported. 
 
If assumed it's chained internally, you could use example for chain panels ```ChainedPanels``` to have a try.
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/ChainedPanels/ChainedPanels.ino

Think it was a 2 * 2 chains, each is 64*64, then 4 panels inside.

The following is a try to change **PANEL_RES_X and NUM_COLS**
```
  Example 1 configuration:
    
    #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
    #define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.
    #define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
    #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
    virtualDisp(dma_display, NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, true, true);

```
If failed, you could think it's a chain of 64*32 panels inside? 
Complicated without knowing their sequence.

Just a suggestion for your case, because I have not same type of panel you have.

mrfaptastic, 2021-06-09:
> I figured out the the two HUB75 connectors are not a chain but two inputs. 

Weird design these panels are. You will have to use a Raspberry PI or something which has enough GPIOs to drive two or more panels in PARALLEL.

jantenne, 2021-06-15:
Thank you for the answers and mrfaptastic for the labeling.

As far as I got now these pales are twi 64x32 in the inside and use A, B, C with SHIFT function. I got some pixels lit, but am far away of have them working. I will do more research and come back if I found something.

For now, thank you again for your feedback.

closed 2021-06-21 by mrfaptastic