mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #129: Strange flicking when scrolling text

xuniuer, 2021-06-09:
Hi there. I found a strange flicking pixels at the edge of screen when to scroll text. Have a look.

https://user-images.githubusercontent.com/1916257/121303464-b1491480-c92d-11eb-8667-0ad3ac878ab4.mp4

The screen is chains of 4 panels (2 * 2), that is 128 width * 64 height. I want to scroll the first 'row' in blue color and the second is in green. It's easier to find when blue char is scrolling at the first row, the edge of the second row has some blue pixels dancing, and vice verse.

The code is as follows:
```
//
void scroll_text(int16_t y, int16_t speed, String flight, String dest_en)
{
    int w_text = 128; // screen width
    int len = 0, len1 = flight.length() * 18, len2 = dest_en.length() * 18; // 18 is width of character when font size is 3
    len = max(len1, len2);
    
    while (1) {
        // display flight
        virtualDisp->flipDMABuffer();
        virtualDisp->fillScreen(0);
        virtualDisp->setCursor(w_text, y);
        virtualDisp->setTextColor(virtualDisp->color444(0, 0, 15));
        virtualDisp->print(flight);

        // display dest_en
        virtualDisp->setCursor(w_text, y + 32);
        virtualDisp->setTextColor(virtualDisp->color444(0, 15, 0));
        virtualDisp->print(dest_en);
        virtualDisp->showDMABuffer();

        w_text--;
        if (w_text + len == 0) {
            w_text = 128;
        }

        delay(speed);
    }
}

```

Any hints appreciated!

mrfaptastic, 2021-06-09:
Are you using the code that's in the git repository or the latest 'release'?

xuniuer, 2021-06-09:
Hi @mrfaptastic, I setup it according to ```Library Installation``` from site at first phrase.
 
Now to install library from github. The version should be latest:

[image]

PlatformIO library info is here:
```
{
"type": "library", 
"name": "ESP32 HUB75 LED MATRIX PANEL DMA Display", 
"version": "2.0.3", 
"spec": {
    "owner": "mrfaptastic", "id": 3577, "name": "ESP32 HUB75 LED MATRIX PANEL DMA Display", "requirements": null, "url": null
 }
}
```

After re-compiling and upload, the problem still exists, not fixed.  Pity.

xuniuer, 2021-06-09:
Plus, I use latest ```Adafruit GFX Library``` (replace GFX_Root 2.0.0), the result is same.

mrfaptastic, 2021-06-09:
I thought I had fixed this issue, but I don't think it's part of 2.0.3.

Just created version 2.0.4 - try that.

xuniuer, 2021-06-09:
Now to use version 2.0.4.

[image]

But the flicking pixels at the edge of screen are still dancing with scrolling.

xuniuer, 2021-06-09:
@mrfaptastic I check file time, only library files are modified several minutes ago, others are not modified.

mrfaptastic, 2021-06-09:
Can you paste the top of the script where you setup the virtual display etc? I can't debug what' going on as I don't know how you've chained the panels together to make the 4x4 display.

xuniuer, 2021-06-09:
```
//------------------------------------------
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2                     // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2                     // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS *NUM_COLS // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false

// library includes
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel *virtualDisp = nullptr;

// Flight / Destination global
String FLIGHT, SFLIGHT, DEST_CN, DEST_EN;

void aled_matrix_setup()
{
    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X, // module width
        PANEL_RES_Y, // module height
        PANEL_CHAIN  // chain length
    );

    //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
    // OK, now we can create our matrix object
    mxconfig.double_buff = true;
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(192); // range is 0-255, 0 - 0%, 255 - 100%

    // Allocate memory and start DMA display
    if (not dma_display->begin())
        Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    // create VirtualDisplay object based on our newly created dma_display object
    virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
}

```

xuniuer, 2021-06-09:
Almost from your example -:) 

This is ```loop()```

```
void loop()
{
    scroll_text(6, 10, "CA1234", "Beijing DaXing");
}
```

mrfaptastic, 2021-06-09:
I have just made an untested change to the ESP32-VirtualMatrixPanel-I2S-DMA.h in the git repository, can you try with using this and see if it fixes the issue?

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/ESP32-VirtualMatrixPanel-I2S-DMA.h

xuniuer, 2021-06-09:
Congratulations! 

https://user-images.githubusercontent.com/1916257/121329820-1dd10d00-c948-11eb-9a5b-7af8fdbdb253.mp4

Clear and clean! I think you could update it.

Thanks for your work!!!

mrfaptastic, 2021-06-09:
Can you show a video of a fill-screen on the entire panel for me though? Looks like it isn't populating the first left row on the top (green) panel.

mrfaptastic, 2021-06-09:
Can you also show a video of the scrolling across all the panels as well ? Just want to check how it looks.

xuniuer, 2021-06-10:
@mrfaptastic Your worry about filling screen is right. I test the following cases:
1. fillScreen() - pass

https://user-images.githubusercontent.com/1916257/121444795-b906c880-c9c2-11eb-90ff-c2e75c56ada3.mp4

2. fillRect() - fail
```
int step = 0;
void loop()
{
    // test for fillRect
    step ++;
    virtualDisp->fillScreen(0);
    delay(1000);
    virtualDisp->fillRect(0, 0, 32*step, 64, virtualDisp->color444(0, 0, 15));
    delay(3000);
    if (step == 4) step = 0;
}

```
https://user-images.githubusercontent.com/1916257/121444811-bf954000-c9c2-11eb-9d9f-09646efeb3b2.mp4

The problem is lack last col at the second row, but the first row is OK.

3. scroll text - pass

https://user-images.githubusercontent.com/1916257/121444815-c3c15d80-c9c2-11eb-8b49-f886c8210433.mp4

Here I fill a 8*64 rect at (0,0), it's also lack last one col at the second row.

mrfaptastic, 2021-06-10:
Should be fixed now. 
```
//------------------------------------------
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2                     // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2                     // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS *NUM_COLS // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN true

// library includes
#include <ESP32-VirtualMatrixPanel-I2S-DMA.h>

// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel *virtualDisp = nullptr;

// Flight / Destination global
String FLIGHT, SFLIGHT, DEST_CN, DEST_EN;

//
void scroll_text(int16_t y, int16_t speed, String flight, String dest_en)
{
    int w_text = 128; // screen width
    int len = 0, len1 = flight.length() * 18, len2 = dest_en.length() * 18; // 18 is width of character when font size is 3
    len = max(len1, len2);
    
    while (1) {
        // display flight
        virtualDisp->flipDMABuffer();
        virtualDisp->fillScreen(0);
        virtualDisp->setCursor(w_text, y);
        virtualDisp->setTextColor(virtualDisp->color444(0, 0, 15));
        virtualDisp->print(flight);

        // display dest_en
        virtualDisp->setCursor(w_text, y + 32);
        virtualDisp->setTextColor(virtualDisp->color444(0, 15, 0));
        virtualDisp->print(dest_en);
        virtualDisp->showDMABuffer();

        w_text--;
        if (w_text + len == 0) {
            w_text = 128;
        }

        delay(speed);
    }
}

void setup()
{
  Serial.begin(112500);
  delay(1000);
    HUB75_I2S_CFG mxconfig(
        PANEL_RES_X, // module width
        PANEL_RES_Y, // module height
        PANEL_CHAIN  // chain length
    );

    //mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
    // OK, now we can create our matrix object
    mxconfig.double_buff = true;
    dma_display = new MatrixPanel_I2S_DMA(mxconfig);

    // let's adjust default brightness to about 75%
    dma_display->setBrightness8(64); // range is 0-255, 0 - 0%, 255 - 100%

    // Allocate memory and start DMA display
    if (not dma_display->begin())
        Serial.println("****** !KABOOM! I2S memory allocation failed ***********");

    // create VirtualDisplay object based on our newly created dma_display object
    virtualDisp = new VirtualMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);

    Serial.print("Virtual Display width: ");
    Serial.println(virtualDisp->width());

    Serial.print("Virtual Display height: ");
    Serial.println(virtualDisp->height());    

   virtualDisp->fillScreen(0);    
}

void loop()
{
    virtualDisp->fillScreen(0);  
    for (int i =0; i < virtualDisp->width(); i++)
    {
      int color = virtualDisp->color444(0, 0, 15);
      if (i % 64 == 0)
      {
        color = virtualDisp->color444(15, 0, 0);
      }
      virtualDisp->drawLine(i, 0, i, 63, color);
      delay(100);
    }

    for (int i =0; i < virtualDisp->height(); i++)
    {
      int color = virtualDisp->color444(0, 15, 0);
      if (i % 32 == 0)
      {
        color = virtualDisp->color444(15, 15, 15);
      }
      virtualDisp->drawLine(0, i, 127, i, color);
      delay(100);
    }
        
}
```

xuniuer, 2021-06-11:
@mrfaptastic Thanks for your test script. The main difference of my test scripts and yours is I use API fillRect() not drawLine(). 

In addition, I compared ```ESP32-VirtualMatrixPanel-I2S-DMA.h``` with that of version 2.0.3, the line 131: 
```
coords.x = (y / panelResY) * (module_cols * panelResX) + (virtualResX - 0 - x); // hack untested 9/june/21
```
Originally, it's 
```
coords.x = (y / panelResY) * (module_cols * panelResX) + (virtualResX - 1 - x); 
```
When I open log function (last two lines) in this function ```getCoords```, to call ```fillRect(0,0,1,64, color)```, ```- 0``` will generate following output:
```
Mapping to x: 0, y: 0 
Mapping to x: 0, y: 1 
Mapping to x: 0, y: 2 
Mapping to x: 0, y: 3 
Mapping to x: 0, y: 4 
Mapping to x: 0, y: 5 
Mapping to x: 0, y: 6 
Mapping to x: 0, y: 7 
Mapping to x: 0, y: 8 
Mapping to x: 0, y: 9 
Mapping to x: 0, y: 10
Mapping to x: 0, y: 11
Mapping to x: 0, y: 12
Mapping to x: 0, y: 13
Mapping to x: 0, y: 14
Mapping to x: 0, y: 15
Mapping to x: 0, y: 16
Mapping to x: 0, y: 17
Mapping to x: 0, y: 18
Mapping to x: 0, y: 19
Mapping to x: 0, y: 20
Mapping to x: 0, y: 21
Mapping to x: 0, y: 22
Mapping to x: 0, y: 23
Mapping to x: 0, y: 24
Mapping to x: 0, y: 25
Mapping to x: 0, y: 26
Mapping to x: 0, y: 27
Mapping to x: 0, y: 28
Mapping to x: 0, y: 29
Mapping to x: 0, y: 30
Mapping to x: 0, y: 31
Mapping to x: 256, y: 31
Mapping to x: 256, y: 30
Mapping to x: 256, y: 29
Mapping to x: 256, y: 28
Mapping to x: 256, y: 27
Mapping to x: 256, y: 26
Mapping to x: 256, y: 25
Mapping to x: 256, y: 24
Mapping to x: 256, y: 23
Mapping to x: 256, y: 22
Mapping to x: 256, y: 21
Mapping to x: 256, y: 20
Mapping to x: 256, y: 19
Mapping to x: 256, y: 18
Mapping to x: 256, y: 17
Mapping to x: 256, y: 16
Mapping to x: 256, y: 15
Mapping to x: 256, y: 14
Mapping to x: 256, y: 13
Mapping to x: 256, y: 12
Mapping to x: 256, y: 11
Mapping to x: 256, y: 10
Mapping to x: 256, y: 9
Mapping to x: 256, y: 8
Mapping to x: 256, y: 7
Mapping to x: 256, y: 6
Mapping to x: 256, y: 5
Mapping to x: 256, y: 4
Mapping to x: 256, y: 3
Mapping to x: 256, y: 2
Mapping to x: 256, y: 1
Mapping to x: 256, y: 0
``` 
If it begins with 0, the last should be 255, not 256. So I think ```- 0``` is not correct. If to use ```- 1``` as version 2.0.3, the output is correct: starts with 0, ends with 255. Maybe I am wrong about x indexing.

Now my workaround based on version 2.0.3 to prevent pixels dancing at the edge is to fill another black row. It's not smart, but it looks good to me.

mrfaptastic, 2021-06-11:
Are you testing the code that's in the git repository directly, NOT the releases?

Version 2.0.4, 2.0.3 etc. don't have any changes. You need to manually download the virtual display class from the git repository and replace in your local copy.

As I won't publish a new version of the library until you confirm the issue is fixed.

xuniuer, 2021-06-11:
Sorry that I don't fetch the latest copy manually before I see your reply. 

Now get the latest, comment out my workaround script, upload then run test. 

WOW: Dancing pixels at the edge of panel disappear, and ```fillRect``` function has also correct behavior.

Just as you said: **Should be fixed! YOU MADE IT!** 

Thank you very much!

mrfaptastic, 2021-06-11:
Excellent, glad to hear it worked. I have published the fix as part of a new release 2.0.5 just created.

Thanks for you help in debugging, and kind words.

closed 2021-06-11 by mrfaptastic