mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #130: problem with panel batch may 2021 PH3 64*64 V4.1 HX

donnersm, 2021-06-09:
I have a problem with a display that was delivered the other day ( ordered in may) The pcb layout is a it different. Same chipsets but the power connected is rotated 90 degrees, compared to the older panels. I have trouble getting this one going....see the video I send to the seller....
Anyone who has an idea what is going on there?

https://user-images.githubusercontent.com/60584210/121424145-50dbc680-c971-11eb-94e5-c651e70aa6ea.mp4

mrfaptastic, 2021-06-09:
Firstly, thank you for making such a great video! This video is the standard of how people should report issues!

Secondly, I think we've seen this before, and @vortigont fixed this as part of: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/75

You need to change the config to invert the clock signal. 

```mx_config.clkphase = true;```

donnersm, 2021-06-10:
Yep, That did the trick?
Thanks a lot!
Also Good to know, I used the newer panel as panel 1 and the older panel as 2.
so the The mx_config.clkphase = true; only applies to panel 1, however, panel 2 is working fine also! looks like the daisy chain of that panel is working like the older onces.

mrfaptastic, 2021-06-10:
Yeah, for some reason if the clock phase isn't inverted it breaks newer panels with some newer chip, but older panels don't seem to care.

I'm going to change the library default to an inverted clock as a result now.

donnersm, 2021-06-20:
https://user-images.githubusercontent.com/60584210/122685010-463eee00-d209-11eb-8125-46379118bb27.mp4

Looks like this issue is not solved completely, No idea what changed but inverting the clkphase doesn't work this time.
I played around with some setting but can't crack it:
```
mxconfig.clkphase = false; // some panels need a  reversed clockpulse I first encountered it in batch may 2021 PH3 64*64 V4.1 HX
mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
```
For those of you who cant open the video, here is an external link:
[http://www.judoles.nl/Projecten/fail.mp4](http://www.judoles.nl/Projecten/fail.mp4)

mrfaptastic, 2021-06-20:
Is this the same problem panel or another one? Can you make sure there's proper grounding between the power source powering the HUB75 and the ground on the signal pins. They need to both be connected together to a common ground.

What is the chipset? 

Try the legacy tree as well, but this has a different setup however (none of this mxconfig stuff): 
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/legacy-1.2.4

Although the legacy branch doesn't have the clock phase inversion, you'll have to hack the copy of the exp32_i2s_parallel.c and add 

```
  if (conf->clkphase)
    GPIO.func_out_sel_cfg[cfg->gpio_clk].inv_sel = 1;
```
... somewhere around line 200

donnersm, 2021-06-20:
No ground issues for sure.
Chipset: 7258
same panel as before.
Ill try the legacy and let you know

mrfaptastic, 2021-06-21:
Also, With the legacy branch you also need to clearscreen every time and repaint before the brightness change takes effect.

donnersm, 2021-06-21:
I made some progress,
I tried the legacy branch but didn't get it going at all and I didn't spend any time to figure out why.
Instead I concentrated on the new lib. Here is what i learned so far:
These new panels have a mix chipset.  Near the power connector there are ships: RUC7258 but at the bottom and top the chips are labelled: MBI5124

I am using the latest Lib version 2.05 which i confirmed in the compilation log
So I changed the config to: mxconfig.driver = HUB75_I2S_CFG::MBI5124;  
The clock phase has no effect as it is hard encoded in the library at line 29
This is getting close!! I can now adjust brightness of the lower part of the display seems like the upper part( I guess the 'E'-pin part..is not working correctly. it is not displaying the right data and it is gosthing when i increase brightness...

https://user-images.githubusercontent.com/60584210/122820652-23313e80-d2dc-11eb-959c-9f40e29f135b.mp4

mrfaptastic, 2021-06-21:
Wow… Now these china manufacturers can’t even be bothered to use the same chip across the same panel.

> This is getting close!! I can now adjust brightness of the lower part of the display seems like the upper part( I guess the 'E'-pin part..is not working correctly. it is not displaying the right data and it is gosthing when i increase brightness...

Actually, the R1, G1 and B1 pins are used to paint the top half, and the R2, G2 and B2 paint the bottom half.

Therefore, can you please swap the pin mapping in your config for the R1/R2, G1/G2, B1/B2 pins around? Yes, this will show a screwed up image with the bottom half painting of the animation painting on the top... but if the top half (per your video) is still blurred then the issue *is* a different chip issue.

I'd try get a refund from the seller as well by the way.

mrfaptastic, 2021-06-21:
Also comment out line 29 and try with true/false in your config and see if that fixes it. Who knows.

donnersm, 2021-06-22:
`Also comment out line 29 and try with true/false in your config and see if that fixes it. Who knows.`
Doesn't improve...it really needs to be set to true or not even the bottom half will work

`Therefore, can you please swap the pin mapping in your config for the R1/R2, G1/G2, B1/B2 pins around? Yes, this will show a screwed up image with the bottom half painting of the animation painting on the top... but if the top half (per your video) is still blurred then the issue is a different chip issue.`
Doing this will swop the bottom half and top half but the issue with the top half remains the same.
I see a lot of improvement when i add a pull up resistor (10K to 3.3v) to the lat pin
I will use an oscilloscope later this week to see what is happening on these pins.
maybe this is a version that really needs that 5V Bus, I will measure and try some level shifters  and report here

donnersm, 2021-06-23:
A little progress update.
Turns out that it is not the pull up resistor that makes a difference but more the long wire I used....This puts me in the capacitance domain. I used the oscilloscope to look at the data bus and all looks fine, no jitter, no explainable voltage levels... But whenever i connect my oscil. probe to the lat pin...the problem is gone....same if i attach a long wire...so...in the end, putting a 100pf cap on the lat pin to gnd solves all...However, I can't help the feeling that this is a workaround. My guess is that this display version has a problem with the bus frequency..... I tried the i2sspeed at 10Mhz and 20Mhz...makes no difference..is there any way to go lower?

I still have to try using the level shifters as that might make a difference too.  In the end we will have a completed topic with hopefully a conclusion and fix or workaround

mrfaptastic, 2021-06-23:
>
> Great work. Physics and parasitic capacitance strike again.

Whilst the DMA can be made to run at lower speeds, flicker starts to be
visible with the naked eye. Hence why the min frequency of 10mhz. I'd fix
the electrical issue rather than degrade image quality.

>

mrfaptastic, 2021-06-23:
In saying that, if you really want to reduce the DMA frequency then you'll have to hack the esp32_i2s_parallel_v2.c file.

Change line 228 from
```
dev->clkm_conf.clkm_div_num = clk_div_main; 
```
To

```
dev->clkm_conf.clkm_div_num = clk_div_main*2; 
```

To halve speeds again.

mrfaptastic, 2021-06-23:
Also, can you try the shorter cable from the esp32 to the first display? Perhaps see if you can get some other better cables as well.

I have had noise issues from using those long 16 pin cables... The price of convenience.

vortigont, 2021-06-24:
> so...in the end, putting a 100pf cap on the lat pin to gnd solves all...However, I can't help the feeling that this is a workaround.

seems to me that the issue is not a speed but bad timing. Putting a cap to LAT you make it's transient more continuous and introduce a delay between LAT transition an OE/addr lines. Need to check timing diagrams for those chips (yeah, mixing different ones on the same panel is a nice way to screw things).
I believe that shifting clk phase to +-90 degrees could fix that, but it is not possible with current i2s design. Reducing clk speed won't help here either. If you are not happy with cap you may try to put a series of 2 or 4 level shifters to the Clk or LAT line. It won't change signal phase but will introduce a propagation delay to the line that might help

mrfaptastic, 2021-06-24:
Personally I think @donnersm should raise a dispute with the seller and get his money back. 

Seems like a garbage panel to me, especially given it has different chipsets for each half.

donnersm, 2021-06-24:
> 
> 
> Personally I think @donnersm should raise a dispute with the seller and get his money back.
> 
> Seems like a garbage panel to me, especially given it has different chipsets for each half.

actually....they say its normal.....( is disagree... but never the less..this is what they say) see foto
[image]

mrfaptastic, 2021-06-24:
That doesn't even make any sense. Just raise a dispute anyway.

mrfaptastic, 2021-06-24:
That doesn't even make any sense because clearly they don't understand that the top and bottom halves of the panel are different shift register chips right (at least that's what you said). Videos prove this in any case.

You won't get anything other than noise from the chinese autotranslation. I doubt they understood your videos as well given the were in English. Just raise a dispute anyway.

DarrylStrong, 2021-06-24:
This does make sense, what they are saying is that the MBI5124 is the low side shift register and the 7258 is the high side row driver.

mrfaptastic, 2021-06-24:
It's irrelevant to what @donnersm stated:
>These new panels have a mix chipset. Near the power connector there are chips: RUC7258 but at the bottom and top the chips are labelled: MBI5124

mrfaptastic, 2021-06-24:
@donnersm 

I've randomly hacked the library in a different tree - no idea what I'm doing, but see if this makes any difference. Use this:

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/donnersm

donnersm, 2021-06-24:
Ok I dit raise a dispute but this topic is not about that...
I tried the hack with factor 2, 4 and 16... at 4 it is already so much flickering that it becomes irritating. More important the hack doesn't solve the issue. Even at lower speed, the display is still messed up. 
In conclusion, I wasn't able to get it to work by changing the software. I got it working by using a very short flatcable and the 100pf cap on the LAT pin to ground.  When I swop the left and right panels, there shouldn't be any difference but, to my surprise, there is!  This confirms that these displays ( this batch with mixed up drivers) is crap. In the end, I got a working setup using the workaround. With the hardware finished, I can finally complete my firmware.
Thanks all!
I shared my progress for anyone who runs into simular problems. Hope this topic can help them.
Here is a short preview of what i am working on:
https://www.youtube.com/watch?v=6ib_ivA0M0Q

mrfaptastic, 2021-06-24:
Thanks for the update. Great work with the equaliser. I'll reference this issue for further people who might recieve the same batch of crap panels.

closed 2021-06-24 by mrfaptastic