mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #131: compile error on changing color depth

tsctrl, 2021-06-10:
Hi @mrfaptastic,

i just try to update the library to the recent version today. when i change the PIXEL_COLOR_DEPTH_BITS to less than 8 the codes cant compile and throw error as below:

```
ESP32-HUB75-MatrixPanel-I2S-DMA.cpp: In member function 'void MatrixPanel_I2S_DMA::updateMatrixDMABuffer(uint8_t, uint8_t, uint8_t)': error: suggest parentheses around '+' inside '<<' [-Werror=parentheses]
 #define COLOR_DEPTH_LESS_THAN_8BIT_ADJUST  +(8-PIXEL_COLOR_DEPTH_BITS)
```

thanks,
tsctrl

mrfaptastic, 2021-06-10:
What environment are you using to compile etc.?

mrfaptastic, 2021-06-10:
Anyway, try again with the version of 'ESP32-HUB75-MatrixPanel-I2S-DMA.cpp' I have just pushed to the git repository.

tsctrl, 2021-06-10:
thank @mrfaptastic thats quite alot of changes i definitely cant solve on my own.

i am building with idf v4, and eclipse with plugin. not sure the C++ version and how to disable those warnings. by right, v4 is latest idf codes as not using arduino v3 or pio platform. this should confirm to be forward compatible with esp library for arduino upgrade in the future.

the fixes works and i am able to boot esp without kaboom or guru as per 8 bits before
thank you so much for the support. 

much love!

closed 2021-06-10 by tsctrl

mrfaptastic, 2021-06-10:
Thanks for your kind words. The code I had there was a hack and was likely to break at some point... which it did. Glad it's fixed now.