mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #132: ghosting on 3x2, off top and bottom for 1px

tsctrl, 2021-06-11:
i am getting pixel off for every character display top and bottom by one pixel. this issue actually happens since the old library i use late 2020 but that time my device are still in breadboard so cant confirm if it's from my jumper connections or from the library. 

now i got my pcb assembled, properly connected and confirm that this should be some algorithm change needed in the library to fit the display that i use.
i really hope if could get some code or area/ variable to play with in the codes to test or replace in which area where i can test to fix this issue.

[image]

thanks @mrfaptastic

tsctrl, 2021-06-11:
i also able to replicate the issue but less noticeable using the ChainedPanels  example as below:
[image]

mrfaptastic, 2021-06-11:
What happens with version 2.0.2 or whatever earlier versions on the library?

mrfaptastic, 2021-06-11:
Reduce the brightness as well.

tsctrl, 2021-06-11:
i remember that it i tested the library earlier than the time vortigont add major changes on the dev branch is also the same where before version 2. let me try on the version 1.2.3

tsctrl, 2021-06-11:
lib version:
"version": "1.0.2",
"frameworks": "arduino",
 "platforms": "esp32"

same result as below:
[image]

changing brightness didnt help

tsctrl, 2021-06-12:
another details to narrow down the issue: 

for the 3(cols)x2(row) setup, 1 row is 3 panels:

1. ghosting happens one pixel off on the top of character on the 1st row of 2 rows. 
2. ghosting happens one pixel off top and one pixel off bottom of character if the character are position in the middle of the screen which rendered in the 1st row and 2nd row. 
3. ghosting happens one pixel bottom of the character if position on the 2nd row of the panel.

settings:
	mxconfig.double_buff = true;
	mxconfig.latch_blanking = 1;
	mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
	mxconfig.clkphase = false;
	mxconfig.min_refresh_rate = 5;

change above configuration to other value: HZ_10M, clkphase false/true, higher refresh rate did not gets rid of the issue

thanks!

tsctrl, 2021-06-12:
basically, if the panels are on 6(col)x1(row) setup, ghosting happens one pixel bottom for all panels. unless its on 2 rows setup and the text are overlapping on both rows which ghosting happens one pixel top and one pixel bottom.

mrfaptastic, 2021-06-12:
I don't think this is a library issue. What is the grounding? Are you using virtual display class at all?

If not, check your grounding. This seems like an electrical issue to me I'm afraid.

mrfaptastic, 2021-06-12:
Set the latch blanking value to something higher like 10.

I might need to introduce a lower frequency... Perhaps there's an issue with these panels and they don't like 10mhz. What happens if you set to 20 MHz?

tsctrl, 2021-06-12:
nice, tested and it works! now clear and plotted correctly.. 
no wonder it's looks like the issue intermittently happens and not sure whats fix it! :D 

awesome @mrfaptastic ! thanks!

closed 2021-06-12 by tsctrl

2BECKHAM2, 2021-07-23:
> another details to narrow down the issue:
>
> for the 3(cols)x2(row) setup, 1 row is 3 panels:
>
> 1. ghosting happens one pixel off on the top of character on the 1st row of 2 rows.
> 2. ghosting happens [...]

@tsctrl 

I've faced the similar issue here. Tried changing the i2sspeed and double_buff values but looks like they don't exist in the `MatrixPanel_I2S_DMA` object. Went through the `ESP32-HUB75-MatrixPanel-I2S-DMA.h` file and there was no function defined to set these variables. Could you share your sample repo for reference? Apart from this, I am able to change the latch_blanking and brightness values. Thanks.