mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #133: 2.0.5 - 64x32 - lower left leds shifted

Daedelos, 2021-06-12:
With 2.0.5 on my 64x32 leds appear to be shifted in the lower left quarter of the panel. e.g.

display.writeFastVLine(17, 0, display.height(), display.color565(0, 0, 255));

results in a broken vertical line in the lower potion of my 64x32 panel.

2.0.3 works correctly with no broken line (shifted leds)

[image]

[image]

mrfaptastic, 2021-06-12:
Is this one panel or a chain of panels? I can't tell.

Are you using the virtual display class. If so, post the sketch code.

Finally, can you reverse what I did in #130 to see if this is a clockphase problem that I've introduced.

Daedelos, 2021-06-12:
Single panel.
Not using the virtual display class. 

I changed the clockphase to false and the line is drawn correctly.

mrfaptastic, 2021-06-12:
Good to know it's fixed. There are too many permutations of these panels now... can't seem to fix an issue without breaking another.

Can you tell me what the chipset your panels are?

Daedelos, 2021-06-12:
ICN2028AP / ICN2012

[image]

vortigont, 2021-06-12:
> There are too many permutations of these panels now... can't seem to fix an issue without breaking another.

Looks like we might need some kind of "config templates" or maybe a WiKi table with confirmed settings

mrfaptastic, 2021-06-12:
We could, but that's another thing we'd have to keep up to date constantly. I'd perfer users of the library to experiment a bit - what I will do is raise an issue of 'common issue' and pin it to the top - give people guidance.

closed 2021-06-12 by mrfaptastic