mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #134: Common Issues - Read This First

mrfaptastic, 2021-06-12:
There are many permutations of these HUB75 LED panels such that the 'default' configuration of the library is likely to not work in some cases. Below are some of the reported issues, and solutions. If you issue is not one of the below, **raise a seperate new issue**

### Issue 1: GHOSTING [#132]
[image]
Solution: Increase the latch_blanking (i.e. the time between clocking data to the panel and then turning the LEDS 'on'. Some panels can't handle the _speed_ of this library.
```
mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;
```

### Issue 2: Vertical output isn't aligned in the two panels of a single 1/16 scan panel. [#133]
[image]
Solution:
```
mx_config.clkphase = false;
```

### Issue 3: Blurred output [#130] 
https://user-images.githubusercontent.com/60584210/121424145-50dbc680-c971-11eb-94e5-c651e70aa6ea.mp4

Solution: Invert the clock signal in your setup configuration. i.e.
```mx_config.clkphase = true;
```
### Issue 4: Flickering or other garbage output [#39] 
[image]

Solution: 
- Get a better power supply that can keep ~5volt output.  
- Make sure you have capacitors parallel to the HUB75 power input to handle spikes.

witnessmenow, 2021-06-22:
Hey

I had a similar issue as "Issue 2" but with a 64x64 (so I presume a 1/32 scan), but it seemed to be impacting the entire display and was shifting what should have been at x=0 to x=63 

[image]

setting `mxconfig.clkphase = false;` worked though:

[image]

marcomilazzo, 2023-01-05:
[image]
HI
i changhed mx_config.clkphase = true;  and i get this . with false i don't get anyrhing
any suggestions?
thank's 
marco

mrfaptastic, 2023-01-05:
You're panel must be some weird scan type. Try using the 1/8 scan example?

The clkphase setting is a physical electrical configuration thing. So the fact you now get an output = correct value.