mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #145: Flickering Issue with "ESP32-HUB75-MatrixPanel-I2S-DMA" only

subashsaraf, 2021-07-05:
I made a clock with this Library actually the code was from Great Project RGB clock I modify the code according to your Lib. Hence I face Flickering issue . Where the code is absolutely fine PxMatrix lib but when I use "ESP32-HUB75-MatrixPanel-I2S-DMA" then the display is become Flickering .So is there any other way to solve the problem

Link To See the Flickering Issue

https://youtu.be/rzNrKGc8oXI

Code :-

Esp32_DS1307_DHT11_Matrix_p4_RGB_64x32_DMA_AMPM_dma_display.zip
[Esp32_DS1307_DHT11_Matrix_p4_RGB_64x32_DMA_AMPM_dma_display.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/6763779/Esp32_DS1307_DHT11_Matrix_p4_RGB_64x32_DMA_AMPM_dma_display.zip)

mrfaptastic, 2021-07-05:
Turn off your LDR checking code and see if that fixes the issues. I think you have conflicting pin mappings causing noise on the DMA output.

subashsaraf, 2021-07-06:
Thanks for reply,

        I  check everything is correct. I attached the full scratch for
your knowledge. I attached the LDR pin to 34 the pin in mint for input only
and all pins are mapped according to the DMA library. So kindly find the
attached file for the

same

Subash Saraf
M: +91 9437357170
Email : ***@***.***

mrfaptastic, 2021-07-06:
What does SimpleTestShapes example look like?

subashsaraf, 2021-07-06:
Kindly Scc the video

Subash Saraf
Manager (WRM E&A)
M: +91 9437357170
Email : ***@***.***
             ***@***.***
BHUSHAN POWER & STEEL LIMITED
Vill: Thelkuli | P.O.: Lapanga | Tehshil : Rengali
Distt.: Sambalpur – 768232 | State : Orissa (INDIA)

mrfaptastic, 2021-07-06:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/1_SimpleTestShapes

?

mrfaptastic, 2021-07-06:
Try: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/1_SimpleTestShapes

Take a video of this sketch

subashsaraf, 2021-07-06:
Sir,
I already checked this this work fine. But in case  scrolling text the
flickering happen. This code is absolutely fine in pxmatrix

Subash Saraf
Manager (WRM E&A)
M: +91 9437357170
Email : ***@***.***
             ***@***.***
BHUSHAN POWER & STEEL LIMITED
Vill: Thelkuli | P.O.: Lapanga | Tehshil : Rengali
Distt.: Sambalpur – 768232 | State : Orissa (INDIA)

2BECKHAM2, 2021-07-22:
@mrfaptastic.

Many thanks for the excellent library.

I'm facing some noise/flickering issues with the 64x32 panel with 1.0.6 esp32 version. I have connected a 3300uf capacitor across  5v in the panel.

https://drive.google.com/drive/folders/1c6tXlEqq_mQJvfQCB-mob6V1B1d418zN?usp=sharing

[image]

[image]

Is there any config changes that I would have to make.

Thanks.

mrfaptastic, 2021-07-22:
Can you check that non-GND pins are not going to ground.

Can you draw a single diagonal line as well? 
Looks like some address line issue, check the A,B,C,D pin mapping - perhaps the pins may have moved? Who knows. There seems to be a few of these issues being raised recently.

[image]

2BECKHAM2, 2021-07-22:
Thanks for the quick response.

Can you please share me the pin mapping details for ESP 32.

I don't have any clue on DR1, DB1, DR2, DB2, A, C OE, DG1DG2 B, D  and LAT
at the ESP 32 Side.

Sorry to bother you.

Thanks in advance.

mrfaptastic, 2021-07-22:
How have you connected the ESP32 to the LED panel? With some custom PCB or pin-by-pin with jumper cable.

You need to check your wiring.

2BECKHAM2, 2021-07-22:
I have not used any custom pcb. Connected one by one as mentioned by you in
the GitHub page link.

mrfaptastic, 2021-07-22:
Looking at your photo I see a 16 wire ribbon cable connecting to the panel. So how do you do that without a pcb?

There's something wrong be with your wiring. You need to keep the wire length between the esp32 and panel as short as possible, like 4-5 inches max, or these issues will occur.

This is an electrical issue I think.

2BECKHAM2, 2021-07-22:
From the ribbon cable, I have used male to female jumper wire ( colored one which normally use for the esp project) female end will be connected to esp 32.

Find attached screenshot for reference.

[image]

[image]

Thanks in advance.

mrfaptastic, 2021-07-22:
That's likely your problem then. Use shorter female to female jumper cables and connecting directly to the LED panel.

mrfaptastic, 2021-07-22:
Also ensure ground pins on LED panel are grounded.

2BECKHAM2, 2021-07-23:
Thanks for your help. Will check as suggested and up date.

2BECKHAM2, 2021-07-23:
@mrfaptastic ,

I  have connected the shorter cables and observed the less flickering and noise. Find below for your reference.

[https://drive.google.com/file/d/1nw4ysHyfHBCSnwHW25mCvjpaQzlTfraj/view?usp=sharing](url)

[https://drive.google.com/file/d/1bRwCGyKM6wTf2oT20CQ6o0N3jbupVB9X/view?usp=sharing](url)

[image]

Also regarding the latch blanking configuration, where exactly should I be making the necessary code changes?

Thanks.

mrfaptastic, 2021-07-23:
In the mxconfig that you pass to the library. Refer to: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/134

2BECKHAM2, 2021-07-24:
@mrfaptastic 

Can you please help me in which file the below configuration needs to be changed.

mxconfig.latch_blanking = 4;
mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_10M;

//and reduce brightness as well

// let's adjust default brightness to about 50%
dma_display->setBrightness8(128);    // range is 0-255, 0 - 0%, 255 - 100%

mx_config.clkphase = false;

Thanks in advance.

mrfaptastic, 2021-07-24:
Please look at the source code of the other examples.

2BECKHAM2, 2021-07-24:
Its working like a charm after doing the config changes. - Thanks a lot for your help.

[image]
[image]
[image]
[image]
[image]

subashsaraf, 2021-07-24:
> 
> 
> Its working like a charm after doing the config changes. - Thanks a lot for your help.
> 
> [image]
> [image]
> [image]
> [image]
> [image]

Sir,
     Can you please post the the code link which you is  working fine without any issue  .So that I will correction my code according to your program 

Subashsaraf@gmail.com

subashsaraf, 2021-07-24:
> 
> 
> > Its working like a charm after doing the config changes. - Thanks a lot for your help.
> > [image]
> > [image]
> > [image]
> > [image]
> > [image]
> 
> Sir,
> Can you please post the the code link which you is working fine without any issue .So that I will correction my code according to your program
> 
> [Subashsaraf@gmail.com](mailto:Subashsaraf@gmail.com)

Sir,
     Can you please post the the code link which you is  working fine without any issue  .So that I will correction my code according to your program 

Subashsaraf@gmail.com

2BECKHAM2, 2021-07-25:
@subashsaraf 

Please check the example code at https://github.com/2BECKHAM2/ESP32-HUB75-MatrixPanel-I2S-DMA-Examples

mrfaptastic, 2021-07-26:
Excellent work @2BECKHAM2 - Would you mind raising a pull request to this repository to update the SimpleTextShapes example to use the new pointer based usage of this library and mxconfig ?

2BECKHAM2, 2021-07-26:
Sure. Will do.

Thanks a lot.

closed 2021-07-27 by mrfaptastic