mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #146: Great job to you guys

Dcmwrnd, 2021-07-07:
This is a fantastic library. I’m using 12 pcs P4 panels (64 x 32 pixels). Did not expect this ESP32 is so powerful.
Thank you to all you guys. Below here are some video links using VirtualMatrixPanel library.

http://www.dcmw.com.my/video/ESP32-Hub75.mp4
http://www.dcmw.com.my/video/ESP32-Spectrum-12x64x32.mp4
http://www.dcmw.com.my/video/LedPanels.mp4

mrfaptastic, 2021-07-07:
Thank you! I especially like 'ESP32-Hub75.mp4' - great selection of hot women. Probably the best use of this library so far.

You're actually driving **12** panels? That's amazing.

Dcmwrnd, 2021-07-07:
Yes, I’m using 12 panels. Total pixels is 256 x 128. All the animated gif files are 256 x 128 and are stored in SD card because spiffs memory got not enough space. If I’m not mistaken I think I tested up to 14 panels & that’s the max I can go. 
[image]
Try to go 15 panels the esp crash, saying memory not enough or something like that.

[image]

Dcmwrnd, 2021-07-07:
Sorry, quoted wrongly… total pixels is 256 x 96.

closed 2021-07-22 by mrfaptastic