mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #151: panel 16*32 and example P6_32x16_1_4_ScanPanel.ino

donnersm, 2021-07-16:
Just wondering..can this QuarterScanMatrixPanel be configured to be used with pointers.
I defined my whole sketch with matric pointer.. but in the example it shows a direct approach.
My definition:

```
// placeholder for the matrix object
MatrixPanel_I2S_DMA *dma_display = nullptr;

// object
dma_display = new MatrixPanel_I2S_DMA(mxconfig);

```
Now i can code the output to the display like this:
dma_display->drawPixelRGB888(x,y,0,0,0);

however, with the Quarterscanmatrixpanel class i cannot define it with pointer like i did, or can I?

mrfaptastic, 2021-07-16:
You'll need to refactor the example as it was written before the library implemented a pointer based approach as well. Feel free to raise a pull request.

Should be as simple as replacing the .'s with ->'s in most cases.

donnersm, 2021-07-18:
replacing the .'s with ->'s in most cases.  that part i understand. However, those are a lot of changes to my code for only one type of panel...to keep it universal for all panels....i would like to keep using that pointer. That way all i need to add is something in the trend of 
```
#ifdef 32x16 
// setup QuarterScanMatrixPanel
..
..
#endif
```
Maybe there is someone who can help me with rewriting that QuarterScanMatrixPanel function?
How about you leave this topic open for a few days to see if anyone offers some help and if not...close it as "not an issue"

vortigont, 2021-07-19:
> who can help me with rewriting that QuarterScanMatrixPanel function?

No need, you can use it as-as.
SOmething like this:

```
MatrixPanel_I2S_DMA *dmaOutput = nullptr;
QuarterScanMatrixPanel * display = nullptr;
void setup() {
dmaOutput = new MatrixPanel_I2S_DMA();
display = new QuarterScanMatrixPanel(*dmaOutput);
}
void loop() {
  display->fillScreen(display->color444(0, 0, 0));
}
```

closed 2021-08-12 by mrfaptastic