mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #154: 64 x 32 panel horizontal lines doubled

donnersm, 2021-07-21:
I just tested a panel i just received today. Its a 64 x 32 panel.
It looks like the horizontal is not working, lines are doubled in an array of 8
Anyone recognise this or is this a defective panel?
(Beside from the defective pixel near one corner)

https://user-images.githubusercontent.com/60584210/126544627-5c615cfe-e9d1-433f-962e-325da9e36a2a.mp4

mrfaptastic, 2021-07-22:
Can you draw a single diagonal line as well? 

Looks like some address line issue, check the A,B,C,D pin mapping - perhaps the pins may have moved? Who knows. There seems to be a few of these issues being raised recently.

donnersm, 2021-07-25:
I Checked the pins on the display and all is "standard" except the LAT pin is called STB. 
I tried a vertical line and a circle:
```
dma_display->clearScreen();
dma_display->drawLine(dma_display->width()-1, 0, 0, dma_display->height()-1, dma_display->color444(15, 0, 0));
dma_display->clearScreen();
delay(500);
dma_display->fillCircle(40, 21, 10, dma_display->color444(15, 0, 15));
```
Here are the photo's of that result:
[image]
[image]

mrfaptastic, 2021-08-12:
Closing this issue as it's superseded by #164

AlfioSaitta, 2021-10-05:
> I Checked the pins on the display and all is "standard" except the LAT pin is called STB. I tried a vertical line and a circle:
>
> ```
> dma_display->clearScreen(); [...]

I have the exact same problem. Have you finally solved it?

DarrylStrong, 2021-10-06:
Do you have a picture of the back of the PCB? Connectors and driver ICs?

donnersm, 2021-10-06:
I didnt solve it.  It turned out to be a 1/8 scan panel which is not supported by the library. We should be able to change the libary because it is al telated to the way the data is sorted bitwise before sending it to the panel but my knowledge of this lib is limited

AlfioSaitta, 2021-10-09:
> Do you have a picture of the back of the PCB? Connectors and driver ICs?

- The ICs are some ICN2037BP
- The connections are direct from my AZ-Delivery DevKitC V4 to the HUB75 connector on the panel

My simple code with the git version of the library:

```
#define MATRIX_WIDTH 64
#define MATRIX_HEIGHT 32
#define CHAIN_LENGTH 1
#define MATRIX_ROWS_IN_PARALLEL 2
#define PIXEL_COLOR_DEPTH_BITS 8

#define R1_PIN  2
#define G1_PIN  15
#define B1_PIN  4
#define R2_PIN  16
#define G2_PIN  27
#define B2_PIN  17

#define A_PIN   5
#define B_PIN   18
#define C_PIN   19
#define D_PIN   -1 // Connected to GND on panel (21 if exist)
#define E_PIN   -1 // Connected to GND on panel

#define LAT_PIN 26
#define OE_PIN  25
#define CLK_PIN 22

#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup() {

  Serial.begin(115200);

  delay(1000);

  HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
    A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
    LAT_PIN, OE_PIN, CLK_PIN
  };

  // Module configuration
  HUB75_I2S_CFG mxconfig(
      MATRIX_WIDTH, // module width
      MATRIX_HEIGHT, // module height
      CHAIN_LENGTH, // Chain length
      _pins);

  //mxconfig.driver = HUB75_I2S_CFG::SHIFTREG;
  //mxconfig.double_buff = true;
  //mxconfig.clkphase = false;
  //mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

  dma_display->begin();
  dma_display->setBrightness8(100); //0-255
  dma_display->clearScreen();

  // First Half Display
  dma_display->drawLine(0, 0, 15, 0, dma_display->color444(0, 15, 0));
  dma_display->drawLine(0, 1, 31, 1, dma_display->color444(15, 0, 0));
  dma_display->drawLine(0, 2, 63, 2, dma_display->color444(0, 0, 15));

  // Second Half Display
  dma_display->drawLine(0, 16, 15, 16, dma_display->color444(0, 15, 15));
  dma_display->drawLine(0, 17, 31, 17, dma_display->color444(15, 15, 0));
  dma_display->drawLine(0, 18, 63, 18, dma_display->color444(15, 0, 15));
}

void loop() {
    //delay(20); 
}
```

[image]

[image]

[image]

[image]

mrfaptastic, 2021-10-09:
Looks like a pretty simple issue to fix with your own custom code of 2 X 1/4th of the panel rows chained together.

Sweet the width of the display of the library to double the physical width and play around?

AlfioSaitta, 2021-10-09:
> Looks like a pretty simple issue to fix with your own custom code, looks like 2 X 1/4th of the panel rows chained together, for both the top and bottom halves of the display.
> 
> Change the width of the display of the library to double the physical width and play around?

Eureka... very strange settings... but work :D

[image]

mrfaptastic, 2021-10-10:
Good work on figuring that out.  As expected, it's actually a chain of panels, with some weird chaining going on there between the 1/4ths of the panel, with the 2nd 1/4th being chained to the first (top) 1/4th of the panel.

What happens when you set:
```HUB75_I2S_CFG mxconfig( 64, 32, CHAIN_LENGTH, _pins);```
to 
```HUB75_I2S_CFG mxconfig( 64, 16, CHAIN_LENGTH, _pins);```

Because if you do the math, 8 pixels of each 'half' of the panel don't actually get used given this library is designed for 1/2 scan panels. If you get what I mean?

mrfaptastic, 2021-10-10:
All you need to do is make your own wrapper to this library, like the 'Virtual Display' cpp file, which will override the operation of the drawPixel() function - then you have youself a working library for these 1/4 scan panels.

Feel free to contribute then :-)

AlfioSaitta, 2021-10-10:
> Good work on figuring that out. As expected, it's actually a chain of panels, with some weird chaining going on there between the 1/4ths of the panel, with the 2nd 1/4th being chained to the first (top) 1/4th of the panel.
> 
> What happens when you set: `HUB75_I2S_CFG mxconfig( 64, 32, CHAIN_LENGTH, _pins);` to `HUB75_I2S_CFG mxconfig( 64, 16, CHAIN_LENGTH, _pins);`
> 
> Because if you do the math, 8 pixels of each 'half' of the panel don't actually get used given this library is designed for 1/2 scan panels, and these panels actually chain the 2nd 1/4th with the 1st 1/4.

Result with your settings and corrected the math a bit:

[image]

```
#define CHAIN_LENGTH 2 // 2 Chained panels
#define PANEL_WIDHT 64
#define PANEL_HEIGHT 16 // 1/8 Scan in the specifics ???

#define R1_PIN  2
#define G1_PIN  15
#define B1_PIN  4
#define R2_PIN  16
#define G2_PIN  27
#define B2_PIN  17

#define A_PIN   5
#define B_PIN   18
#define C_PIN   19
#define D_PIN   -1 // Connected to GND on panel (21 if exist)
#define E_PIN   -1 // Connected to GND on panel

#define LAT_PIN 26
#define OE_PIN  25
#define CLK_PIN 22

#include <Arduino.h>
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

MatrixPanel_I2S_DMA *dma_display = nullptr;

void testScreen() {
  
  for (int x=0; x <= 63; x++) {

    // Red 8 Rows
    dma_display->drawPixel(64+(x),0,dma_display->color565(255, 0, 0)); // ROW 1
    dma_display->drawPixel(64+(x),1,dma_display->color565(210, 0, 0)); // ROW 2
    dma_display->drawPixel(64+(x),2,dma_display->color565(180, 0, 0)); // ROW 3
    dma_display->drawPixel(64+(x),3,dma_display->color565(150, 0, 0)); // ROW 4
    dma_display->drawPixel(64+(x),4,dma_display->color565(125, 0, 0)); // ROW 5
    dma_display->drawPixel(64+(x),5,dma_display->color565(90, 0, 0)); // ROW 6
    dma_display->drawPixel(64+(x),6,dma_display->color565(60, 0, 0)); // ROW 7
    dma_display->drawPixel(64+(x),7,dma_display->color565(40, 0, 0)); // ROW 8

    // Blue 8 Rows
    dma_display->drawPixel(0+x,0,dma_display->color565(0, 0, 255)); // ROW 9
    dma_display->drawPixel(0+x,1,dma_display->color565(0, 0, 210)); // ROW 10
    dma_display->drawPixel(0+x,2,dma_display->color565(0, 0, 180)); // ROW 11
    dma_display->drawPixel(0+x,3,dma_display->color565(0, 0, 150)); // ROW 12
    dma_display->drawPixel(0+x,4,dma_display->color565(0, 0, 125)); // ROW 13
    dma_display->drawPixel(0+x,5,dma_display->color565(0, 0, 90)); // ROW 14
    dma_display->drawPixel(0+x,6,dma_display->color565(0, 0, 60)); // ROW 15
    dma_display->drawPixel(0+x,7,dma_display->color565(0, 0, 40)); // ROW 16

    // White 8 Rows
    dma_display->drawPixel(64+x,8,dma_display->color565(255, 255, 255)); // ROW 17
    dma_display->drawPixel(64+x,9,dma_display->color565(210, 210, 210)); // ROW 18
    dma_display->drawPixel(64+x,10,dma_display->color565(180, 180, 180)); // ROW 19
    dma_display->drawPixel(64+x,11,dma_display->color565(150, 150, 150)); // ROW 20
    dma_display->drawPixel(64+x,12,dma_display->color565(125, 125, 125)); // ROW 21
    dma_display->drawPixel(64+x,13,dma_display->color565(90, 90, 90)); // ROW 22
    dma_display->drawPixel(64+x,14,dma_display->color565(60, 60, 60)); // ROW 23
    dma_display->drawPixel(64+x,15,dma_display->color565(40, 40, 40)); // ROW 24

    // Green 8 Rows
    dma_display->drawPixel(0+x,8,dma_display->color565(0, 255, 0)); // ROW 25
    dma_display->drawPixel(0+x,9,dma_display->color565(0, 210, 0)); // ROW 26
    dma_display->drawPixel(0+x,10,dma_display->color565(0, 180, 0)); // ROW 27
    dma_display->drawPixel(0+x,11,dma_display->color565(0, 150, 0)); // ROW 28
    dma_display->drawPixel(0+x,12,dma_display->color565(0, 125, 0)); // ROW 29
    dma_display->drawPixel(0+x,13,dma_display->color565(0, 90, 0)); // ROW 30
    dma_display->drawPixel(0+x,14,dma_display->color565(0, 60, 0)); // ROW 31
    dma_display->drawPixel(0+x,15,dma_display->color565(0, 40, 0)); // ROW 32
  }
}

void setup() {

  Serial.begin(115200);
  delay(1000);

  HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
    A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
    LAT_PIN, OE_PIN, CLK_PIN
  };

  HUB75_I2S_CFG mxconfig( PANEL_WIDHT, PANEL_HEIGHT, CHAIN_LENGTH, _pins);

  mxconfig.double_buff = true; // ????
  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(200); //0-255
  dma_display->clearScreen();
}

void loop() {
    //delay(40); 
    testScreen();
}
```
After researching in various Chinese retailers and factories, i found some more information on the panel:

```
Pixel Pitch: P4
Dimension: 128X256
Lamp bead: RGB
LED Type: SMD1921
Scanning Mode: 1/8
```

mrfaptastic, 2021-10-10:
Hi @AlfioSaitta - thanks again for the great work with trying this out, really helps me understand what's going on here!

I did notice one thing with you photo, what's the go on the top left? Looks to be a row of missing pixels? Or is the pixels broken in that area?

[image]

AlfioSaitta, 2021-10-10:
> Hi @AlfioSaitta - thanks again for the great work with trying this out, really helps me understand what's going on here!
> 
> I did notice one thing with you photo, what's the go on the top left? Looks to be a row of missing pixels? Or is the pixels broken in that area?
> 

I think it's a power problem. I'm using the raspberry power supply to power the panel :dancers: 

https://user-images.githubusercontent.com/10280718/136692822-862c8240-afb5-4614-b84d-488f79f492cf.mp4

mrfaptastic, 2021-10-10:
Good work. I've created a 'VirtualDisplay' like class that does co-ordinate re-mapping, like it seems you have coded manually.

I don't have one of these 1/8 scan panels - can you try this out?

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/One_Eighth_1_8_ScanPanel

AlfioSaitta, 2021-10-10:
> Good work. I've created a 'VirtualDisplay' like class that does co-ordinate re-mapping, like it seems you have coded manually.
>
> I don't have one of these 1/8 scan panels - can you try this out? [...]

Yes, of course

AlfioSaitta, 2021-10-10:
Generate error here: rows and cols are not defined
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/ad2cb3c9767125e682fa4a494f47e357bbb0fb3f/examples/One_Eighth_1_8_ScanPanel/OneEighthScanMatrixPanel.h#L175

mrfaptastic, 2021-10-10:
Whoops. I've fixed that.

AlfioSaitta, 2021-10-10:
Some tests with lines:

```
  virtualDisp->drawLine(0, 0, 63, 0, virtualDisp->color565(255,255,255)); // VISIBLE ON ROW 1
  virtualDisp->drawLine(0, 8, 63, 8, virtualDisp->color565(255,255,0)); // VISIBLE ON ROW 13 (WITH LAST PIXEL RED)
  virtualDisp->drawLine(0, 9, 63, 9, virtualDisp->color565(255,0,0)); // VISIBLE ON ROW 13 (COVERS THE PREVIOUS ROW)
  virtualDisp->drawLine(0, 17, 63, 17, virtualDisp->color565(255,0,255)); // VISIBLE ON ROW 17
  virtualDisp->drawLine(0, 31, 63, 31, virtualDisp->color565(0,255,0)); // VISIBLE ON ROW 32
```

mrfaptastic, 2021-10-10:
Sorry. I figured out the problem and my last commit in my previous comment was wrong.

Hopefully it's fixed now !

AlfioSaitta, 2021-10-10:
With your latest changes:

[image]

```
#define PANEL_WIDHT 64
#define PANEL_HEIGHT 32 // 1/8 Scan in the specifics ???

#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW

#define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
  
// Change this to your needs, for details on VirtualPanel pls read the PDF!
#define SERPENT true
#define TOPDOWN false

#define R1_PIN  2
#define G1_PIN  15
#define B1_PIN  4
#define R2_PIN  16
#define G2_PIN  27
#define B2_PIN  17

#define A_PIN   5
#define B_PIN   18
#define C_PIN   19
#define D_PIN   -1 // Connected to GND on panel (21 if exist)
#define E_PIN   -1 // Connected to GND on panel

#define LAT_PIN 26
#define OE_PIN  25
#define CLK_PIN 22

#include <Arduino.h>
//#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
#include <OneEighthScanMatrixPanel.h>

MatrixPanel_I2S_DMA *dma_display = nullptr;
OneEighthMatrixPanel *virtualDisp = nullptr;

void drawPicture() {

  int counter = 0;

  for (int y = 0; y <= 7; y++){
    for (int x=0; x <= 63; x++) {
      dma_display->drawPixel(64+x , y, mycar[counter]);
      counter++;
    }
  }

  for (int y = 0; y <= 7; y++){
    for (int x=0; x <= 63; x++) {
      dma_display->drawPixel(0+x , y, mycar[counter]);
      counter++;
    }
  }

  for (int y = 8; y <= 15; y++){
    for (int x=0; x <= 63; x++) {
      dma_display->drawPixel(64+x , y, mycar[counter]);
      counter++;
    }
  }

  for (int y = 8; y <= 15; y++){
    for (int x=0; x <= 63; x++) {
      dma_display->drawPixel(0+x , y, mycar[counter]);
      counter++;
    }
  }
}

void setup() {

  Serial.begin(115200);
  delay(100);

  HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
    A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
    LAT_PIN, OE_PIN, CLK_PIN
  };

  HUB75_I2S_CFG mxconfig( PANEL_WIDHT*2, PANEL_HEIGHT/2, PANEL_CHAIN, _pins);

  mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(30); //0-255
  dma_display->clearScreen();

  virtualDisp = new OneEighthMatrixPanel((*dma_display), 1, 1, PANEL_WIDHT, PANEL_HEIGHT, SERPENT, TOPDOWN);

  virtualDisp->drawLine(0, 0, 63, 0, virtualDisp->color565(255,255,255)); // VISIBLE ON ROW 1
  virtualDisp->drawLine(0, 8, 63, 8, virtualDisp->color565(255,255,0)); // VISIBLE ON ROW 13 (WITH LAST PIXEL RED)
  virtualDisp->drawLine(0, 9, 63, 9, virtualDisp->color565(255,0,0)); // VISIBLE ON ROW 13 (COVERS THE PREVIOUS ROW)
  virtualDisp->drawLine(0, 17, 63, 17, virtualDisp->color565(255,0,255)); // VISIBLE ON ROW 17
  virtualDisp->drawLine(0, 31, 63, 31, virtualDisp->color565(0,255,0)); // VISIBLE ON ROW 32
}

```

mrfaptastic, 2021-10-10:
Yeah, I figured out the problem and my last commit in my previous comment was wrong.

Hopefully it's fixed now !

mrfaptastic, 2021-10-10:
Try again now. 5th time lucky.

Wrote some code to emulate my logic: http://cpp.sh/6skpy

mrfaptastic, 2021-10-10:
So it's ok? Not sure why there are missing bits of the red box but that was your power supply right?

AlfioSaitta, 2021-10-10:
I'm doing various tests... but i think my panel doesn't like the color red without proper power supply, but the library is working perfectly. Great work ;)

https://user-images.githubusercontent.com/10280718/136715954-d978209b-1994-4407-b95c-2bae184db484.mp4

AlfioSaitta, 2021-10-10:
Tomorrow in the office i will check if it works well in chain with other panels, and especially with their power supplies :kissing_closed_eyes:

AlfioSaitta, 2021-10-11:
I can't draw starting after the first panel

```
#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 3 // Number of INDIVIDUAL PANELS per ROW

virtualDisp = new OneEighthMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_WIDHT, PANEL_HEIGHT, SERPENT, TOPDOWN);

void drawPicture(uint16_t Picture[2048], int StartX, int StartY) {
  int counter = 0;
  for (int y = StartY; y <= StartY+(PANEL_HEIGHT-1); y++){
    for (int x=StartX; x <= StartX+(PANEL_WIDHT-1); x++) {
      virtualDisp->drawPixel(x , y, Picture[counter]);
      counter++;
    }
  }
}

drawPicture(mario, 64, 0);
```
[image]

AlfioSaitta, 2021-10-11:
> No problems. What are you trying to make? I'll be interested to see what frame rate you can achieve eventually.

I want to divide the tasks to be performed on the two cores. 

- Define an array (or series of arrays based on the number of panels) shared between the two cores (task)
- On the first one draw the array of rgb values ​​on the panel (as in the previous drawPicture function)
- And on the second core have a udp connection which refreshes my array(s)

Actually i don't know the potential of the Esp, but i hope it behaves well :P

AlfioSaitta, 2021-10-11:
What frame rate did you get in your tests?

macduyhai, 2021-11-03:
> Result with your settings and corrected the math a bit:
>
> [image]
>
> ```
> #define CHAIN_LENGTH 2 // 2 Chained panels
> #define PANEL_WIDHT 64
> #define PANEL_HEIGHT 16 // 1/8 Scan in the specifics ??? [...]

This code on your brand ?? patch-1 or legacy-1.2.4 ?? thanks

macduyhai, 2021-11-03:
Thank you @mrfaptastic  but I try example "One_Eighth_1_8_ScanPanel" and change GPIO define 
Link git : https://github.com/AlfioSaitta/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/patch-1

`/*************************************************************************
 * Description: 
 * 
 * The underlying implementation of the ESP32-HUB75-MatrixPanel-I2S-DMA only
 * supports output to 1/16 or 1/32 scan panels - which means outputting 
 * two lines at the same time, 16 or 32 rows apart. This cannot be changed
 * at the DMA layer as it would require a messy and complex rebuild of the 
 * library's DMA internals.
 *
 * However, it is possible to connect 1/8 scan panels to this same library and
 * 'trick' the output to work correctly on these panels by way of adjusting the
 * pixel co-ordinates that are 'sent' to the ESP32-HUB75-MatrixPanel-I2S-DMA
 * library (in this example, it is the 'dmaOutput' class).
 * 
 * Supports chaining of multiple 1/8 panels...
 *
 **************************************************************************/
   
  // Panel configuration
  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
  
  #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
  #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
  
  // Change this to your needs, for details on VirtualPanel pls read the PDF!
  #define SERPENT true
  #define TOPDOWN false
  
  // GPIO Configuration
  #define R1_PIN  25
  #define G1_PIN  26
  #define B1_PIN  27
  #define R2_PIN  14
  #define G2_PIN  12
  #define B2_PIN  13
  
  #define A_PIN   23
  #define B_PIN   19
  #define C_PIN   5
  #define D_PIN   35 // Connected to GND on panel (21 if exist)
  #define E_PIN   -1 // Connected to GND on panel
  
  #define LAT_PIN 4
  #define OE_PIN  15
  #define CLK_PIN 33 
  
  #include "OneEighthScanMatrixPanel.h" // Virtual Display to re-map co-ordinates such that they draw correctly on a32x16 1/4 Scan panel 
  
  // placeholder for the matrix object
  MatrixPanel_I2S_DMA *dma_display   = nullptr;
  
  // placeholder for the virtual display object
  OneEighthMatrixPanel  *virtualDisp = nullptr;
  
  /******************************************************************************
   * Setup!
   ******************************************************************************/
  void setup() {
    
    delay(2000);
    Serial.begin(115200);
    Serial.println(""); Serial.println(""); Serial.println("");
    Serial.println("*****************************************************");
    Serial.println("*         1/8 Scan Panel Demonstration              *");
    Serial.println("*****************************************************");
  
  
    /******************************************************************************
     * Create physical DMA panel class AND virtual (chained) display class.   
     ******************************************************************************/
  
  	/*
  	The configuration for MatrixPanel_I2S_DMA object is held in HUB75_I2S_CFG structure,
  	All options has it's predefined default values. So we can create a new structure and redefine only the options we need
  
  	Please refer to the '2_PatternPlasma.ino' example for detailed example of how to use the MatrixPanel_I2S_DMA configuration
  	*/

	HUB75_I2S_CFG::i2s_pins _pins = {
		R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
		A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
		LAT_PIN, OE_PIN, CLK_PIN
	 };

	HUB75_I2S_CFG mxconfig(
				PANEL_RES_X*2,   	// DO NOT CHANGE THIS
				PANEL_RES_Y/2,   	// DO NOT CHANGE THIS
				PANEL_CHAIN    		// DO NOT CHANGE THIS
				//,_pins			// Uncomment to enable custom pins
	);
  
  	//mxconfig.driver = HUB75_I2S_CFG::FM6126A;     // in case that we use panels based on FM6126A chip, we can set it here before creating MatrixPanel_I2S_DMA object
  
  	// OK, now we can create our matrix object
  	dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  
  	// let's adjust default brightness to about 75%
  	dma_display->setBrightness8(50);    // range is 0-255, 0 - 0%, 255 - 100%
  
  	// Allocate memory and start DMA display
  	if( not dma_display->begin() )
  	  Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
  
  	// create VirtualDisplay object based on our newly created dma_display object
  	virtualDisp = new OneEighthMatrixPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
  
//  	virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));
//  	virtualDisp->setCursor(0, virtualDisp->height()/2); 
  
  	// Red text inside red rect (2 pix in from edge)
    virtualDisp->setCursor(0, 0); 
  	virtualDisp->print("  1234");
//  	virtualDisp->drawRect(1,1, virtualDisp->width()-2, virtualDisp->height()-2, virtualDisp->color565(255,0,0));
  
  	// White line from top left to bottom right
//  	virtualDisp->drawLine(0,0, virtualDisp->width()-1, virtualDisp->height()-1, virtualDisp->color565(255,255,255));
  }
  
  
  void loop() {
    
  
  } // end loop`
  
  
--> There's something incorrect here ????? 

[image]

mrfaptastic, 2021-11-09:
Unfortunately the 1/8 Scan panel workaround was never completed. I do not have one of these panels either so I cannot test.

Can you try drawing a line from pixel 0,0 to 32,63 without the use of the virtualDisp library and tell me what it looks like?

donnersm, 2021-11-09:
> Unfortunately the 1/8 Scan panel workaround was never completed. I do not have one of these panels either so I cannot test.
> 
> Can you try drawing a line from pixel 0,0 to 32,63 without the use of the virtualDisp library and tell me what it looks like?

Since i have one of those panels and without working lib its useless to me.., might as well put a stamp on it and ship it to someone @mrfaptastic pm me if your interested

mrfaptastic, 2021-11-09:
That would be great if you're happy to ship one. Shoot us an email at github@tty0.uk.to and we can sort something out. Hopefully you're in Europe or the UK to make this a cheap exercise?

AlfioSaitta, 2021-11-10:
> That would be great if you're happy to ship one. Shoot us an email at [github@tty0.uk.to](mailto:github@tty0.uk.to) and we can sort something out. Hopefully you're in Europe or the UK to make this a cheap exercise?

If you want i can send you various panels of different sizes and with different scan rates

mrfaptastic, 2021-11-10:
> If you want i can send you various panels of different sizes and with different scan rates.

If you are happy to donate then I certainly won't say no. Mark is kindly sending a 1/8 which I can use to get those panels working properly (and that will help you as well with your 3 x 1/8 scan panel display). 

Feel free to send what you think is of value to get supported and is most used. Can start continue expanding support in the library.

dickshaydle, 2021-11-22:
I have a 64x32 RGB Panel that worked with a setting for 1/8 scan with another library so i guess it is an 1/8 scan display.
[image]

I couldn't get it to run with current master branch of mrfaptastic but with the [patch-1 branch of Alfio](https://github.com/AlfioSaitta/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/patch-1).

But there were a some errors in the displayed image.
[image]

After a lot of trial and error i found some changes to the `OneEighthScanMatrixPanel.h`
[image]
``` c++
if(   ( _s_chain_party && !_chain_top_down && (row % 2 == 0) )  // serpentine vertically stacked chain starting from bottom row (i.e. ESP closest to ground), upwards
      ||
      ( _s_chain_party && _chain_top_down  && (row % 2 != 0) )  // serpentine vertically stacked chain starting from the sky downwards
)
{
  // First portion gets you to the correct offset for the row you need
  // Second portion inverts the x on the row
	coords.x = ((y / panelResY) * (virtualResX)) + (virtualResX - x) - 1;

  // inverts the y the row
  coords.y = panelResY - 1 - (y % panelResY);  
} 
else 
{
// Normal chain pixel co-ordinate
  coords.x = x + ((y / panelResY) * (virtualResX)) +1 ;
  coords.y = y % panelResY;
}

/* *******
* START: 1/8 Scan Panel Pixel Re-Mapping
*
* We have calculated the x, y co-ordinates as if we have a chain of standard panels this library is designed 
* for, this being 1/8 or 1/16 scan panels. We have to do some further hacking to convert co-ords to the 
* double length and 1/2 physical dma output length that is required for these panels to work electronically.
*/

 if ( (coords.y /8) % 4 == 0) { // 1st 8 rows
    coords.x += 64;
 }
 if ( (coords.y /8) % 4 == 1) { // 2nd 8 rows
    coords.y -= 8;
 }

 if ( (coords.y /8) % 4 == 2) { // returns true/1 for the 1st and 3rd 8-pixel 1/4th of a 32px high panel
    coords.x += 64;
    coords.y -= 8;
 }
 
 if ( (coords.y /8) % 4 == 3) { // returns true/1 for the 1st and 3rd 8-pixel 1/4th of a 32px high panel
    coords.y -= 16;
 }

/* 
 * END: 1/8 Scan Panel Pixel Re-Mapping
 * *******
 */
```

Any ideas why it behaves this way? 

What is needed to make this run with the aurora demo or with displaying images?
The aurora demos looks like it has double 8row blocks. The blocks are not equal but look very similar,

mrfaptastic, 2021-11-22:
Hi @dickshaydle. I am working on a fix for 1/8 scan panels as @donnersm has kindly provided me a 1/8 scan panel to test with.

I hope to have something that works with chaining as well, soon.

mrfaptastic, 2021-11-22:
You can't use the Aurora demo as is, it uses the underlying library which is 1/16 only.

You will need to merge the Aurora demo code with the new '1_8_ScanPanel.h' example and call this object instead of the underlying dma_display object which is deigned electrically for 1/16 panels only.

danieldecesaro, 2021-11-22:
Hello mrftastic,
I'm using 1/8 scan panels and I'm testing two types of panels with their example firmware.
In one of them everything works as expected, but in another there is an error in the coordinates.

The video of the two panels follows.

1. This one works:
[image]

https://user-images.githubusercontent.com/40169434/142909792-e9c91b03-2da8-442d-bece-91d73ca3fb30.mp4

2. This one has problems:
[image]

https://user-images.githubusercontent.com/40169434/142909637-3c0ece68-f4d8-4607-8948-7fd5d35881f7.mp4
 
I'm grateful if you can help me.

Thanks for your work!

Att.
  Daniel

mrfaptastic, 2021-11-22:
Wow. Sorry to say that is yet another 1/8 panel variant with some super weird addressing scheme. They are different panels it seems. Can't help you as I don't have one of these I'm afraid.

It's almost like these Chinese manufacturers try and make a new variant every week.

closed 2021-12-20 by mrfaptastic

muslihrzl, 2024-01-27:
> > Kerja bagus untuk mencari tahu hal itu. Seperti yang diharapkan, ini sebenarnya adalah rangkaian panel, dengan beberapa rangkaian aneh terjadi di antara 1/4 panel, dengan 1/4 panel ke-2 dirantai ke 1/4 panel pertama (atas).
> > Apa yang terjadi bila Anda menyetel: `HUB75_I2S_CFG mxconfig( 64, 32, CHAIN_LENGTH, _pins);`ke`HUB75_I2S_CFG mxconfig( 64, 16, CHAIN_LENGTH, _pins);`
> > Karena jika Anda menghitungnya, 8 piksel dari masing-masing 'setengah' panel tidak benar-benar digunakan mengingat perpustakaan ini dirancang untuk 1/2 panel pindaian, dan panel ini sebenarnya merangkai 1/4 ke-2 dengan 1/4 ke-1 4.
> 
> Hasilkan dengan pengaturan Anda dan koreksi sedikit perhitungannya:
> 
> [image]
> 
> ```
> #define CHAIN_LENGTH 2 // 2 Chained panels
> #define PANEL_WIDHT 64
> #define PANEL_HEIGHT 16 // 1/8 Scan in the specifics ???
> 
> #define R1_PIN  2
> #define G1_PIN  15
> #define B1_PIN  4
> #define R2_PIN  16
> #define G2_PIN  27
> #define B2_PIN  17
> 
> #define A_PIN   5
> #define B_PIN   18
> #define C_PIN   19
> #define D_PIN   -1 // Connected to GND on panel (21 if exist)
> #define E_PIN   -1 // Connected to GND on panel
> 
> #define LAT_PIN 26
> #define OE_PIN  25
> #define CLK_PIN 22
> 
> #include <Arduino.h>
> #include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
> 
> MatrixPanel_I2S_DMA *dma_display = nullptr;
> 
> void testScreen() {
>   
>   for (int x=0; x <= 63; x++) {
> 
>     // Red 8 Rows
>     dma_display->drawPixel(64+(x),0,dma_display->color565(255, 0, 0)); // ROW 1
>     dma_display->drawPixel(64+(x),1,dma_display->color565(210, 0, 0)); // ROW 2
>     dma_display->drawPixel(64+(x),2,dma_display->color565(180, 0, 0)); // ROW 3
>     dma_display->drawPixel(64+(x),3,dma_display->color565(150, 0, 0)); // ROW 4
>     dma_display->drawPixel(64+(x),4,dma_display->color565(125, 0, 0)); // ROW 5
>     dma_display->drawPixel(64+(x),5,dma_display->color565(90, 0, 0)); // ROW 6
>     dma_display->drawPixel(64+(x),6,dma_display->color565(60, 0, 0)); // ROW 7
>     dma_display->drawPixel(64+(x),7,dma_display->color565(40, 0, 0)); // ROW 8
> 
>     // Blue 8 Rows
>     dma_display->drawPixel(0+x,0,dma_display->color565(0, 0, 255)); // ROW 9
>     dma_display->drawPixel(0+x,1,dma_display->color565(0, 0, 210)); // ROW 10
>     dma_display->drawPixel(0+x,2,dma_display->color565(0, 0, 180)); // ROW 11
>     dma_display->drawPixel(0+x,3,dma_display->color565(0, 0, 150)); // ROW 12
>     dma_display->drawPixel(0+x,4,dma_display->color565(0, 0, 125)); // ROW 13
>     dma_display->drawPixel(0+x,5,dma_display->color565(0, 0, 90)); // ROW 14
>     dma_display->drawPixel(0+x,6,dma_display->color565(0, 0, 60)); // ROW 15
>     dma_display->drawPixel(0+x,7,dma_display->color565(0, 0, 40)); // ROW 16
> 
>     // White 8 Rows
>     dma_display->drawPixel(64+x,8,dma_display->color565(255, 255, 255)); // ROW 17
>     dma_display->drawPixel(64+x,9,dma_display->color565(210, 210, 210)); // ROW 18
>     dma_display->drawPixel(64+x,10,dma_display->color565(180, 180, 180)); // ROW 19
>     dma_display->drawPixel(64+x,11,dma_display->color565(150, 150, 150)); // ROW 20
>     dma_display->drawPixel(64+x,12,dma_display->color565(125, 125, 125)); // ROW 21
>     dma_display->drawPixel(64+x,13,dma_display->color565(90, 90, 90)); // ROW 22
>     dma_display->drawPixel(64+x,14,dma_display->color565(60, 60, 60)); // ROW 23
>     dma_display->drawPixel(64+x,15,dma_display->color565(40, 40, 40)); // ROW 24
> 
>     // Green 8 Rows
>     dma_display->drawPixel(0+x,8,dma_display->color565(0, 255, 0)); // ROW 25
>     dma_display->drawPixel(0+x,9,dma_display->color565(0, 210, 0)); // ROW 26
>     dma_display->drawPixel(0+x,10,dma_display->color565(0, 180, 0)); // ROW 27
>     dma_display->drawPixel(0+x,11,dma_display->color565(0, 150, 0)); // ROW 28
>     dma_display->drawPixel(0+x,12,dma_display->color565(0, 125, 0)); // ROW 29
>     dma_display->drawPixel(0+x,13,dma_display->color565(0, 90, 0)); // ROW 30
>     dma_display->drawPixel(0+x,14,dma_display->color565(0, 60, 0)); // ROW 31
>     dma_display->drawPixel(0+x,15,dma_display->color565(0, 40, 0)); // ROW 32
>   }
> }
> 
> void setup() {
> 
>   Serial.begin(115200);
>   delay(1000);
> 
>   HUB75_I2S_CFG::i2s_pins _pins = {
>     R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, 
>     A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, 
>     LAT_PIN, OE_PIN, CLK_PIN
>   };
> 
>   HUB75_I2S_CFG mxconfig( PANEL_WIDHT, PANEL_HEIGHT, CHAIN_LENGTH, _pins);
> 
>   mxconfig.double_buff = true; // ????
>   mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
> 
>   dma_display = new MatrixPanel_I2S_DMA(mxconfig);
>   dma_display->begin();
>   dma_display->setBrightness8(200); //0-255
>   dma_display->clearScreen();
> }
> 
> void loop() {
>     //delay(40); 
>     testScreen();
> }
> ```
> 
> Setelah melakukan penelitian di berbagai pengecer dan pabrik Tiongkok, saya menemukan beberapa informasi lebih lanjut di panel:
> 
> ```
> Pixel Pitch: P4
> Dimension: 128X256
> Lamp bead: RGB
> LED Type: SMD1921
> Scanning Mode: 1/8
> ```

[image]

For results like this, try the code above, is the scanning different, can anyone help

muslihrzl, 2024-01-27:
[image]
 for the panel, use P5 64x32-8S, CaiLiang

board707, 2024-01-27:
You need a kind of reverse engineering to determine scan.

muslihrzl, 2024-01-27:
[image]

Can you help with the coding, for scanning like this?

muslihrizal, 2024-01-29:
> You need a kind of reverse engineering to determine scan.

can help with setting the coordinates, still confused

board707, 2024-01-29:
try this
```
inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
  VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  if ( (y / 8) == 0) { 
    coords.x += ((coords.x / 32)+1)*32; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }
  else {
    coords.x += (coords.x / 32)*32; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }

  // http://cpp.sh/4ak5u
  // Real number of DMA y rows is half reality
  // coords.y = (y / 16)*8 + (y & 0b00000111);   
  coords.y = (y >> 4)*8 + (y & 0b00000111);   

  return coords;
}
```

muslihrzl, 2024-01-29:
> try this
>
> ```
> inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
>   VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach
>
>   if ( [...]

[image]

Thank you very much for your help, but there are still defects in the appearance, which parts I need to fix

board707, 2024-01-29:
sorry, just typo.
try this
```
inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
  VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach

  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
    return coords;
  }

  if ( (y & 8) == 0) { 
    coords.x += ((coords.x / 32)+1)*32; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }
  else {
    coords.x += (coords.x / 32)*32; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
  }

  // http://cpp.sh/4ak5u
  // Real number of DMA y rows is half reality
  // coords.y = (y / 16)*8 + (y & 0b00000111);   
  coords.y = (y >> 4)*8 + (y & 0b00000111);   

  return coords;
}
```

muslihrzl, 2024-01-30:
> sorry, just typo. try this
>
> ```
> inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
>   VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining [...]

[image]

Thank you very much, it has worked out the way it looks

muslihrzl, 2024-02-09:
I tried 1 row and 2 col panel, it seems to appear normal

[image]

muslihrzl, 2024-02-09:
> I tried 1 row and 2 col panel, it seems to appear normal
> 
> ![Image 2024-02-09 at 23 08 20](https://private-user-images.githubusercontent.com/157888752/303711820-52cb4312-0e16-4d94-b9cf-e118d8d0a04c.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0OTYxMjAsIm5iZiI6MTcwNzQ5NTgyMCwicGF0aCI6Ii8xNTc4ODg3NTIvMzAzNzExODIwLTUyY2I0MzEyLTBlMTYtNGQ5NC1iOWNmLWUxMThkOGQwYTA0Yy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMTYyMzQwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OGZkNWE0ZTJkNWY1YzAzOGJlYjkyNTY3ODIwYzY5ZDY5ZmY0NTRkYjMyMjBhMmFjZmM0NzI2OGI0MzY4OGU0NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.4I_d5C5QIBXoTYBK3HA0OQEV2b2c1S8oK-md-naydY0)

`
      #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
      #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
      
      #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
      #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
      #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
      
      // Change this to your needs, for details on VirtualPanel pls read the PDF!
      #define SERPENT true
      #define TOPDOWN true
  `

muslihrzl, 2024-02-09:
> I tried 1 row and 2 col panel, it seems to appear normal
>
> ![Image 2024-02-09 at 23 08 [...]

`
      #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
      #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
      
      #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
      #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
      #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
      
      // Change this to your needs, for details on VirtualPanel pls read the PDF!
      #define SERPENT true
      #define TOPDOWN true
  `

muslihrzl, 2024-02-09:
Then I tried 2 rows and 1 col panel, some panels didn't appear
[image]

`
      #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
      #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
      
      #define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
      #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
      #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
      
      // Change this to your needs, for details on VirtualPanel pls read the PDF!
      #define SERPENT true
      #define TOPDOWN true
    `

please help me to appear normal

muslihrizal, 2024-02-09:
> > Saya mencoba 1 baris dan 2 panel kolom, sepertinya terlihat normal
> > ![Gambar 09-02-2024 pukul 23 08 20](https://private-user-images.githubusercontent.com/157888752/303711820-52cb4312-0e16-4d94-b9cf-e118d8d0a04c.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc0OTY1MDEsIm5iZiI6MTcwNzQ5NjIwMSwicGF0aCI6Ii8xNTc4ODg3NTIvMzAzNzExODIwLTUyY2I0MzEyLTBlMTYtNGQ5NC1iOWNmLWUxMThkOGQwYTA0Yy5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIwOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMDlUMTYzMDAxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmMzNmMyY2VhYTBiZGJmYmE4ODk5ZDAxMTM5MjkzMTRkYTEzOGU2OTgyMWQzZDQxNmM0MWFkYjg1NDNjYmY4NyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.IjHO_Tt9K0T2AJznr101uDiqhTelmWSNHoS86P599kw)
> 
> ` #define PANEL_RES_X 64 // Jumlah piksel lebar setiap modul panel INDIVIDU. #define PANEL_RES_Y 32 // Jumlah piksel tinggi setiap modul panel INDIVIDU.
> 
> ```
>   #define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
>   #define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
>   #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
>   
>   // Change this to your needs, for details on VirtualPanel pls read the PDF!
>   #define SERPENT true
>   #define TOPDOWN true
> ```
> 
> `

[image]

board707, 2024-02-09:
Hi
Where you added my virtual co-ordinates code that I wrote to help you two weeks ago?

muslihrzl, 2024-02-10:
> Hi Where you added my virtual co-ordinates code that I wrote to help you two weeks ago?

`	#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>
	#include <ESP32-HUB75-VirtualMatrixPanel-I2S-DMA.h>

	class OneEightScanPanel : public VirtualMatrixPanel
	  {
	  public:
		using VirtualMatrixPanel::VirtualMatrixPanel; // inherit VirtualMatrixPanel's constructor(s)

	  protected:
			/* Convert Real World 'VirtualMatrixPanel' co-ordinates (i.e. Real World pixel you're looking at
			   on the panel or chain of panels, per the chaining configuration) to a 1/8 panels
			   double 'stretched' and 'squished' coordinates which is what needs to be sent from the
			   DMA buffer.
			   Note: Look at the One_Eight_1_8_ScanPanel code and you'll see that the DMA buffer is setup
			   as if the panel is 2 * W and 0.5 * H !
			*/
		
		VirtualCoords getCoords(int16_t &x, int16_t &y);
	  };
	  
	inline VirtualCoords OneEightScanPanel::getCoords(int16_t &x, int16_t &y) {
		VirtualMatrixPanel::getCoords(x, y); // call to base to update coords for chaining approach
	 
	  if ( coords.x == -1 || coords.y == -1 ) { // Co-ordinates go from 0 to X-1 remember! width() and height() are out of range!
	   
		return coords;
		}
		  coords.x = x + ((y / panelResY) * (virtualResX)) ;
		  coords.y = panelResY - 1 - (y % panelResY);

	  if ( (y & 8) == 0) { 
		  coords.x += ((coords.x / 32)+1)*32; // 1st, 3rd 'block' of 8 rows of pixels, offset by panel width in DMA buffer
		  }
	  else {
		  coords.x += (coords.x / 32)*32; // 2nd, 4th 'block' of 8 rows of pixels, offset by panel width in DMA buffer
		   }
		   
			coords.y = (y >> 4)*8 + (y & 0b00000111);    
					  
		  return coords;
		}
	   
	  // Panel configuration
	  
		  #define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
		  #define PANEL_RES_Y 32 // Number of pixels tall of each INDIVIDUAL panel module.
		  
		  #define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
		  #define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
		  #define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another
		  
		  // Change this to your needs, for details on VirtualPanel pls read the PDF!
		  #define SERPENT true
		  #define TOPDOWN true
		
	  // placeholder for the matrix object
	  MatrixPanel_I2S_DMA *dma_display    = nullptr;
	  OneEightScanPanel   *virtualDisp    = nullptr;
	

	void setup()
	{
		HUB75_I2S_CFG mxconfig( PANEL_RES_X*2, PANEL_RES_Y/2, PANEL_CHAIN );
		dma_display = new MatrixPanel_I2S_DMA(mxconfig);
		dma_display->setBrightness8(20);    // range is 0-255, 0 - 0%, 255 - 100%
		virtualDisp = new OneEightScanPanel((*dma_display), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y,SERPENT,TOPDOWN);
		dma_display->begin();

		virtualDisp->setTextColor(virtualDisp->color565(0, 0, 255));
		virtualDisp->setCursor(0, virtualDisp->height()/2); 
	  
		// Red text inside red rect (2 pix in from edge)
		virtualDisp->print("  1234");
		virtualDisp->drawRect(1,1, virtualDisp->width()-2, virtualDisp->height()-2, virtualDisp->color565(255,0,0));
	  
		// White line from top left to bottom right
		virtualDisp->drawLine(0,0, virtualDisp->width()-1, virtualDisp->height()-1, virtualDisp->color565(255,255,255));

	}

	void loop()
	{
		
	}
`

board707, 2024-02-10:
Please show your ESP32-HUB75-VirtualMatrixPanel-I2S-DMA.h file

muslihrzl, 2024-02-10:
> Silakan tunjukkan file ESP32-HUB75-VirtualMatrixPanel-I2S-DMA.h Anda

muslihrzl, 2024-02-10:
> Silakan tunjukkan file ESP32-HUB75-VirtualMatrixPanel-I2S-DMA.h Anda

`

		#ifndef _ESP32_VIRTUAL_MATRIX_PANEL_I2S_DMA
		#define _ESP32_VIRTUAL_MATRIX_PANEL_I2S_DMA

		/*******************************************************************
			Class contributed by Brian Lough, and expanded by Faptastic.
			
			Originally designed to allow CHAINING of panels together to create
			a 'bigger' display of panels. i.e. Chaining 4 panels into a 2x2
			grid.
			
			However, the function of this class has expanded now to also manage
			the output for 1/16 scan panels, as the core DMA library is designed
			ONLY FOR 1/16 scan matrix panels.

			YouTube: https://www.youtube.com/brianlough
			Tindie: https://www.tindie.com/stores/brianlough/
			Twitter: https://twitter.com/witnessmenow
		 *******************************************************************/

		#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
		#ifndef NO_GFX
		 #include <Fonts/FreeSansBold12pt7b.h>
		#endif

		struct VirtualCoords {
		  
		  int16_t x;
		  int16_t y;
		  int16_t virt_row; // chain of panels row
		  int16_t virt_col; // chain of panels col
		  
		  VirtualCoords()  : x(0), y(0)
		  { }
			  
		};

		#ifndef NUM_COLS
		#define NUM_COLS 1 // Number of INDIVIDUAL PANELS per ROW
		#endif

		#ifndef NUM_ROWS
		#define NUM_ROWS 1 // Number of rows of chained INDIVIDUAL PANELS
		#endif

		enum PANEL_SCAN_RATE {NORMAL_ONE_SIXTEEN, ONE_EIGHT_32, ONE_EIGHT_16,ONE_EIGHT_8};

		#ifdef USE_GFX_ROOT
		class VirtualMatrixPanel : public GFX
		#elif !defined NO_GFX
		class VirtualMatrixPanel : public Adafruit_GFX
		#else
		class VirtualMatrixPanel
		#endif	
		{

		  public:
			int16_t virtualResX;
			int16_t virtualResY;

			int16_t vmodule_rows;
			int16_t vmodule_cols;

			int16_t panelResX;
			int16_t panelResY;
			
			int16_t dmaResX; // The width of the chain in pixels (as the DMA engine sees it)
[... 251 lines cut ...]
			int top_left_x = (panel == 0) ? 0:(panel*panelResX);
			this->display->drawRect( top_left_x, 0, panelResX, panelResY, this->display->color565( 0, 255, 0));
			this->display->setCursor(panel*panelResX, panelResY-3); 
			this->display->print((vmodule_cols*vmodule_rows)-panel);    
		  } 

		}
		#endif
		/*
		// need to recreate this one, as it wouldn't work to just map where it starts.
		inline void VirtualMatrixPanel::drawIcon (int *ico, int16_t x, int16_t y, int16_t icon_cols, int16_t icon_rows) {
		  int i, j;
		  for (i = 0; i < icon_rows; i++) {
			for (j = 0; j < icon_cols; j++) {
			  // This is a call to this libraries version of drawPixel
			  // which will map each pixel, which is what we want.
			  //drawPixelRGB565 (x + j, y + i, ico[i * module_cols + j]);
			  drawPixel (x + j, y + i, ico[i * icon_cols + j]);
			}
		  }
		}
		*/

		#endif
`

[13 more replies]