mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #159: SmartMatrix style multiplexing of addr and RGB data?

twinotter, 2021-07-30:
Hi,

I need to find a way to lower the memory footprint of my software which is, so far, based around the SmartMatrix library. I'm just starting adapting your code to the SmartMatrix style of latching out RGB and addr lines on the gpio. (Since I can't easily change the hardware I already have.) It's a bit sticky, to put it mildly. Do you have any suggestions along this route? I'll post in this thread when I finish working my way through the two libraries...

Thanks!
Rudy

mrfaptastic, 2021-07-31:
This library doesn't support any form of multiplexing / dual use of GPIOs, you would be best suited to sticking with the SmartMatrux library.

closed 2021-08-12 by mrfaptastic

twinotter, 2021-09-13:
I've patched your libraries to work with multiplexed addr and RBG data using the method SmartMatrix uses. The main reason to do this is for the memory savings. One 64x64 panel at an lsbMsbTransitionBit of 3 (for speed) uses only 23kB. I got eight 64x64 panels working on the PatternPlasma demo (at a paltry frame rate of 9fps), but I was still only using 133kB.

I'm not sure if this is something you'd like to include in your library or not. And I don't understand Git well enough to know the best way to upload the patch to let you use it. I tried to stick with your coding style where possible.

Let me know if you're interested and how best to get the code to you.

Best,
Rudy

mrfaptastic, 2021-09-13:
Hi @twinotter,

Raise a pull request and I'll take a look.

If there's a 'clean' way of doing this then I may merge. I want to try avoid making this library needlessly complex for rare use cases - but given you have made these changes and I am working on support for the 'newer' ESP32's which have a pitiful amount of RAM (Only 320kB vs. 520kB on the 'original'), this could actually be quite handy.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/386SX-33

mrfaptastic, 2021-09-13:
Attach a .zip file here of your modified library as well if you want, if that's easier.

twinotter, 2021-09-14:
here's the zip file. I'll see if I can figure out pull requests. I've been meaning to learn git. :)

[ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/7165737/ESP32_HUB75_LED_MATRIX_PANEL_DMA_Display.zip)