mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #161: Buffered writing of text

lkaino, 2021-08-03:
I have understood that only way to write a changing text to the display is to clear the display in between. This makes the display flicker.

If I don't clear the screen in between the text is added on top of the previous pixels.

Double buffering would solve this, but there is not enough RAM available on the ESP32 for both WiFi and this.

Is it possible to buffer the whole frame before sending it out (single frame buffer)?

DarrylStrong, 2021-08-03:
Perhaps just clear the part of the display that's being overwritten. This will produce less flicker than a full screen clear.

What size display are you driving? I am using this library to drive a 384x64 display with WiFi, ethernet and full buffering as I need to be able to read pixel colour.

mrfaptastic, 2021-08-12:
> Is it possible to buffer the whole frame before sending it out (single frame buffer)?
> 
> I'm using clearScreen() for clearing, but is there a way to clear the buffer only?

If you turn on double-buffering or you decide to have your screen buffer - the result would be the same wouldn't it? You will use double the memory to 'store' what 'was' (or should be) on the display. 

However, who knows how the ESP32 allocates memory or your sketch. Try turning double-buffering off, and then using a seperate graphics layer to 'store' what you want to print to screen, before 'sending' it to the display.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/GraphicsLayer

If you're printing everything the same color, then technically you could implement your own 1-bit bitmap graphics buffer, and then print to screen with the same color, if you know what I mean.

closed 2021-08-12 by mrfaptastic