mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #163: Trouble with V2.0.5 (Flickering, Ghost, offset) and example code to check

FrenchLab47, 2021-08-09:
Hello mrfaptastic

Forgive again my English assisted by Google

I meet a problem of display with version 2.0.5 of your library.
After some tests I isolated this defect, and I offer you an easy example to reproduce.

A few days ago, I presented my SwinGraf project in your forum.
The models shown in the videos work perfectly with ESP32-HUB75-MatrixPanel-I2S-DMA version 1.2.4.
When you have published version 2.0.0, I tried some quick tests but I had too many offsets and ghosts, and not a lot of time to test the settings.

Last week I reload my tests with your version 2.0.5.
By reading all your recommendations, I could operate 90% of my program, but there are still some visible defects on moving frames around texts, and audio spectrums.
And also pixels that appear at left when the text scrolls.

I read your "common issues" and settled according to your recommendations.
After that I got a suitable display for fixed texts and graphics.
But in the case of scrolling texts, some LEDs light on the left on column 1 (like a mirror of column 128 or 129).
Another fault appears with animated frames (animated points that move around the panel).
When a frame is displayed, it causes a "definitive" offset on lines 1, 15, 16, and 32.
I say "definitive" because once triggered, the whole display is affected by this defect, and you have to restart the device to restore the display.
The SwinGraf code is bulky and I thought it could cause slowdowns in signals or memory.

***************** Issue noted and easily verifiable ****************************

To simplify all this, I used your first example "1_SimpleTestShapes" on two panels (2 x 64x32).
I used a new ESP32, and a new PCB.

[image]

[image]

The display of your example was normal and everything worked until I draw lines or out-of-frame points.

`dma_display.drawFastVLine(0, 0, dma_display.height() + 1,  myRED);`

An offset occurs if I trace a line or pixel that overflows from the panel.

[image]

The attached file contains some tests to cause this defect.

[1_SimpleTestShapesWithFrame.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/6955180/1_SimpleTestShapesWithFrame.zip)

Just activate the corresponding lines at each test, and read my remarks.

- Your example use virtual mode. I restored classic mode and compared with your two library versions.
- I raised all the possible doubts with different models of panels (64x32 p4 or p5 and 64x64) and I got the same faults.
- I tested several power supplies and added capacitors.
- As the wiring of my PCB is different from your standard wiring, I wired a new one with the same connections as you.

Despite all these tests I confirm that the defects observed never occur with version 1.2.4, and always with version 2.0.5.

Finally I tried to compare your two versions to find differences between certain variables, but my skills have limits, and I did not find anything weird.
Now, you have the details of the problem, and surely the solution, but that it does not disturb your holidays.

Cordially.

FrenchLab47, 2021-08-12:
Hello mrfaptastic

Thanks for your intervention.
I just did a quick test, and I note the change, but still some problems.
I would try better tomorrow, and I could give you a more precise answer.

mrfaptastic, 2021-08-12:
Can you try and **not** use the drawFastVLine or drawFastHLine functions as they exist in 2.x branch?

That is, use the old 'drawLine' function. I think there might be an issue with drawFastVLine or drawFastHLine

FrenchLab47, 2021-08-13:
I just tested (again) with "drawline" and there are also problems when it overflow the panel.

I had already done these tests with "drawline". They are in the file I have provided at the beginning. You can activate them to check the differences.

This morning I also compared what you have changed yesterday, but the result is worse. Now the offset fault is visible on several lines. I tried to play with the > and =, but chance was not with me.

Another thing: I work with a fixed power supply that delivers a little over 5V (5.4V) and I remember old p5 panels that are sensitive to voltage. To lift doubt this morning I used an adjustable power supply. But nothing of Flagrand, the variation of 4.8V to 5.4V changes a little the color of the staggered points, but does not solve the offset.

If you are not at home or if you do not have the opportunity to test. I can to do other tests, but in my case there is no urgency. Version 1.2.4 always works perfectly.

mrfaptastic, 2021-08-13:
Any ideas @vortigont ?

vortigont, 2021-08-13:
Interesting.. let me set back my panel and check this.
Only one question so far
@FrenchLab47 how do you build your sketch?
I can see panel defines in your .ino file 

#define MATRIX_WIDTH 128
#define MATRIX_HEIGHT 32

but later on you use old style constructor which default's are 64x32 
dma_display.begin(R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN); // my pins

If you build this under Arduino IDE your defines won't propagate while building the lib itself.
This is just to make things clear.

FrenchLab47, 2021-08-13:
Hello vortigont,

Yes I define 128 and 32 in Arduino Ide just to remember.
And especially in the .h of the Library for spread it.

But I tested with a single panel of 64x32 with the original settings, and the problem is identical. If you have the opportunity to test my .ino, you must be able to reproduce it.

vortigont, 2021-08-15:
OK! I will set my panel back and check this out.

mrfaptastic, 2021-08-16:
I've found the problem and fixed it in the SM5266P tree. 
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/SM5266P

``` dma_display->drawFastVLine(0, 0, dma_display->height() + 1,  myRED); // Defect on the text (and line) ```

Exposed an out of bounds error with the code, as you shouldn't be able to draw a line that's greater than the height of the chain of panels (the +1). 

``` dma_display->drawFastVLine(0, 0, dma_display->height() - 1,  myRED); ``` works perfectly for example.

Also, @FrenchLab47, you should use the new format with the pointer based used of the library:

```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define PANEL_RES_X 64     // Number of pixels wide of each INDIVIDUAL panel module.
#define PANEL_RES_Y 32     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

// Change these to whatever suits
/*
#define R1_PIN 25
#define G1_PIN 26
#define B1_PIN 27
#define R2_PIN 14
#define G2_PIN 12
#define B2_PIN 13
#define A_PIN 3
#define B_PIN 22
#define C_PIN 21
#define D_PIN 33
#define E_PIN -1 // required for 1/32 scan panels, like 64x64. Any available pin would do, i.e. IO32
#define LAT_PIN 4
#define OE_PIN 15
#define CLK_PIN 2
*/

MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myBLACK = dma_display->color565(0, 0, 0);
uint16_t myWHITE = dma_display->color565(255, 255, 255);
uint16_t myRED = dma_display->color565(255, 0, 0);
uint16_t myGREEN = dma_display->color565(0, 255, 0);
uint16_t myBLUE = dma_display->color565(0, 0, 255);

// Input a value 0 to 255 to get a color value.
// The colours are a transition r - g - b - back to r.
// From: https://gist.github.com/davidegironi/3144efdc6d67e5df55438cc3cba613c8
uint16_t colorWheel(uint8_t pos) {
  if (pos < 85) {
    return dma_display->color565(pos * 3, 255 - pos * 3, 0);
  } else if (pos < 170) {
    pos -= 85;
    return dma_display->color565(255 - pos * 3, 0, pos * 3);
  } else {
    pos -= 170;
    return dma_display->color565(0, pos * 3, 255 - pos * 3);
  }
}

void drawText(int colorWheelOffset)
{

  // draw text with a rotating colour
  dma_display->setTextSize(1);     // size 1 == 8 pixels high
  dma_display->setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display->setCursor(21, 0);    // start at top left, with 8 pixel of spacing
  uint8_t w = 0;
  const char *str = "ESP32 DMA TEST";
  for (w = 0; w < strlen(str); w++) {
[... 102 lines cut ...]

  /* ***************************************************************************************
    Test 5 : Now a test with free lines
  */

  //dma_display->drawLine(-5, 5, 50, 25,  myRED); // OK even if the line begin outside the frame

  //dma_display->drawLine(15, 5, 50, 25,  myRED); // OK

  //dma_display->drawLine(-5, 25, 50, 5,  myRED); // OK even if the line begin outside the frame

  /* ***************************************************************************************
    Test 6 : I finish this tests with drawLine and horizontal lines
  */

  //dma_display->drawLine(5, 25, 50, 25,  myRED); // Horizontal line in the frame OK

  //dma_display->drawLine(-1, 25, 50, 25,  myWHITE); // OUPS - This horizontal line that begin outside the frame is INVISIBLE (replace -1 by 1 to see)

  //dma_display->drawLine(67, 25, 130, 25,  myRED); // If the line overflows on the right, it is well drawn, but it always remains a pixel on the left

  delay(2000);

}
```

FrenchLab47, 2021-08-16:
I'm sorry to disturb you with that.

I have just downloaded your files with the last correction.
I also tested the new "matrix->" format.

We agree, and as I also said at first, the display is correct, as long as we do not overflow the frame!
This is valid with "drawline" (in the example), but parasitic pixels also appear when animated texts or images that come out of the frame.

In my current project, I have big text scrolls that can overflow 800 pixels or more.
Animated gifs that move in all directions, and come out of the frame.
It seems to me complicated or impossible to control each pixel that passes the limit.

For the moment your version LEGACY 1.2.4 accepts all these freedoms, without displaying the slightest parasite pixel.
It gives me satisfaction as long as I do not need several lines of panels. 

Thanks.

mrfaptastic, 2021-08-16:
Add `mxconfig.latch_blanking = 4;` before the  `dma_display = new MatrixPanel_I2S_DMA(mxconfig);`

Does that fix the issue?

mrfaptastic, 2021-08-16:
Je dois également noter que la bibliothèque doit toujours vérifier si un pixel est hors de portée et l'ignorer. J'utilise aussi google translate ici pour vous aider :-)

C'est ennuyeux pour moi qu'il y ait une régression dans la nouvelle version de la bibliothèque, donc j'aimerais idéalement résoudre ce problème.

mrfaptastic, 2021-08-16:
Essayez ceci dans setup():

```

void setup() {
  
    HUB75_I2S_CFG::i2s_pins _pins = {R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
    HUB75_I2S_CFG mxconfig(
      PANEL_RES_X,   // module width
      PANEL_RES_Y,   // module height
      PANEL_CHAIN//,    // Chain length
      _pins
    );

    mxconfig.latch_blanking = 4;
    mxconfig.clkphase = false;  // Pouvez-vous essayer cela aussi comme vrai ou faux et voir si cela fait une différence ?
    //mxconfig.driver = HUB75_I2S_CFG::FM6126A;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);

```

FrenchLab47, 2021-08-16:
Effectivement c'est bien mieux !

A propos de
mxconfig.latch_blanking = 4;
mxconfig.clkphase = false;
Effectivement je n'ai pas pensé à les régler sur le dernier fichier.

Il faut bien ces deux lignes ensemble, PLUS votre dernière correction.
Et j'obtiens un affichage net, même si des lignes débordent du cadre.

Pour l'instant ça fonctionne avec les objets fixes de l'exemple ci-joint,
Demain je vais adapter tout mon code au nouveau format mais ça risque de m'occuper la matinée.
Je vous tiens informé... 
Bonne soirée.

mrfaptastic, 2021-08-16:
Juste pour vous assurer que votre réponse n'est pas "perdue dans la traduction". Mes suggestions ont-elles fait fonctionner parfaitement votre code maintenant?

FrenchLab47, 2021-08-17:
Je suis de retour ce matin avec mes dernières observations :
Comme je le disais dans mon message précédent votre correctif fonctionne parfaitement.

Hier soir je n'ai testé que le fichier .ino que nous utilisons comme exemple, je n'ai relevé aucun défaut.
Aucune différence avec votre version legacy, à part cette ligne qui disparait encore complètement, c'est bizare mais facile sans importance.
`dma_display->drawLine(-1, 25, 50, 25,  myWHITE); // OUPS - This horizontal line that begin outside the frame is INVISIBLE (replace -1 by 1 to see)   `

Maintenant les choses sérieuses.
Ce matin j'ai entrepris de faire fonctionner mon code (SwinGraf) avec votre librairie et ce nouveau correctif.

Première étape : 
Avant de m'attaquer a modifier le format objet dans 40000 lignes de code, j'ai commencé par le format classique et j'ai gardé l'écriture "matrix." avec un point. (et les réglages en dur dans le fichier .h)
J'ai testé les textes défilant, les encadrements animés, les gifs qui se déplacent.
J'ai noté des petits ralentissements que je n'ai pas avec la version legacy. (peut-être des détails que je devrais optimiser)
Mais aucun défaut d'affichage
Tout fonctionne parfaitement avec cette version corrigée, BRAVO !

Deuxième étape : 
Le passage au nouveau format.
Après modification de mon code avec la nouvelle écriture, je n'arrive plus à le compiler   
Il me reste une erreur au moment de l'initialisation de la librairie U8G2 utilisée pour les polices de texte.
`no matching function for call to 'U8G2_FOR_ADAFRUIT_GFX::begin(MatrixPanel_I2S_DMA*&)'`

Mais ceci n'est plus de votre ressort, et je vais essayer de comprendre ce qu'il se passe.

Merci encore pour votre écoute et votre aide.

mrfaptastic, 2021-08-17:
> dma_display->drawLine(-1, 25, 50, 25, myWHITE); // OUPS - This horizontal line that begin outside the frame is INVISIBLE (replace -1 by 1 to see)

C'est le comportement de la bibliothèque sous-jacente AdaFruit GFX. Je n'ai jamais vu l'utilisation de coordonnées négatives.

Il ne devrait pas y avoir de ralentissement avec cette bibliothèque non plus - elle est beaucoup plus rapide que la version héritée de nos tests, donc cela pourrait être autre chose.

> Après modification de mon code avec la nouvelle écriture, je n'arrive plus à le compiler
> Il me reste une erreur au moment de l'initialisation de la librairie U8G2 utilisée pour les polices de texte.
> `no matching function for call to 'U8G2_FOR_ADAFRUIT_GFX::begin(MatrixPanel_I2S_DMA*&)'`

Je ne peux pas t'aider car c'est une autre bibliothèque.

En toute honnêteté, si la version héritée fonctionne correctement, il semble qu'il serait plus facile de continuer à l'utiliser. Au moins à partir de cet exercice, nous avons trouvé un vrai bogue dans cette bibliothèque et l'avons corrigé - alors merci pour votre aide.

vortigont, 2021-08-18:
Sorry, but my last french lessons are decades behind :)
But there is something defenetely wrong with lines, I could see artifacts when drawing multiple single lines also.
Will take a look into this more closely, not sure if this my panel/wiring issue or lib.

FrenchLab47, 2021-08-18:
Hello vortigont,

Here is a small summary in my translated French.

I did all my tests by comparing with the LEGACY version 1.2.4
The last fix proposed by mrfaptastic fix the display defects I had described in the attached .ino file.
Only Test 6 (White Line) at the end of the Sketch still does not appear on the new version, but it is unimportant.

Then I compiled my big code (swingraf) with the new library.
I do not use virtual mode and the new format (matrix->) because my code refuses to compile with the U8G2_FOR_ADAFRUIT_GFX library

Even in normal mode, the result is perfect.
The following settings are the ones that are best for my panels:
mxconfig.latch_blanking = 4;
mxconfig.clkphase = false;

And this morning, I even solved slowdowns with big texts that scroll by increasing the value min_refresh_rate
min_refresh_rate 255;
I would be well mounted higher, but this value is limited by uint8_t

Conclusion:
Everything works for me!
Thank you.

vortigont, 2021-08-18:
well, glad it all OK for you now, @FrenchLab47 
I will check the latest changes from @mrfaptastic might solve my garbage too.
BTW, bumping refresh rate to high results in degraded dynamic range, mostly for shadows. But I think it's OK if you do text mostly.
It could also be of a great help here to reduce PIXEL_COLOR_DEPTH_BITS build-time option.

commit 1fa0b424 (2021-08-16): Fix #163

Array out of bounds DMA change.