mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #164: Display boards with SM5266P Row drivers

DarrylStrong, 2021-08-10:
Hi,

I have recently had delivered some display boards which use the SM5266P drivers for the row select. This is a shift register type using only ABC signals to drive a 1:32 scan 128x64 PCB. 

https://datasheetspdf.com/pdf-file/1328896/Sunmoon/SM5266P/1

Before I have a go at modifying the library to drive these has anyone had any joy previously with these?

They are a shift register with Clock, Data and Blank inputs rather that a straight 5:32 demux.

Cheers!

Daz

donnersm, 2021-08-10:
I have the same panel here, received last month and the rows is giving me trouble...
I addressed it in isue:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/154
But so far no solution.
If you are going to modify the lib to drive this one successfully, please keep us updated.

DarrylStrong, 2021-08-12:
Okay - problem solved.

I have modified the function void MatrixPanel_I2S_DMA::clearFrameBuffer(bool _buff_id) in ESP32-HUB75-MatrixPanel-I2S-DMA.cpp

Essentially the DE row drive signals for multiplexing stay the same but the ABC signals are clk, di, bk now instead of being binay coding in three bits for the row drivers. This means that the rows to be turned on are defined by shifting data into the SM5266P chip. (Datasheet here- [https://datasheetspdf.com/pdf-file/1328896/Sunmoon/SM5266P/1])

Page six shows the setup on my 128x64 P2.5 panels.

Here is the modified code :

///////////////////////////////////////////////////////////////////////////////////////////////
//
// modifications here for row shift register type SM5266P
//
///////////////////////////////////////////////////////////////////////////////////////////////

// get last pixel index in a row of all colordepths
int x_pixel = dma_buff.rowBits[row_idx]->width * dma_buff.rowBits[row_idx]->color_depth;
//Serial.printf(" from pixel %d, ", x_pixel);
// fill all x_pixels except color_index[0] (LSB) ones, this also clears all color data to 0's black
do {
  --x_pixel;
#ifdef SM5266P
  row[x_pixel] = abcde & (0x18 << BITS_ADDR_OFFSET); // mask out the bottom 3 bits which are the clk di bk inputs
#else
  row[x_pixel] = abcde;
#endif
} while (x_pixel != dma_buff.rowBits[row_idx]->width);
// color_index[0] (LSB) x_pixels must be "marked" with a previous's row address, 'cause  it is used to display
//  previous row while we pump in LSB's for a new row
abcde = ((ESP32_I2S_DMA_STORAGE_TYPE)row_idx - 1) << BITS_ADDR_OFFSET;
do {
  --x_pixel;
#ifdef SM5266P
  row[x_pixel] = abcde & (0x18 << BITS_ADDR_OFFSET); // mask out the bottom 3 bits which are the clk di bk inputs
#else
  row[x_pixel] = abcde;
#endif
} while (x_pixel);

// now time to add the clk di and bk bits to the end of the scan to update the row select shift register SM5266P
#ifdef SM5266P
uint16_t serialCount;
uint16_t latch;
x_pixel = dma_buff.rowBits[row_idx]->width - 16; // come back 8*2 pixels to allow for 8 writes
serialCount = 8;
do {
  serialCount--;
  latch = row[x_pixel] | (((((ESP32_I2S_DMA_STORAGE_TYPE)row_idx) % 8) == serialCount) << 1) << BITS_ADDR_OFFSET; // data on 'B'
  row[x_pixel++] = latch | (0x05 << BITS_ADDR_OFFSET); // clock high on 'A'and BK high for update
  row[x_pixel++] = latch | (0x04 << BITS_ADDR_OFFSET); // clock low on 'A'and BK high for update
} while (serialCount);
#endif

///////////////////////////////////////////////////////////////////////////////////////////////
//
// modifications end here for row shift register type SM5266P
//
///////////////////////////////////////////////////////////////////////////////////////////////

This modified code is invoked if SM5266P is defined at the start of this file.

#define SM5266P
#include <Arduino.h>
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"

I am unsure of the best way to include this as an option when setting the library up as the defines are not carried through to the library from the source code so any help there would be appreciated.

Cheers :)

donnersm, 2021-08-12:
If this code mod is working then this is definitely something that should be added to the library. With the SM5266P comes a whole lot of panels ...so worth looking into @mrfaptastic 
On a side note, i tried the code by modifying the ESP32-hub75-matrixpanel-i2s-dma.cpp but all i get is errors compiling ...double defined variables or different expectations of a functions..etc.
Maybe its an option for you to share the modified file completely? That way we can check side to side what needs to be changed.

mrfaptastic, 2021-08-12:
@DarrylStrong - Can you submit a pull request to the SM5266P tree, donnersm can test, and I might get my resident config expert @vortigont to look at the changes and see how it could be incorporated.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/SM5266P

vortigont, 2021-08-12:
Will be my pleasure to help here! I'll take a look at it this weekend or once confirmed that it all ready to go. That great if we all could keep this lib up to date and make it even better
Cheers!

DarrylStrong, 2021-08-13:
I am sorry - you will have to guide me through this process as I am not familiar with GitHub and especially Pull Requests. I have gone to Pull Requests and compared two branches but can't get any further.

mrfaptastic, 2021-08-13:
Not a problem. How about just attaching a zip of your full modified version of this library.

DarrylStrong, 2021-08-13:
Now that I can do :)

One file attached.

Cheers!
[ESP32-HUB75-MatrixPanel-I2S-DMA.zip](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/files/6981435/ESP32-HUB75-MatrixPanel-I2S-DMA.zip)

donnersm, 2021-08-13:
I tried using it by replacing the lib file with the modified file, however, it will not compile. In the end it still has some definition errors. Also the clearscreen function is double declared. don't know if Darryl added that function to the i2s-sma.cpp or that it is a library version conflict. I am using 2.0.5 ( according to arduino)

DarrylStrong, 2021-08-14:
I modified an older version of the library so definitely a version conflict.

Once I have working code for one of our signs I leave well alone 😊

mrfaptastic, 2021-08-16:
@donnersm - Can you try the SM5266P branch of this library? 

You can pass ` mxconfig.driver = HUB75_I2S_CFG::SM5266P;` on setup

mrfaptastic, 2021-08-16:
@vortigont - FYI

mrfaptastic, 2021-08-16:
@donnersm - Can you try the SM5266P branch of this library?  @vortigont - fyi.

You can pass ` mxconfig.driver = HUB75_I2S_CFG::SM5266P;` on setup

donnersm, 2021-08-16:
I tried the SM5266P Branch by deleting all the lib files and replacing them with the onces from the SM5266P repo..however, cannot compile..getting error on the mxconfig.driver = HUB75_I2S_CFG::SM5266P;

C:\Users\chord\AppData\Local\Temp\arduino_build_650897\sketch\LEDDRIVER.H:131:43: error: expected unqualified-id before ';' token
   mxconfig.driver = HUB75_I2S_CFG::SM5266P;     // in case that we use panels based on different chip, we can change that

donnersm, 2021-08-16:
> 
> 
> @donnersm - Can you try the SM5266P branch of this library? @vortigont - fyi.
> 
> You can pass ` mxconfig.driver = HUB75_I2S_CFG::SM5266P;` on setup

Tried it. It compiles and runs but result is the same, still double horizontals

DarrylStrong, 2021-08-16:
I shall try this tomorrow.

Donnersm, do you have a picture of the back of your pcb without the plastic frame? I took mine off to see the complete chipset.

My panel behaved according to the data sheet specs. for the 5266 driver.

The pcb was marked with ABC inputs only but it also had DE lines. ABC are clock data and strobe for the address latch and the DE lines select which of the 4 row drivers are active.

mrfaptastic, 2021-08-16:
> I hope we are not on a wild goose chase here and that i am dealing with a broken panel. 

I can't unfortuately test because I don't have one of these panels. You probably have yet another variant of the panels. You sure it isn't some 1/4 or 1/8 scan panel? These don't work with this library out of the box.

@DarrylStrong - Thanks for testing the branch of the library to see if my copy-paste of your 'SM5266P' code works.

DarrylStrong, 2021-08-17:
Glad to report all working well!

Thank you for incorporating it into the library.

Video here...

https://youtu.be/Mwpp63a9-Eo

Picture of the back of the module and my CPU board. 

[image]

Thank you again!

Darryl

closed 2021-08-17 by mrfaptastic

hugojgl, 2024-02-25:
> I can't unfortuately test because I don't have one of these panels. You probably have yet another variant of the panels. You sure it isn't some 1/4 or 1/8 scan panel? These don't work with this [...]

I have a 1/8 scan panel, is there no modification that works for it?

commit 11eaf34d (2021-08-16): Update to include SM5266P

Attempt to resolve #164