mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #165: change panel display size without restarting

tsctrl, 2021-08-14:
Hi @mrfaptastic ,

is there way to reinitialize display with different mx_width, mx_height config?

thanks!

mrfaptastic, 2021-08-14:
reboot the esp with different settings.
ESP.restart() or .reboot() - I forget which one it is.

mrfaptastic, 2021-08-14:
Ah I see you want it without reboot.

The answer is no. You need to configure this on boot.

vortigont, 2021-08-16:
I'm currently working on implementing this. Keep in touch :)

mrfaptastic, 2021-08-16:
@witnessmenow - FYI

witnessmenow, 2021-08-16:
Awesome! I've been working around at the moment but would be interesting to have it!

vortigont, 2021-08-17:
Actually it has been already implemented for most of the internal structs during v2 changes, only needed a proper destructor.
So the idea is that you simply destroy and recreate matrix object with new config. It works pretty fine (mostly). My FM6124 panels have problems with proper initialization but that is another issue.

```
// for reference
HUB75_I2S_CFG mxconfig ={
    .mx_width = 64,
    .mx_height = 32,
  };
MatrixPanel_I2S_DMA dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->fillScreenRGB888(255,0,0);
delay(1000);
delete dma_display;
dma_display = nullptr;
// switch to a larger matrixes :)
mxconfig.mx_height = 64;
mxconfig.chain = 2;
dma_display = new MatrixPanel_I2S_DMA(mxconfig);
dma_display->fillScreenRGB888(0,255,0);
```

P.S. Do not forget to do the same with VirtualMatrixPanel object if using combined display.

tsctrl, 2021-08-17:
hi @vortigont, 

how about the begin()? release and reallocate the memory?

thanks!

mrfaptastic, 2021-08-17:
> It works pretty fine (mostly). My FM6124 panels have problems with proper initialization but that is another issue.

Hahah! Amazing. What's more amazing is the DMA Engine hasn't been shut down and restarted as part of this, so I think it's probably terminating itself in error when it finds its allocated memory block disappears, but for whatever reason we don't see it on the output to the panel.

mrfaptastic, 2021-08-17:
> I've added a destructor that stops DMA and releases linked list chains, buffer structs already have their own destructors.

Ahhh ok. Great work!

tsctrl, 2021-08-17:
thanks @vortigont, @mrfaptastic

mrfaptastic, 2021-08-17:
Feel free to merge into the main tree @vortigont

vortigont, 2021-08-17:
Pushed to master. Testers are welcome )

P.S. Actually now I'm working on a nice testbed project using that same feature, hope it will make our life easier. Will share it soon.

closed 2021-08-17 by vortigont

vortigont, 2021-08-17:
>     2\. is it possible to reflect dynamic color depth bits to higher value on smaller display? possible to use dynamic heap_caps_malloc for different panel size?

currently color depth is a build-time option, not possible to change. But since we have properly destructable object now, I think it makes sense to have color depth changeable also someday.
Feel free to open a feature request.

tsctrl, 2021-08-18:
above is when i do check nullptr before deleting the object. when i remove the check i got different error as below:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
`PC: 0x400e0647: MatrixPanel_I2S_DMA::updateMatrixDMABuffer(unsigned char, unsigned char, unsigned char) at line 583`

tsctrl, 2021-08-18:
i guess this is because i have separate task that was driving the led and that is unloading the display at the same time

vortigont, 2021-08-18:
Looks like a use-after-free issue. Need to be careful not to access the pointer after destructor called.

commit 32a02a5f (2021-08-17): Add proper destructor to the class

Closes #165