mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #169: Screen flickering issue

tsctrl, 2021-08-17:
hi @mrfaptastic @vortigont ,

wondering if this related to the library but i notice high speed flickering. i havent go thru which part causing this but just to come out with some question regarding the double buffer and display synchronization. 

1. somecodes are running fast in a cycle and the next cycle will be a bit slow due to some function processing, probably due to font processing. how the frequency and the double buffer sync to each other when the loop speed is inconsistent? 
2. is there a wait to process clock or how to configure clocking latch delay per display feed. look like the switch buffer is slow to swap in my case. happens even no display changes where the swap cant keep up with the loop cycle or actually the previous swap came first before the function finish printing all details.
3. is there a way to confirm swap did full flush before running a new swap? 
4. is there a way to detect last clock, if same keep swap and if have changes do swap?

my guess is my code are not consistently running and swapping printing the display. but just to see how i could improve based on the library behavior or flow.

thanks!
ts

tsctrl, 2021-08-17:
https://user-images.githubusercontent.com/75555993/129705701-56410b9b-bc98-4082-93ba-4413e47069c9.mp4

sample video as above.

mrfaptastic, 2021-08-17:
I take it you have flipDMABuffer and showDMABuffer on a never ending loop right?

The speed and frequency that this occurs is completely managed by the ESP32 SoC and the library has no control over performance whatsoever. If you are using multiple perhiphals, WiFi, Bluetooth etc - all of these may generate interrupts which slow down the performance of the ESP32 flipping from one buffer to another.

I try avoid flipping buffers unless necessary. I suspect you might also be doing a clearscreen() somewhere which is causing flicker, because flipping buffers will NOT cause flicker unless one buffer is a black screen, and the other isn't.

mrfaptastic, 2021-08-17:
For example, this is the double buffer example without any delay.
Red goes into the background of one buffer
Blue into another

It'll flip so fast between buffers you should see a seamless yellow. So if you're experiencing flicker you'll need to look at your code.

```
#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

/* 
 * Below is an is the 'legacy' way of initialising the MatrixPanel_I2S_DMA class.
 * i.e. MATRIX_WIDTH and MATRIX_HEIGHT are modified by compile-time directives.
 * By default the library assumes a single 64x32 pixel panel is connected.
 *
 * Refer to the example '2_PatternPlasma' on the new / correct way to setup this library
 * for different resolutions / panel chain lengths within the sketch 'setup()'.
 * 
 */

MatrixPanel_I2S_DMA *display = nullptr;

const byte row0 = 2+0*10;
const byte row1 = 2+1*10;
const byte row2 = 2+2*10;

void setup() 
{
 
  // put your setup code here, to run once:
  delay(1000); Serial.begin(115200); delay(200);

  Serial.println("...Starting Display");
  
  HUB75_I2S_CFG mxconfig;
  mxconfig.double_buff = true; // Turn of double buffer

  // OK, now we can create our matrix object
  display = new MatrixPanel_I2S_DMA(mxconfig);  
  
  
  display->begin();  // setup display with pins as per defined in the library
  display->setTextColor(display->color565(255, 255, 255));  

  
  // Buffer 0 test
  display->fillScreen(display->color565(128, 0, 0));  
  display->setCursor(3, row0);
  display->print(F("Buffer 0"));
  display->setCursor(3, row1);
  display->print(F(" Buffer 0"));
  Serial.println("Wrote to to Buffer 0");
  display->showDMABuffer(); 
  delay(1500);
  
  // Buffer 1 test
  display->flipDMABuffer();
  display->fillScreen(display->color565(0, 128, 0)); // shouldn't see this
  display->setCursor(3, row0);
  display->print(F("Buffer 1"));
  display->setCursor(3, row2);
  display->print(F(" Buffer 1"));
 
  Serial.println("Wrote to to Buffer 1");
  display->showDMABuffer();  
  delay(1500);

}

void loop() {
  
  // Flip the back buffer
  display->flipDMABuffer();  
  display->showDMABuffer();  
  display->flipDMABuffer();   
  display->showDMABuffer();   

}

```

mrfaptastic, 2021-08-17:
Finally, changing stuff on the display using drawing routines (i.e. AdaFruit GFX) and flipping DMA buffers, are totally independant operations.

From my limited experience, both can seemingly occur at the same time on the ESP32. Perhaps RTOS allows one to change data *in a buffer* whilst another process will change what buffer the hardware DMA 'pumps out' to the GPIOs.

1. display->fillscreen(xxx)
2. display->flipdmabuff
3. display->showdmabuff

1 and 2&3 might could execute at the same time.

So you'll need to write your own syncronisation code, not sure how. Perhaps put a delay() after a big drawing routine just to make sure it's completed before flipping a buffer, or implementing a callback routing in whatever gfx code you have to then call the dma buffer flip operation.

This isn't an issue with this library.

vortigont, 2021-08-18:
Agreed, flickering comes when you clear the screen. If you check the examples for sinus wibble effect - it could run flickerless even without double buffer. The idea is that it updates buffer pixel by pixel slowly, not synced but w/o clearing the screen.

tsctrl, 2021-08-18:
hi @mrfaptastic delay is a bit expensive operation since the codes run in main thread and i need to sync the display up to 200ms to make sure clock updates are consistent. so i guess the lib does not control on race and concurrency.

i guess 
> The idea is that it updates buffer pixel by pixel slowly,
only possible to update when display are updated by part of the display that change. or it is kind of a shader/grads that lit the display gradually? we have this in the library?   

i remember that on other issue mentioning that clearScreen only now clearing the back buffer. supported by the comment here:
```
    /**
     * A wrapper to fill the entire Screen with black
     * if double buffering is used, than only back buffer is cleared
     */
    inline void clearScreen(){ updateMatrixDMABuffer(0,0,0); };
```
so it do effecting the display on the main display buffer? 
basic question, what happen to back buffer after the flip operation? safe to draw without clear the buffer using clearScreen()?

thanks!
TS

tsctrl, 2021-08-18:
update: removing clearScreen() confirm to fix the flickering issue, but the issue is my display are all now mixed up and not refreshing.

mrfaptastic, 2021-08-18:
Hi @tsctrl. I can't fault your persistence on trying to find a solution here  :-).

I've made a change to the library in the git - I have no idea if this will work...

```
loop(){
      display->flipDMABuffer();      // set active buffer to back buffer
      while(!display->clearScreen());  // wait for true to wipe the back buffer

  // do some graphic updates here..
  
  delay(1); // Do this anyway for sync, see if it makes a difference
  display->showDMABuffer();      

  }

```

No idea if it'll help

mrfaptastic, 2021-08-18:
This might be a dumb question, but you have definately enabled double buffering? because calling flip/showDMABuffer won't actually do anything otherwise, it'll just use the one buffer.

tsctrl, 2021-08-18:
yes, i have enable double buffering

tsctrl, 2021-08-18:
hi @mrfaptastic ,

should have this:
```
    bool clearScreen() {
      return display->clearScreen();
    }
```
here:
https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/3b13645e7ff4427694a8daa448e66d77882d5f92/ESP32-VirtualMatrixPanel-I2S-DMA.h#L82

mrfaptastic, 2021-08-18:
Can you just make the change locally and confirm if it helps or not with the flicker? If not i'll revert the changes I've made entirely.

tsctrl, 2021-08-18:
yup, the changes work with delay(1);
without delay is still flickering.

mrfaptastic, 2021-08-18:
So it's solved? I don't have any other solution as this isn't really an issue with the library per say.

tsctrl, 2021-08-18:
it solved, but hope to have other solution then adding delay(1) to the main loop though

thanks @mrfaptastic

mrfaptastic, 2021-08-18:
Unfortunately I'll have to leave it to you to find a plausable solution here as if a 1ms delay isn't acceptable then potentially you may need to do some larger refactoring.

try various values into delayMicroseconds() 

Perhaps just having any form of 'delay' helps the CPU/DMA/Thread/RTOS/dma flip get 'synced' - not sure. I'm going to remove the true/false return code from clearscreen from the library as that doesn't fix it.

mrfaptastic, 2021-08-18:
https://community.particle.io/t/delay-timer-in-function-without-delaying-main-loop/23789/2

Write code like this to check the system nanosecond time.... only flip to new buffer a few ms after completing GFX updates to buffer.

closed 2021-08-18 by mrfaptastic

tsctrl, 2021-08-19:
hi @mrfaptastic how to know GFX done updates to buffer?

tsctrl, 2021-08-19:
possible to expose one callback for LAT timing? not sure lat or oe

mrfaptastic, 2021-08-19:
nope. some callback on latch won't solve your issue

mrfaptastic, 2021-08-19:
re-write your sketch to use its own pixel buffer and when finished then iterate through every row and column and write to the dma matrix

mrfaptastic, 2021-08-19:
I've replaced the double buffer example with a better one, and we've made a change to the library and got rid of the 'showDMABuffer' function.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/SmoothDoubleBuffer

Not sure if it'll help