mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #172: Accidentally commented code

tsctrl, 2021-08-18:
hi @mrfaptastic 

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/ab641b964272d121c5f7db6212324f5bfbd1650b/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L418

look something not right there

mrfaptastic, 2021-08-18:
I want to remove the legacy begin function so had commented it out, but have uncommented it for now.

closed 2021-08-18 by mrfaptastic

tsctrl, 2021-08-18:
also this issue @mrfaptastic ,

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/ab641b964272d121c5f7db6212324f5bfbd1650b/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L376

undefined reference to `MatrixPanel_I2S_DMA::shiftDriver(HUB75_I2S_CFG const&)'

tsctrl, 2021-08-18:
sorry the tittle a bit misleading

tsctrl, 2021-08-18:
yup build with no issue, thanks @mrfaptastic