mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #173: Delay() in a lib code

vortigont, 2021-08-19:
@mrfaptastic, while working on [esp32-hub75-testbed](https://github.com/vortigont/esp32-hub75-testbed) project I've hit this peace of code in the lib, just wanted to discuss it.

https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/e12eaf15b5598703ff76ded68d5b575e64a7476a/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L506-L507

So the issue is it conflicts with Async Webserver - any call that has a delay() crashes when processing async web requests.
A delay() in the user code is not so good but could be dialed with, a delay() in a lib is a real pain that you have no so many options to take.

I'm not quite sure why that loop is required there - have not tested double-buffered setups too much, so any hints would be much appreciated.
For sure I can work around it my code somehow, but wanted to discuss the ideas on how to get rid of it at all?

mrfaptastic, 2021-08-19:
Hi @vortigont - Get rid of the delay. I can't remember why it's there. The `while(!i2s_parallel_is_previous_buffer_free());` itself should do the job anyway.

vortigont, 2021-08-19:
yea, maybe you right - just kill it in place and deal with what comes next later :) OK, I will give it a try then

tsctrl, 2021-08-19:
hi @vortigont maybe can try to reduce the esp wdt timings to minimum value so can make sure the free buffer without delay() will not trigger any wdt on large display. but the option is to increase the settings in esp rather than adding a delay. else is to disabled the task wdt temporarily

vortigont, 2021-08-19:
the issue is not with WD, buffer depletes pretty fast here. It's due to switch between SYS and CONT, async lib (as some others ISRs) does not like this. The best approach would be event driven programming, quite complex though.

closed 2021-08-19 by mrfaptastic