mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #176: FM6124 panel flicker

songxiaopeng1, 2021-08-24:
Hi：
  My English is poor, sorry...
  I have two kinds of panel ，ICN2028BP，FM6124，They're both 64 * 32。I use the ICN2028BP very well。But using FM6124 has a twinkle。

https://user-images.githubusercontent.com/38554894/130563733-f193b590-ac60-4840-9d47-fb87b3248d47.mp4

Although the definition is poor, dark shadows can be clearly seen in the middle of the panel

I'm running the  PatternPlasma  example and made the following changes：
`  HUB75_I2S_CFG mxconfig;
  mxconfig.mx_height = PANEL_HEIGHT;      // we have 64 pix heigh panels
  mxconfig.chain_length = PANELS_NUMBER;  // we have 2 panels chained
  //mxconfig.gpio.e = PIN_E;  
  mxconfig.latch_blanking = 4;
  mxconfig.clkphase = false; 
  mxconfig.driver = HUB75_I2S_CFG::FM6124;`

`#define PANEL_WIDTH 64
#define PANEL_HEIGHT 32  	// Panel height of 64 will required PIN_E to be defined.
#define PANELS_NUMBER 1 	// Number of chained panels, if just a single panel, obviously set to 1
#define PIN_E 32`

This code works very well for ICN2028BP，But FM6124 is flashing。I tried to change the parameters many times, but none of them worked, including  i2sspeed 、driver、clkphase...

And I added a capacitor and The connection wire is very short.

[image]

mrfaptastic, 2021-08-24:
`mxconfig.latch_blanking = 4;` will cause flicker if it is not required by the panels.

Try with `mxconfig.latch_blanking = 1;`

songxiaopeng1, 2021-08-24:
The problem is still，I checked the timing of icN2028 and FM6124 chips are exactly the same.so these two chips should be fully compatible,I feel very strange.Is my panel broken?

mrfaptastic, 2021-08-24:
> Is my panel broken?

Most probably if the other panel works perfectly and they are the same size etc.

songxiaopeng1, 2021-08-25:
Hello, I received another two fm6124 panels today, and I have conducted a test, and the three FM6124 panels are basically flashing in the same position. In this case, can we rule out the possibility that the panels are broken? In addition, I can basically rule out the problem of the power supply, because my power supply is 5V-60A, now I am very confused.Could you please give me a direction to solve the problem?Thank you very much.

https://user-images.githubusercontent.com/38554894/130725739-54b32be4-6d9d-4347-8e6d-c33019e66945.mp4

The lower panel is ICN2028, and the top three panels are FM6124, and they all flash in one place

vortigont, 2021-08-25:
Test it with solid color fill - red, green, blue, white, but not with video. Panels, this lib and ESP32 are not perfect, so there could be some glitches, does not mean that panels are defected.

mrfaptastic, 2021-08-25:
Might need a 3.3v to 5v logic level converter between the ESP32 and the first panel?

It could be anything. 

Also try the smooth buffer test example. 

Change the Mhz to 20Mhz as well. etc.

mrfaptastic, 2021-09-06:
The first panel looks OK in that video. It's an undefined behaviour having 4 panels chained, but the library's MATRIX_WIDTH set to only the width of one physical panel.

closed 2021-09-06 by mrfaptastic

mrfaptastic, 2021-09-10:
Try latest release as well. Please report back results. 
@songxiaopeng1