mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #178: Random crash: Cannot access memory at address XXX

tsctrl, 2021-08-29:
Hi @mrfaptastic , @vortigont 

i have to add back the delay for now on my end, error happens in random location mentioning unable to access memory at address as below:
error: Cannot access memory at address XXX
below are the example errors that was happens without the delay():

example error 1:
```
======================== THREADS INFO =========================
  Id   Target Id         Frame
* 1    process 1073667900 0x4000c59a in ?? ()
  2    process 1073486916 0x400dd3f0 in Adafruit_GFX::drawChar (this=0x3ffc0a94, x=-527, y=19, c=<optimized out>, color=63509, bg=<optimized out>, size_x=1 '\\001', size_y=1 '\\001') at ..libraries/Adafruit_GFX_Library/Adafruit_GFX.cpp:1225
  3    process 1073479328 0x4000bff0 in ?? ()
  4    process 1073492140 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) at esp32/crosscore_int.c:112
  5    process 1073480556 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) at esp32/crosscore_int.c:112
  6    process 1073488316 0x401c9aea in phy_rfcal_data_check (check_flag=0 '\\000', rf_cal_data=0x1 <error: Cannot access memory at address 0x1>, init_param=0x80089228 <error: Cannot access memory at address 0x80089228>) at phy_chip_v7.c:3305
  7    process 1073489716 0x401c9aea in phy_rfcal_data_check (check_flag=0 '\\000', rf_cal_data=0x0, init_param=0x80084a4a <error: Cannot access memory at address 0x80084a4a>) at phy_chip_v7.c:3305
  8    process 1073598880 0x40089956 in xQueueReceive (xQueue=0x3ffdc4cc, pvBuffer=0x3ffdd0dc, xTicksToWait=<optimized out>) athal/esp32/include/hal/cpu_ll.h:39
  9    process 1073649560 0x401cd688 in cpu_ll_get_core_id () at hal/esp32/include/hal/cpu_ll.h:35
  10   process 1073659616 0x401cd688 in cpu_ll_get_core_id () athal/esp32/include/hal/cpu_ll.h:35
  11   process 1073654588 0x401cd688 in cpu_ll_get_core_id () at hal/esp32/include/hal/cpu_ll.h:35
  12   process 1073588972 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) atesp32/crosscore_int.c:112
  13   process 1073586496 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) at esp32/crosscore_int.c:112
  14   process 1073603700 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) at esp32/crosscore_int.c:112
  15   process 1073608172 0x40084b00 in esp_crosscore_int_send_yield (core_id=0) at esp32/crosscore_int.c:112
```

example error 2:
```
==================== THREAD 1 (TCB: 0x3ffc1c44, name: 'main') =====================
#0  0x4000c46c in ?? ()
#1  0x400e2b5e in GifDecoder<192, 64, 12, true>::freeMalloc (this=0x3ffb6dbc <decoder>) at ..libraries/GifDecoder/src/GifDecoder_Impl.h:327

==================== CURRENT THREAD STACK =====================
#0  0x4000c46c in ?? ()
#1  0x400e2b5e in GifDecoder<192, 64, 12, true>::freeMalloc (this=0x3ffb6dbc <decoder>) at libraries/GifDecoder/src/GifDecoder_Impl.h:327
#2  0x400e3070 in loadGIF (name=0x1 <error: Cannot access memory at address 0x1>, duration=120) at ../src/animation.cpp:102
```

i guess memory allocation on same address are being use by the display, not sure, but this is what i have to report.

thanks!
TS

vortigont, 2021-08-29:
delay() issues is absolutely unrelated to I2S lib code since is does not allocate/release dynamic memory after begin() finished. You should dig through the rest of your/other lib code and check for congestion/use-after-free places.

vortigont, 2021-08-29:
> IRAM_ATTR as i disabled the isr to ram in config to preserve memory

bad idea also. Doing this you must be 100% sure your code never triggers ISR during SPI access routines, otherwise - crash. And I doubt you do since you are working with GIFs. My advise - never play with ram-saving features in the areas you have no confidence, it may lead you to the very dark areas.

tsctrl, 2021-08-29:
noted, 
i guess disabling the ISR in ram will impacting performance and not crash. the call will bypass ram and take from the ISR directly.

tsctrl, 2021-08-29:
`
By default, all functions are compiled into flash, which means that the cache may kick in for that code. However, the cache currently can’t be used during hardware interrupts. That means that, if you use a hardware ISR, such as attachInterrupt(gpio, myISR, CHANGE) for a GPIO change, the ISR must have the IRAM_ATTR attribute declared. Not only that, but the entire function tree called from the ISR must also have the IRAM_ATTR declared. Be aware that every function that has this attribute reduces available memory.

In addition, it is not possible to execute delay() or yield() from an ISR, or do blocking operations, or operations that disable the interrupts, e.g.: read a DHT.

Finally, an ISR has very high restrictions on timing for the executed code, meaning that executed code should not take longer than a very few microseconds. It is considered best practice to set a flag within the ISR, and then from within the loop() check and clear that flag, and execute code.
`
from: [here](https://arduino-esp8266.readthedocs.io/en/latest/faq/a02-my-esp-crashes.html) 

its mentioned ISR time critical, since the lib run at main thread. any delay will cause an issue. i also encountered crash at delay(100) on my other codes as well when the lib is using IRAM_ATTR

closed 2021-09-06 by mrfaptastic