mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #179: Double buffer now changed?

witnessmenow, 2021-09-13:
Hey, it seems the double buffer process has changed, and `showDMABuffer` has been removed, and looking at the double buffer example it doesn't look like anything replaces it, so I'm guessing the second buffer is shown as soon as `display->flipDMABuffer();` is called, is this correct?

One of my main use cases for the double buffer was to be able to draw whatever was needed to buffer while it was off screen and only cut over to it when it was ready, which doesn't seem possible anymore, is this right?

Pre-Submit edit:

[Seems like it automatically draws on the off-screen one](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/ESP32-HUB75-MatrixPanel-I2S-DMA.h#L510), I think that simplifies it a bit ok, but maybe some additional text in the example might be useful. I'll take a stab at it if you want?

mrfaptastic, 2021-09-13:
The redundant showDMAbuffer has been removed and the flipDMAbuffer does both. So when double buffering is enabled it'll write to the back buffer until you flip.

witnessmenow, 2021-09-13:
Ok sure, just tested there and it seems to work fine, just need to adjust the way I am using it.

```
void oldDrawFunction()
{
    display->flipDMABuffer();
    //draw something nice
    display->showDMABuffer();
}
```

```
void newDrawFunction()
{
    //draw something nice
    display->flipDMABuffer();
}
```

witnessmenow, 2021-09-13:
Maybe the old way was wrong before 😄

closed 2021-09-13 by witnessmenow

mrfaptastic, 2021-09-13:
The the way you were using it in the old method was correct. However it was easy to stuff up, and I often found myself prematurely flipping myself / getting confused and drawing to the active buffer.

So @vortigont suggested to simplify, and we did.