mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #183: The output data through I2S (LCD mode, 8-bit width), high and low 16bit is inverted. is this normal?

hyzgit, 2021-09-15:
For example, the data I'm going to send

uint8_t sendData[]={ 0x00,0x01,0x02,0x03,0x04,0x05,0x06,0x07}

But the order of actual output is

0x02,0x04,0x00,0x01,0x06,0x07,0x04,0x05

mrfaptastic, 2021-09-15:
Hello. I'm going to need more detail here.

You're using the DMA setup code for your own purposes right, to transmit data, not light up LEDs?

You're using 8 bit mode?

What ESP32 model are you using?

hyzgit, 2021-09-15:
ESP32 is ESP32 (not S2 or C3) with 8bit mode.

Also used to light up LED panels.

It is also used to light up LED panel, but I2S is only used to transmit data, excluding A,B,C,D,E and LAT. If these are also output through DAM, they will take up too much memory. I want to carry them in A larger size, such as 128*128 or larger

hyzgit, 2021-09-15:
I just tried it out and it's still a problem if you use 16bit

For example, the data I'm going to send

uint16_t sendData[]={ 0x0000,0x0001,0x0002,0x0003,0x0004,0x0005,0x0006,0x0007}

But the order of actual output is

0x0001,0x0000,0x0003,0x0002,0x0005,0x0004,0x0007,0x0006

mrfaptastic, 2021-09-15:
Can you try the old 1.2.4 branch in 16 bit mode and see if that still has an issue?

Fyi - A pull request was made litterally in the last couple of hours to implement a working 8 bit mode for this library. No ETA on when I'll integrate this though, refer to the other discussion with the .zip of the code attached.

hyzgit, 2021-09-15:
Ok, I'll try. Thank you

hyzgit, 2021-09-15:
#ifndef ESP32_S2
    // We need to update the correct uint16_t in the rowBitStruct array, that gets sent out in parallel
    // 16 bit parallel mode - Save the calculated value to the bitplane memory in reverse order to account for I2S Tx FIFO mode1 ordering
	// Irrelevant for ESP32-S2 the way the FIFO ordering works is different - refer to page 679 of S2 technical reference manual
    x_coord & 1U ? --x_coord : ++x_coord;
#endif 

I see this comment in your code, is ESP32 itself the reverse of high and low 16bit?

hyzgit, 2021-09-16:
> Can you try the old 1.2.4 branch in 16 bit mode and see if that still has an issue?
>
> Fyi - A pull request was made litterally in the last couple of hours to implement a working 8 bit mode for this [...]

I see this comment in your code, is ESP32 itself the reverse of high and low 16bit?

mrfaptastic, 2021-09-16:
Yes.. Esp32 hardware flips the order for solve stupid reason. So we flip before.

I take it you have built your own code and are asking me questions about how DMA works, if so, can you close this issue please?

hyzgit, 2021-09-16:
> Yes.. Esp32 hardware flips the order for some reason. So we flip to compensate.
> 
> I take it you have built your own code and are asking me questions about how DMA works, if so, can you close this issue please?

I see, thank you, can close, my display module driver IC is ICN2053, so it is a big change, need to know how the underlying DMA is implemented.

mrfaptastic, 2021-09-16:
Fair enough. But do we have a bug in this library we've never noticed? Are we actually sending out the bytes in parallel in the wrong order?

I would have thought there would be some display artifacts if we were flipping bytes wrongly or should have never done this in the first place. Can you confirm as I'm confused.

mrfaptastic, 2021-09-20:
I am closing this as I see there is no issue with this library.

closed 2021-09-20 by mrfaptastic