mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #186: External Latch Support (8 bit parallel mode)

mrfaptastic, 2021-09-22:
Tracking potential enhancement https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/pull/180 here.

Will need to think about this as the ESP32 (2017 edition) is horrible for TX FIFO byte reordering in 8 bit mode.

Much easier on S2 where bytes won't get messed around (FIFO is actually FIFO).

Will need to implement compiler conditional code depending on MCU type (WROOM, S2/3, C3/H2).

mrfaptastic, 2021-10-16:
The use of an 8-bit latch is only really of value if there is a need for:
* GPIO pins (currently this library uses about ~12, so you'll get 4 'free')
* A larger resolution display. Not sure how much memory is saved, but probably about 30%?

If this library is to implement latch support, the first thing that needs to be designed is a open source 'sheild' similar to 
https://github.com/pixelmatix/SmartMatrix/tree/master/extras/hardware/ESP32 with 8->16 bit shift registers / buffers.

Doesn't seem to be a real use case for a shield, as it seems most people just want to drive one or two panels in a chain.

mrweaver, 2021-11-03:
I'd definitely be interested in this.
My current project is fairly pin heavy and I ended up using the Teensy 4.1 and SmartMatrix for my prototype, but would prefer to use a ESP32 based approached.
I'll be keeping an eye on it, and may have a look at making a ESP32 board solution for this project if I get time (probably ESP32-WROOM-32E module or similar)