mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #195: P4 64x32 with ICN2037BP

AlfioSaitta, 2021-10-06:
After several attempts i was unable to get the library to work in any way with a P4 panel with ICN2037BP chip.

The panel in question is a 64x32, and with a quick check on the connections of the HUB75 connector, i noticed that D and E  pins are connected to GND, so i assume the chip only uses A, B and C for row selection.

The same panel, connected to an esp8266 with the PxMatrix library works "well".

My MatrixPanel-I2S-DMA code:

```
#include <Arduino.h>
#include <Wire.h>
#include <SPI.h>

#define MATRIX_WIDTH 64
#define MATRIX_HEIGHT 16
#define CHAIN_LENGTH 1
#define PIXEL_COLOR_DEPTH_BITS 8

#include <ESP32-HUB75-MatrixPanel-I2S-DMA.h>

#define R1_PIN_DEFAULT 25
#define G1_PIN_DEFAULT 26
#define B1_PIN_DEFAULT 27
#define R2_PIN_DEFAULT 14
#define G2_PIN_DEFAULT 12
#define B2_PIN_DEFAULT 13
#define A_PIN_DEFAULT 23
#define B_PIN_DEFAULT 19
#define C_PIN_DEFAULT 5
#define D_PIN_DEFAULT 17
#define E_PIN_DEFAULT 32
#define LAT_PIN_DEFAULT 4
#define OE_PIN_DEFAULT 15
#define CLK_PIN_DEFAULT 18

MatrixPanel_I2S_DMA *dma_display = nullptr;

void setup()
{
  Serial.begin(115200);

  HUB75_I2S_CFG::i2s_pins _pins = {
    R1_PIN_DEFAULT, 
    G1_PIN_DEFAULT, 
    B1_PIN_DEFAULT, 
    R2_PIN_DEFAULT, 
    G2_PIN_DEFAULT, 
    B2_PIN_DEFAULT, 
    A_PIN_DEFAULT, 
    B_PIN_DEFAULT,
    C_PIN_DEFAULT, 
    D_PIN_DEFAULT, 
    E_PIN_DEFAULT, 
    LAT_PIN_DEFAULT, 
    OE_PIN_DEFAULT, 
    CLK_PIN_DEFAULT
  };

  // Module configuration
  HUB75_I2S_CFG mxconfig(
      MATRIX_WIDTH, // module width
      MATRIX_HEIGHT, // module height
      CHAIN_LENGTH, // Chain length
      _pins);

  //mxconfig.driver = HUB75_I2S_CFG::FM6124;
  //mxconfig.double_buff = true;
  //mxconfig.clkphase = true;

  //mxconfig.mx_height = MATRIX_HEIGHT;
  //mxconfig.mx_width = MATRIX_WIDTH;

  // Display Setup
  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  
  dma_display->begin();
  dma_display->setBrightness8(10); //0-255
  dma_display->clearScreen();

  dma_display->drawLine(0, 0, 15, 0, dma_display->color444(0, 15, 0));
  dma_display->drawLine(0, 1, 31, 1, dma_display->color444(15, 0, 0));
  dma_display->drawLine(0, 2, 63, 2, dma_display->color444(0, 0, 15));
}

void loop()
{
  // ....
}
```

Result:
[image]

The serial log:
```
******************************************************************
We're going to need 8192 bytes of SRAM just for the frame buffer(s).
The total amount of DMA capable SRAM memory is 289720 bytes.
Largest DMA capable SRAM memory block is 126376 bytes.
******************************************************************
Malloc'ing 1024 bytes of memory @ address 1073448600d for frame row 0.
Malloc'ing 1024 bytes of memory @ address 1073449704d for frame row 1.
Malloc'ing 1024 bytes of memory @ address 1073450808d for frame row 2.
Malloc'ing 1024 bytes of memory @ address 1073451912d for frame row 3.
Malloc'ing 1024 bytes of memory @ address 1073422440d for frame row 4.
Malloc'ing 1024 bytes of memory @ address 1073423480d for frame row 5.
Malloc'ing 1024 bytes of memory @ address 1073424520d for frame row 6.
Malloc'ing 1024 bytes of memory @ address 1073425560d for frame row 7.
lsbMsbTransitionBit of 0 with 128 DMA descriptors per frame row, requires 12288 bytes RAM, 126376 available, leaving 114088 free: 
Raised lsbMsbTransitionBit to 0/7 to fit in remaining RAM
lsbMsbTransitionBit of 0 gives 76 Hz refresh: 
lsbMsbTransitionBit of 1 gives 152 Hz refresh: 
Raised lsbMsbTransitionBit to 1/7 to meet minimum refresh rate
Recalculated number of DMA descriptors per row: 64
Descriptors for lsbMsbTransitionBit of 1/7 with 8 frame rows require 6144 bytes of DMA RAM with 64 numDMAdescriptorsPerRow.
*** ESP32-HUB75-MatrixPanel-I2S-DMA: Memory Allocations Complete ***
Total memory that was reserved: 14 kB.
... of which was used for the DMA Linked List(s): 6 kB.
Heap Memory Available: 359984 bytes total. Largest free block: 126376 bytes.
General RAM Available: 274648 bytes total. Largest free block: 126376 bytes.
configureDMA(): Starting configuration of DMA engine.

Row 0 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 1 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 2 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 3 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 4 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 5 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 6 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
Row 7 DMA payload of 1024 bytes. DMA_MAX is 4092.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 2 is: 1.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 3 is: 2.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 4 is: 4.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 5 is: 8.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 6 is: 16.
configureDMA(): DMA Loops for PIXEL_COLOR_DEPTH_BITS 7 is: 32.
configureDMA(): Configured LL structure. 512 DMA Linked List descriptors populated.
Performing I2S setup:
configureDMA(): DMA setup completed on I2S_NUM_0.
```

DarrylStrong, 2021-10-06:
The panel you have is a 1/8 scan panel and therefore not supported by this library. (well not without some fudgery and bodgery)

Darryl

AlfioSaitta, 2021-10-06:
> The panel you have is a 1/8 scan panel and therefore not supported by this library. (well not without some fudgery and bodgery)
> 
> Darryl

No way to fix? If there really is no solution with this library, could you tell me another equally performing library?