mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #201: Led P5 outdoor drawPixel conflict position--> setCursor(0,0) but display (0,8)

macduyhai, 2021-11-02:
Code test:
  dma_display.setTextSize(1);     // size 1 == 8 pixels high
  dma_display.setTextWrap(false); // Don't wrap at end of line - will do ourselves

  dma_display.setCursor(0, 0);   // start at top left, with 8 pixel of spacing

  uint8_t w = 0;
  char *str = "ESP32 DMA 123456789123456789123456789123456789123456789123456789";
  Serial.println(strlen(str));
  for (w = 0; w < strlen(str); w++) {
    dma_display.setTextColor(dma_display.color565(255, 0, 255));
    dma_display.print(str[w]);

  }
  
  --> Start character start in (0,8) --> conflict position
  
  
  
[image]

mrfaptastic, 2021-11-02:
What's the actual issue?

macduyhai, 2021-11-02:
Im setting cursor at pos (0,0) but when i run this code i see character begin at pos(0,8).
(0,0);(0,8) output the same

macduyhai, 2021-11-02:
> What's the actual issue? Does a drawPixel (0,8) light up the correct LED?

drawPixel (0,0) light up the incorrect LED 
Code: dma_display.drawPixel(0, 0, dma_display.color444(15, 0, 0));
Look at this picture (draw a pixel ,red led) 

[image]

macduyhai, 2021-11-03:
I run example testshapes_32x64 . Results showing errors
[image]

mrfaptastic, 2021-11-03:
Looks like a 1/8 scan panel right which isn't supported.

Post technical details of the panel as well as your full sketch. May need to use that other 1/8 scan example that hasn't been fully confirmed as working.

macduyhai, 2021-11-03:
> Looks like a 1/8 scan panel right which isn't supported.
> 
> Post technical details of the panel as well as your full sketch. May need to use that other 1/8 scan example that hasn't been fully confirmed as working.

My panel :
Led maxtrix P5 full color outdoor 
IC : ICN2037BP
1/8 Scan

How  can I do for panel run correct  ? Thanks

mrfaptastic, 2021-11-03:
Refer to and read: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/154

mrfaptastic, 2021-11-25:
Try new example: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/One_Eight_1_8_ScanPanel

closed 2021-12-13 by mrfaptastic