mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #207: Displaying a image-sequence without memory issues

kzka90, 2021-11-25:
Hello everyone. 😀

I've succesfully run and used this library to display the examples, everything went perfect. Now I'm trying to do something more complex.

I've setup a WiFiClient to download pixel data from a server to transform it into an image, it works perfect when I display the pixel-data immediately. My data strcuture looks as follow:

`0x00 0x00 0x10 0x30 0x10 0x50 0x00 0x00 0x10 0x30 0x10 0x50 ...`

Which corresponds to:

`R G B R G B R G B R G B R G B R G B R G B R G B R G B R G B R G B R G B ...`

Where every **three bytes** of the array it's the color of the X,Y pixel. In my case my panel is **128x64**, so my byte array is `24,576` long (128x64x3)

Would this be the most efficient way to build a 128x64 RGB image? (In my case from the Internet)

I'm asking this because, I haven't been able to store this byte array in RAM, I've only been able to put it right away from my `client.read()` loop.

Why do I want to store it in an array? Because I want to download multiple images from my server and display them one after the other, so I must store them in some any way.

I've seen on the **README** there's a pretty smooth image-sequence video:

[image]

How has this been achieve without having memory issues?

I'd appreciate if you could guide me.

Is it even possible to handle such big arrays on RAM? Or would it be better to store my data on flash and read it every X milliseconds?  Wouldn't read it from Flash every time you want to display an image make the image-sequence look choppy?

Thank you in advance.

mrfaptastic, 2021-11-25:
This type of network thing has been done lots of times before.

WiFi display example:

https://github.com/NeoCat/ESP32-P3RGB64x32MatrixPanel/

https://github.com/mrfaptastic/Glediator3

mrfaptastic, 2021-11-25:
> How has this been achieve without having memory issues?

Buy an ESP32 with a few MB of 'PSRAM' (aka. add-on memory). Note: You CAN NOT use PSRAM for the DMA buffer. So an ESP32 is limited in what resolution it can display to the SRAM.

WiFi -> Buffer in PSRAM -> Pixel, by pixel update of DMA memory in SRAM (what gets sent out to the physical hardware)

closed 2021-11-25 by mrfaptastic