mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #209: Adafruit BusIO 1.9.7 breaks ESP32-HUB75-MatrixPanel-I2S-DMA for ESP32

han-fastolfe, 2021-11-25:
If you try to build fro ESP32 with the latest Adafruit BusIO lib version 1.9.7, you'll get an error that looks like this:

```
rocessing wemos_d1_mini32 (platform: espressif32; board: wemos_d1_mini32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/wemos_d1_mini32.html
PLATFORM: Espressif 32 (3.4.0) > WeMos D1 MINI ESP32
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <ESP32 HUB75 LED MATRIX PANEL DMA Display> 2.0.6
|   |-- <Adafruit GFX Library> 1.10.12
|   |   |-- <Adafruit BusIO> 1.9.7
|   |   |   |-- <Wire> 1.0.1
|   |   |   |-- <SPI> 1.0
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <FastLED> 3.4.0
|   |   |-- <SPI> 1.0
|-- <Adafruit BusIO> 1.9.7
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <Wire> 1.0.1
|-- <Adafruit GFX Library> 1.10.12
|   |-- <Adafruit BusIO> 1.9.7
|   |   |-- <Wire> 1.0.1
|   |   |-- <SPI> 1.0
|   |-- <Wire> 1.0.1
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/FastLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/bitswap.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorpalettes.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/colorutils.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/hsv2rgb.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/lib8tion.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/noise.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/platforms/esp/32/clockless_rmt_esp32.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/power_mgt.cpp.o
Compiling .pio/build/wemos_d1_mini32/libdef/FastLED/wiring.cpp.o
Compiling .pio/build/wemos_d1_mini32/src/main.cpp.o
Generating partitions .pio/build/wemos_d1_mini32/partitions.bin
Compiling .pio/build/wemos_d1_mini32/lib2ea/Wire/Wire.cpp.o
Compiling .pio/build/wemos_d1_mini32/libcd9/SPI/SPI.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_BusIO_Register.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_SPIDevice.cpp.o
Archiving .pio/build/wemos_d1_mini32/libcd9/libSPI.a
Indexing .pio/build/wemos_d1_mini32/libcd9/libSPI.a
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp: In static member function 'static void ESP32RMTController::interruptHandler(void*)':
/home/cory/Development/Arduino/sketchbook/libraries/FastLED/src/platforms/esp/32/clockless_rmt_esp32.cpp:318:10: warning: unused variable 'stuff_to_do' [-Wunused-variable]
     bool stuff_to_do = false;
          ^
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GFX.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_GrayOLED.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/Adafruit_SPITFT.cpp.o
Compiling .pio/build/wemos_d1_mini32/lib4cd/Adafruit GFX Library/glcdfont.c.o
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp: In member function 'void Adafruit_I2CDevice::end()':
.pio/libdeps/wemos_d1_mini32/Adafruit BusIO/Adafruit_I2CDevice.cpp:44:10: error: 'class TwoWire' has no member named 'end'
   _wire->end();
          ^
Compiling .pio/build/wemos_d1_mini32/lib5ba/ESP32 HUB75 LED MATRIX PANEL DMA Display/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp.o
*** [.pio/build/wemos_d1_mini32/lib53d/Adafruit BusIO/Adafruit_I2CDevice.cpp.o] Error 1
========================================================== [FAILED] Took 2.28 seconds ==========================================================
The terminal process "platformio 'run'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
```
The change in `Adafruit BusIO/Adafruit_I2CDevice.cpp` for 1.9.7 relies on a `#define ESP8266` which is not there for ESP32.

mrfaptastic, 2021-11-25:
Please raise the issue with the Adafruit project, or simply use and earlier version of that library.

solhuebner, 2021-11-26:
PlatformIO just implemented IDF 4.3.1 (https://github.com/platformio/platform-espressif32/releases/tag/v3.4.0) but Arduino-ESP32 is based on 4.4 (https://github.com/espressif/arduino-esp32/releases/tag/2.0.1).

So you would either need to wait or do the following in your platformio.ini:
change platform to: `platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream`
and add:
```
platform_packages =
    framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
```
This is how mine looks like:
[image]

Ideally run a "Clean All" afterwards:
[image]

han-fastolfe, 2021-11-27:
Adafruit has fixed it on their end with 1.9.8, however, @solhuebner has detailed [a fix for PlatformIO](https://github.com/platformio/platform-espressif32/issues/680#issuecomment-980577179) to get the latest Esp32-Arduino 2.0.1 framework instead of the 1.0.6 that ships with a PlatformIO/VSCode install.

For anybody using ESP32-HUB75-MatrixPanel-I2S-DMA in PlatformIO, this will fix the broken build issue.

closed 2022-02-06 by mrfaptastic