mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #213: [Suggestion] Default PIO Project/Config?

ModischFabrications, 2021-12-09:
I was unable to quickstart a project with PIO, even after looking around in the Readme and attached examples. I managed to get it running in the end, but what started as a quick Test wioth this (awesome) library turned into a 1 hour debugging session. 

I tried combining the singular [PIO example](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/PIO_TestPatterns) with the simple [pattern check](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/1_SimpleTestShapes), but it seems like PIO is unable to find the dependencies of this library natively and is therefore unable to build. I had to check out the library and look into the imports to find out that a manual import of GFX/FastLED is necessary. 

I think a generic default project or at leat a default platformio.ini could greatly improve onboarding experience. My initial findings can be found at https://github.com/ModischFabrications/HUB75-DMA-Test, but I haven't grasped what is needed and what isn't yet, so take it with a grain of salt. A FAQ entry for "missing definition for matrix->setText" or similar could also help newcomers.

mrfaptastic, 2021-12-20:
Thanks for this suggestion, @vortigont has implemented something here: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/tree/master/examples/PIO_TestPatterns

closed 2021-12-20 by mrfaptastic

vortigont, 2021-12-20:
@ModischFabrications you can find another example in this repo [repo](https://github.com/vortigont/esp32-hub75-testbed)
Most ppl still uses Arduino IDE, so the examples are still .ino based :)
Introducing CI job (#222 ) for library build testing based on PIO ci feature might be a good start to switch one or two more examples to PIO format. At least one including Adafruit GFX text/primitives methods would be nice. Will try to look into this any time soon.
Cheers!

ModischFabrications, 2021-12-21:
> has implemented something

That's exactly the example I linked in my statement, but thanks for looking into it. 

I can't fully agree that most people use Arduino IDE, especially not for complex systems including this library, but it's not my decision to make. No hard feelings on my side, feel free to ignore this issue. 

> another example

That's helpful, thanks for the additional example. I'm going to copy some of these patterns.

mrfaptastic, 2021-12-21:
The library started off as an Arduino library before PlatformIO was even useful, but it makes sense to create some a few more PIO examples. I'll look to copy paste your example into our examples folder.

However, in PIO you can import an Arduino sketch as well I think.