mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #217: The color of the top half of the screen is not saturated enough. 64x64Screen

FettLuo, 2021-12-12:
Porting from Arduino to esp32 IDF without Adafruit GFX or FastLED.
E_PIN_DEFAULT set to GPIO18. other default.
Matrix Screen is 64x64.
The program draws some random pixels.

**Issue:**
The color of the top half of the screen is not saturated enough, the bottom half is normal.
Unplugging and inserting R1G1B1 will affect the color indicating that the wiring is correct.

[image]

mrfaptastic, 2021-12-13:
Looks like a bad panel to me. Check your writing.

Worse comes to worse you'll have to write some colour adjustment hack in your software for any pixels in the bottom half of the panel.

FettLuo, 2021-12-14:
> Looks like a bad panel to me. Check your writing.
> 
> Worse comes to worse you'll have to write some colour adjustment hack in your software for any pixels in the bottom half of the panel.

Call fillScreen to fill pure white with no problem.

Swapped R1G1B1 and R2G2B2, it exchanged the upper and lower states, indicating that the I2S code distinguishes between upper and lower screen processing?

I compared the brightness of the R, G and B pins separately, it seems that blue is normal and red and green are muted.

mrfaptastic, 2021-12-14:
This isn't an issue of the library, there is no special treatement between upper and lower halves of the panel - it's the same digital signal.

Either the wiring/wires that is used between halves is different, there's some power related issue with your setup, or perhaps, even some weird issue with one of the GPIO's being used. Try different pins and shorter wires.

FettLuo, 2021-12-14:
> This isn't an issue of the library, there is no special treatement between upper and lower halves of the panel - it's the same digital signal except for a different set of GPIOs.
> 
> Either the wiring/wires that is used between halves is different, there's some power related issue with your setup, or perhaps, even some weird issue with one of the GPIO's being used. Try different pins and shorter wires.
> 
> Also, check the panel type and see if there's some setup routine you're missing. Clock phase etc. FM6XXXX etc.

If it's a wire problem, why is the white color saturated!
I tried changing the driver (cfg.driver) and tried each one with the same result.
Another I2S DMA library (SmartMatrix) uses a similar technique to drive the screen and the colors are normal. It just crashes a lot when it's in use, so I switched to try here.

mrfaptastic, 2021-12-14:
> If it's a wire problem, why is the white color saturated!

Who knows. Do you have another panel to test with? Can't really identify the root cause otherwise.

FettLuo, 2021-12-14:
> > If it's a wire problem, why is the white color saturated!
> 
> Who knows. Do you have another panel to test with? Can't really identify the root cause otherwise.
> 
> There are a lot of faulty panels being sold these days with weird issues. I guess chinese sellers are selling the rejects because of the 'chip shortage'.

Not so much yet, these chips should be low-end in which country, there is no shortage.

FettLuo, 2021-12-14:
Thanks guys, the problem is solved.

GPIO25 and GPIO26 of ESP32-WROOM-32 (4M) are not suitable for use and will cause insufficient voltage.

Change to GPIO0, GPIO2 to solve:
```
    #define R1_PIN_DEFAULT 0
    #define G1_PIN_DEFAULT 2
```

closed 2021-12-14 by FettLuo