mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #223: Using the dev version!

MythicalForce, 2021-12-20:
i was trying to use the dev version by doing this in the ini

```
lib_deps =
    ${env.lib_deps}
;   use dev version of the lib
    https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA.git
```
but only 2 of my 64x64 matrix display would work under the dev version :S

using the normal one in the ini all 4 panels work :S

```
lib_deps = fastled/FastLED
	       mrfaptastic/ESP32 HUB75 LED MATRIX PANEL DMA Display
	       adafruit/Adafruit GFX Library
	       adafruit/Adafruit BusIO
	       bitbank2/AnimatedGIF
```

this is my setup

```
#define PANEL_RES_X 64 // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64 // Number of pixels tall of each INDIVIDUAL panel module.

#define NUM_ROWS 2 // Number of rows of chained INDIVIDUAL PANELS
#define NUM_COLS 2 // Number of INDIVIDUAL PANELS per ROW
#define PANEL_CHAIN NUM_ROWS*NUM_COLS    // total number of panels chained one to another

// Change this to your needs, for details on VirtualPanel pls see ChainedPanels example
#define SERPENT true // Should be false
#define TOPDOWN false // Should be true

// Virtual Panl dimensions - our combined panel would be a square 4x4 modules with a combined resolution of 128x128 pixels
#define VPANEL_W PANEL_RES_X*NUM_COLS // Kosso: All Pattern files have had the MATRIX_WIDTH and MATRIX_HEIGHT replaced by these.
#define VPANEL_H PANEL_RES_Y*NUM_ROWS //

// placeholder for the matrix object
MatrixPanel_I2S_DMA *matrix = nullptr;

// placeholder for the virtual display object
VirtualMatrixPanel  *virtualDisp = nullptr;

namespace MatrixConfig
{
    void begin()
    {
        HUB75_I2S_CFG mxconfig(PANEL_RES_X, PANEL_RES_Y, PANEL_CHAIN);

        mxconfig.gpio.e = 18; // Must be pin 18 for 64x64 else -1

        matrix = new MatrixPanel_I2S_DMA(mxconfig);

        matrix->setBrightness8(127);

        if( not matrix->begin() )
        {
            Serial.println("****** !KABOOM! I2S memory allocation failed ***********");
        }

        virtualDisp = new VirtualMatrixPanel((*matrix), NUM_ROWS, NUM_COLS, PANEL_RES_X, PANEL_RES_Y, SERPENT, TOPDOWN);
    }
};
```

mrfaptastic, 2021-12-20:
Can you post a photo? So you have four of these chained together in a 2x2 grid?

I take if if you don't use the VirtualDisplay class, you can address all four panels as if its a 64x256 pixel horizontal chain (just want to confirm that the core library isn't busted)?

mrfaptastic, 2021-12-20:
Actaully, try the dev version again now, I think I found the bug

MythicalForce, 2021-12-21:
> Actaully, try the dev version again now, I think I found the bug

yes i have 4 x 64x64 in a 2x2 grid

just tried it again using the VirtualDisplay class and the normal Matrix Class but still only 2 panels working :S

mrfaptastic, 2021-12-21:
Can you double check your wiring as you should get output in all four panels when NOT using the virtual display class, and just using the underlying dma_display configured for a 256x64 resolution horizonal screen.

MythicalForce, 2021-12-21:
the ribbon cable is connected correctly as it works with the stabel version just fine and power is not  an issue either as im running them all on a 60A PSU with good thick power wires!

you mean i have to test all 4 in a layed out "grid"? so all 4 side by side and not in a 2x2?

mrfaptastic, 2021-12-21:
Can you please run the 'SimpleTestShapes' example: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/1_SimpleTestShapes/1_SimpleTestShapes.ino

With:
```
#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1     // Total number of panels chained one to another
```
All that needs to happen is all panels to show something (the same thing probably).

Then try it again with:
```
#define PANEL_RES_X 64      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 64     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 4     // Total number of panels chained one to another
```

closed 2022-03-09 by mrfaptastic