mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #229: ESP32-HUB75-MatrixPanel-I2S-DMA.cpp:771:11: error: this 'if' clause does not guard...

solhuebner, 2022-01-02:
```
        if(coloridx && coloridx <= lsbMsbTransitionBit) {
          // divide brightness in half for each bit below lsbMsbTransitionBit
          int lsbBrightness = brt >> (lsbMsbTransitionBit - coloridx + 1);
          if((x_coord) >= lsbBrightness)
            row[x_coord] |= BIT_OE;  // Disable output after this point.
            continue;
        }
```

Where does the **continue;** belong to?
- if(coloridx && coloridx <= lsbMsbTransitionBit)
- if((x_coord) >= lsbBrightness)

uklooney, 2022-01-02:
do {
        --x_coord;

solhuebner, 2022-01-02:
> 751 do { 752 --x_coord;

Maybe I should rephrase the question. Which version is the correct one? (which if statement should the continue belong to?) 

A:
```
if(coloridx && coloridx <= lsbMsbTransitionBit) {
          // divide brightness in half for each bit below lsbMsbTransitionBit
          int lsbBrightness = brt >> (lsbMsbTransitionBit - coloridx + 1);
          if((x_coord) >= lsbBrightness) {
            row[x_coord] |= BIT_OE;  // Disable output after this point.
            continue;
	  } 
        }
```

B:
```
if(coloridx && coloridx <= lsbMsbTransitionBit) {
          // divide brightness in half for each bit below lsbMsbTransitionBit
          int lsbBrightness = brt >> (lsbMsbTransitionBit - coloridx + 1);
          if((x_coord) >= lsbBrightness) {
            row[x_coord] |= BIT_OE;  // Disable output after this point.
	  } 

	  continue;
        }
```

mrfaptastic, 2022-01-03:
I've fixed it in the git repo. Should have been A

closed 2022-01-03 by mrfaptastic