mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #232: Add S-PWM

daveythacher, 2022-01-09:
I tried this out on the Pi, however I pushed it under a different account. Here it is: https://github.com/greatballoflazers/rpi-rgb-led-matrix (I did not want to be associated directly with that project.)

Anyhow this is a reduced trick called S-PWM. What it does is increases the refresh rate and possibly the quality or size of the display. Normally this is done with PWM, which is problematic. I was able to scale it back with BCM and remove the color redistribution. Which results in no extra CPU or memory usage. (Some increase is possible in interrupts or state machine logic; however, you can configure that to your liking.)

There are three version of the S-PWM implementation in contrast to traditional PWM which has two. Meaning there is a total of five implementations, but two of them are not practical. The Pi may struggle a little bit in terms of determinism but will have more CPU and memory available compared to ESP32. For the time being I think only use the traditional PWM/S-PWM hybrid with only BCM is recommended.

A rough outline is possible as shown in that repo. I did not create the other two versions in that repo. I have some disagreements with Henner's rpi-rgb-led-matrix project, and I am not willing to disclose the others to them at this time. Note I do have some ideas for the memory mapped drivers, but I do not have a ESP32.

I did test this out on the Pi and it does improve a few things. However, I did not do extreme testing as it seemed a little pointless on that platform to some extent.

mrfaptastic, 2022-01-09:
I don't understand. This library has no semblance to the RPI library under the hood and PWM is completely irrelevant to this library.

The RPI library essentially bit-bangs the GPIO output using the CPU.

daveythacher, 2022-01-09:
It is based on traditional PWM using BCM right? Single period per row. Do you use a timer or DMA to keep track of the BCM bit-planes? As long as you are doing traditional PWM with or without BCM you can use this. Basically, you just divide the single period per row into multiple periods per row.

Under traditional PWM the refresh is wasted somewhat. You have really high frame rate, which unless needed could be used to increase chain length or increase color depth. However, in some cases this could also be bad. However, in that case it could be used to lower the serial clock rate required which can be useful in low voltage, long chains and slower power supplies.

This can lead to increase in interrupts with could be fixed with multicore, scatter gather, DMA chaining, etc. This trick optimizes the CPU and memory consumption for microcontrollers. Microcontrollers are generally based on SRAM and have higher levels of determinism than say the Pi. The Pi has to play rounding tricks in the bit-bang but has more memory and processing power. Ironically this means the full version is not likely very useful to either platform. Hardware like FPGA could possible do it, but it is likely more expensive to do so. Would be cheaper to use MM LED driver.

MM LED drivers are beast of themselves. Two or three approaches possible there for the most part. However, these are great for two different application areas, high density and low multiplex with high contrast.  Standard panels can accomplish more overlap using S-PWM. It will not be able to compete directly, but it can close the window some. To the point it may not matter. Handling the 16 bit per pixel is hard for many microcontrollers, resulting in memory shortage or complex software models. ESP32's multicore does provide a break from this. DMA and scatter gather may also provide options.

mrfaptastic, 2022-01-09:
I don't really understand what the proposal is to be honest.

With this library we construct a 16 bit (2 byte) wide binary blob bitstream to send out the GPIO's in parallel at the clock-speed the I2S hardware provides. There's no use of PWM hardware or anything else - whatever is in this DMA memory blob allocation is spat out repeatedly as quickly as possible.

So the concept of S-PWM isn't applicable to this library or hardware, and in any case, I think what we have is pretty good for what it does and I don't wish to complicate the code any further.

daveythacher, 2022-01-09:
From a configuration standpoint there is an increase in complexity. However, there is also an increase in flexibility.

What I am proposing is to allow scaling the frame rate on the serial bus to be as low as 1, without losing any refresh. In fact, I am proposing we allow refresh above 1kHz. The way this is done is perceptive refresh scaling. Basically, we divide the period into sections of 64, which for a 30FPS serial bus would yield a multiplexing refresh of 1920Hz. 

Now this is not for free. With traditional PWM you have to do 2^x shifts/interrupts per row times frame rate (refresh). With traditional PWM using BCM you can do 2^x or x shifts/interrupts per row times frame rate (refresh). With the scaled back version of S-PWM that I am proposing you would have 2^x or 2^upper * lower shifts per row times frame rate. Now the refresh becomes 2^upper times frame rate. No longer does the frame have to at least 100Hz for decent multiplexing, which means more serial bus time.

From the programming side it is fairly straight forward. However, the user may experience additional confusion. Overall, this can be very powerful. Again, this is not full S-PWM. This trick does not do PWM, it does BCM. However, it is a little more intense.

The core logic you have will not change. You will simply modify the state machine to change the rows a little bit more frequently. DMA should not care. It will still push data out from memory buffer. It will duplicate shifts, assuming such duplication is not already done to keep track of time via DMA. The state machine will run more, which can lead to overhead. However, the multiplier for refresh is also the interrupt multiplier. So, there is a catch.

Going back to my example a 16x32 matrix if ran at 25MHz could have 10-bit color depth with 95 Hz refresh. This is what traditional PWM gives you. However, with S-PWM you could do 11-bit color depth (6-bit upper and 5-bit lower) with 1920 Hz refresh using 15.8MHz. This is possibly since we moved the 95 FPS to 30 FPS, which means a difference of 3.17. If we increase the color depth by one bit this becomes 1.59. If you divide 25 by 1.59 you get 15.8.

I can get really crazy with it too if I play with the ratio of upper to lower. Note I would only really get 11-bit color depth if the panel had low multiplex and fairly fast PSU. However, the lower clock will help with this on standard panels. (The other way to help would be to increase the chain length which would require more number tinkering.)

The comes in handy in certain places. 24bpp is nearly impossible on 16x128 with traditional PWM above 95Hz refresh on single connector. Assuming 25MHz is even stable or possible. You can do time averaging tricks like what the Pi does to average in 8-bit planes.

In a personal project on POE. I could do at most 15-21bpp for 16x128. (LED current is limitation.) If I used S-PWM the refresh would up to 960-3840 Hz. With a serial clock of around 1-4MHz. Assuming my power supply could handle 7.7-30.8 kHz response. (This should not be hard to accomplish.) Note this is on single chain with 30FPS. Max interrupt rate would be 32-128 times 8 per frame. Meaning per second my interrupt rate would be my PSU response rate.

Just a suggestion.

mrfaptastic, 2022-01-09:
Very interesting. I think I understand, but we're not in the same page here 

For silicon hardware and more importantly, memory reasons, this can't be done in an ESP32.