mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #238: VirtualMatrixPanel is missing fillScreenRGB888()

vortigont, 2022-01-10:
Found during CI tests, looks like it's missing in a VirtualMatrixPanel class. Have to be fixed someday.
So as Fast functions, but this would be a bit trickier to fix. Maybe we should at least stub it with a meaningful compiler message instead of just throwing no such func error.

solhuebner, 2022-01-10:
@vortigont the PR should fix the missing function but the fast DMA stuff looks a bit like Voodoo to me atm ;)

vortigont, 2022-01-11:
Tnx, @solhuebner!
Yeah... it would require some coordinates transformation magic to implement fast functions for virtual canvas. Let it be another todo item in a list :)

mrfaptastic, 2022-01-13:
Thanks for this addition @solhuebner.

@vortigont - I don't think we should implement fast functions for this virtual canvas at the DMA level. It's too complex, it'll be messy and it breaks the abstraction we currently have. 

ESP32-HUB75-MatrixPanel-I2S-DMA = 1/16 Scan for a Horizontal Chain as fast as possible
ESP32-VirtualMatrixPanel-I2S-DMA = All the other panel specific junk.

commit 5b54b56a (2022-01-13): Added missing fillScreenRGB888 to VirtualMatrixPanel

Fixes #238
closed 2022-01-13 by mrfaptastic