mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #245: Scan issue

tsctrl, 2022-01-13:
hi @mrfaptastic , @vortigont 

this is not library issue. sorry as i am asking this here
i have issue on the scan lines pattern shown as below:

[image]

could you guess what potential the culprit for this? is it a a,b,c,d lines might be not properly connected. or lat oe or clock issue or something else. 

your experience on this and suggestion to investigate are really appreciated.
thanks at the earliest!

mrfaptastic, 2022-01-13:
Looks like an address line issue. I assume the image that you were trying to display looks fine for the lines that do show right?

tsctrl, 2022-01-13:
yes, issue on horizontal lines

tsctrl, 2022-01-13:
tested all address lines are all connected and have 1.6v from gpio

mrfaptastic, 2022-01-13:
Yes. I mean abcde lines.

These panels are supposed to have 5v input. They work with 3.3v from the ESP 32 for a few chained panels... But 1.6? Way too low

tsctrl, 2022-01-14:
yup, thats the reading i got from gpio on address pin referenced to main ground. i have fix the issue by replacing the esp32 mcu. not sure is a chip defect or bad soldering.

thanks @mrfap!

closed 2022-01-14 by tsctrl