mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #246: Random Pixel on Master

tsctrl, 2022-01-15:
Hi @mrfaptastic,

i am having random pixel ghosting on p4 panels. latch blanking 1-4 did not fix this issue.

[image]

i could not replicate this issue on arduino with simple test shape example with 2.0.6 library.

thank you

mrfaptastic, 2022-01-15:
Change the 'clkphase' value?

tsctrl, 2022-01-15:
changing 'clkphase' to true will mess up the display. become unreadable text

mrfaptastic, 2022-01-17:
Does reducing the brightness a bit, help?

See what happens if you remove the 'continue' statements (two of them) in the library here: https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/c682a7e5e313d7db66eec589273cb64dc6310748/ESP32-HUB75-MatrixPanel-I2S-DMA.cpp#L768

Not sure if a bug has been introduced or not.

tsctrl, 2022-01-18:
Hi @mrfaptastic ,

i did the update to latest master today with some issue on my end:
[image]

left is latest master changes and right is the old codes.
by updating the codes (right) to be as master changes (left). i am getting the following error during set brightness after the initialization:

*Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 
Core  0 was running in ISR context:
EPC1    : 0x401d85c7  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x40082bfd

re-adding back those condition or changing I2S_INT_CLR_REG(0) instead of I2S_INT_CLR_REG(1) the make the library works again. i not sure if I2S1 need to be enable in config. wondering why forcing I2S1 to be default for 2 I2S device? can you shed some light for me to further check, might be direct related to pin that i am using or any configuration to enable? what benefits to use 2nd I2S instead the first one? 

for the random pixel issue, changing brightness level from max to the lowest did not fix the issue. i have commented two continue condition on line 768 and 776 but still having the same issue.

Thanks!

mrfaptastic, 2022-01-18:
You can't mix and match file versions, you're creating issues for yourself.

tsctrl, 2022-01-18:
no, i am not matching file version. i just integrating the latest master while checking what change from the last update. all files are the latest master.

that was the issue i figure out during updates

tsctrl, 2022-01-22:
hi @mrfaptastic ,

some updates on my end. i have updated my arduino ide matrix lib to current master and run 1_SimpleTestShapes.ino

2 changes done to have fixed display:
1. assign to my custom mcu pin.
2. uncomment drawText(0); in setup and remove the color wheel from the loop.

result as below:
[image]

i have tested with other boards and other panels with different chip with the same results.

tsctrl, 2022-01-22:
anything else are default codes no configuration was added, brightness are at 90. not sure if this was related to the random pixel above. 

thanks!

tsctrl, 2022-01-28:
hi @mrfaptastic , update on my end.

issue happens only when PIXEL_COLOR_DEPTH_BITS was reduced.
8 bits, no random pixel. 5, start to appear. 3, lots of random pixels

hope this help.

Thanks

mrfaptastic, 2022-02-06:
> actually, i could reproduce the same issue when using the library from arduino lib manager version: 2.0.6
hope this help.

Just to confirm, this issue exists in all versions of the library then? Even the old versions?

tsctrl, 2022-02-07:
yes, old version i can tweak the latch blanking to 4 to remove the pixels. but in master it is not reflecting. it is minor. but still visible in dark

tsctrl, 2022-02-10:
the latest arduino lib 2.0.6 that can be install using arduino version manager.

with single p4 64x32 1/16. any chip i guess will do. basically the issue was visible in lower pixelcolor depth bits. 5-3. highly visible on 3. 5 to 8 the random pixel was start to disappeared and 8 has no random pixel.

tsctrl, 2022-03-13:
Hi @mrfaptastic , had sometime to investigate. 

i test the library by switching in arduino IDE lib manager.
default config, DEPTH_BITS 8 and simple text display and a frame react as below:

`
    dma_display->clearScreen();
    dma_display->setFont(Font0);
    dma_display->drawRect(0, 0, 64, 32, myWHITE);
    dma_display->setTextColor(myRED);
    dma_display->setTextWrap(false);
    dma_display->setCursor(9, 20);
    dma_display->print("TEST MATRIX");
`

issue still exist in lib version 2.0.7 but minor effecting some text and Rect only. **fixed** by mxconfig.clkphase = false;
issue worst in lib version 2.0.6, has random pixel most of the screen
issue still exist in lib version 2.0.5 but minor effecting the Rect only
issue did not exist in lib version 2.0.3 display is clean

overall. from the arduino lib manager the library version 2.0.7 is fine with DEPTH_BITS 8. with **example test above**
changing DEPTH_BITS 3 in lib version 2.0.7 seems working fine too.

**SimpleTestShapes sketch test**
using the 1_SimpleTestShapes sketch again with DEPTH_BITS 8 in lib version 2.0.7 and mxconfig.clkphase = false;
no random ghosting.

using the 1_SimpleTestShapes sketch again with **DEPTH_BITS 3** in lib **version 2.0.7** and mxconfig.clkphase = false;
**have random ghosting as mentioned in the ticket.**

using the 1_SimpleTestShapes sketch again with **DEPTH_BITS 8** in lib **version 2.0.3** and mxconfig.clkphase = false;
**have random ghosting as mentioned in the ticket**

using the 1_SimpleTestShapes sketch again with **DEPTH_BITS 3** in lib **version 2.0.3** and mxconfig.clkphase = false;
**have random ghosting as mentioned in the ticket more visible.**

conclusion:
The random pixel exist in all version when bit depth is set low, but less visible in the latest lib.

thanks!

mrfaptastic, 2022-03-13:
Excellent analysis @tsctrl !

I think I might know what the issue is, and it's likely to do with toggling of the Output Enable not quite correct with the lower colour depth. Will investigate further.

mrfaptastic, 2022-03-13:
Excellent analysis @tsctrl !

I think I might know what the issue is now, and it's likely to do with toggling/timing of the Output Enable needing to be different with a lower colour depth. I'll investigate.

nguyedinhduc, 2022-09-11:
> Excellent analysis @tsctrl !
> 
> I think I might know what the issue is now, and it's likely to do with toggling/timing of the Output Enable needing to be different with a lower colour depth. I'll investigate.

@mrfaptastic, Can you help me?, i have the same error as yours @tsctrl !

nguyedinhduc, 2022-09-12:
> Hi @mrfaptastic,
> 
> i am having random pixel ghosting on p4 panels. latch blanking 1-4 did not fix this issue.
> 
> [image]
> 
> i could not replicate this issue on arduino with simple test shape example with 2.0.6 library.
> 
> edit: it exist in 2.0.6 but i fix it using latch blanking 4. but could not fix it using latch blanking on recent codes update from master. chip is tc5020a. code last update is on 25 dec 2021.
> 
> thank you

Did you fix it?  , can you help me ?

closed 2022-11-12 by mrfaptastic