mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #249: Restart esp32

dmb-220, 2022-01-27:
[image] 64x64 4 panel. Poeer supply 5v 30A,  restart esp32, is not working properly. Connect wifi ntp and openweather, draws on matrix. Restart 3-5 times for day. Push restart esp32 module,  working ok.

DarrylStrong, 2022-01-27:
Which panels are those please?

What does it do to require a restart? Totally stop running?

solhuebner, 2022-01-27:
Can you capture the serial output?

davemaster, 2022-01-27:
May be a Power supply issue, consider 10-12A per panel

Best regards

dmb-220, 2022-01-27:
> Which panels are those please?
> 
> What does it do to require a restart? Totally stop running?

the controller restarts itself, and shows how the photo is pinned. you need to press restar to display OK

dmb-220, 2022-01-27:
https://github.com/dmb-220/RGB-matrix

solhuebner, 2022-01-28:
A couple of things to test:

> Use of xTaskCreatePinnedToCore for TaskORAI and pin to core 1

> use https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv427uxTaskGetStackHighWaterMark12TaskHandle_t to find the right stack size for that task or just ramp it up a bit

> use the latest version of Arduino ESP32

mrfaptastic, 2022-02-06:
Does it restart if you do not have the LED matrix panels connected and turned on? You can still run the sketch and ESP32 without the LED matrix panels of course.... just keep a watch of the serial output.

If it still crashes without the LED panels then you a memory issue, else you have a power issue.

Either way, this isn't an issue caused by this library.