mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #251: Support FastLED CRGB for drawPixel

ModischFabrications, 2022-01-29:
Manual conversion from CRGB to raw draw Call is needlessly complicated: 

```
CRGB color = CRGB(1,2,3);
screen->drawPixelRGB888(sn.pos[0].x, sn.pos[0].y, color.r, color.g, color.b);
```

could be 
```
screen->drawPixel(sn.pos[0].x, sn.pos[0].y, color);
```

ModischFabrications, 2022-01-30:
Some more background: I often need advanced functions for gradients, blending and more, which is much easier with FastLED. On the other hand, text support from Adafruit GFX is nice. Seems like both could be supported, at least with a hidden conversion. 

Looking through it it seems like these methods are excluded with compiler flag `USE_GFX_ROOT`, but it seems like that would enable and disable a whole lot of other stuff...

mrfaptastic, 2022-02-06:
CRGB is supported with enabling 'USE_GFX_ROOT'

mrfaptastic, 2022-02-06:
I suggest you modify the library and enable the code that's not compiled when 'USE_GFX_ROOT' is disabled.

Then include FastLED instead of 'GFX Root' etc.

I don't want this library to have a dependancy on FastLED.

> Looking through it it seems like these methods are excluded with compiler flag USE_GFX_ROOT, but it seems like that would enable and disable a whole lot of other stuff...

Not quite sure what that is? GFX_ROOT is my hack of AdafruitGFX but modified for CRGB.