mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #253: ESP-IDF crash only if doublebuff is enabled

solhuebner, 2022-01-31:
I do get the following error message as soon as I enable double buffering.
I am using ESP-IDF 4.4 and the most recent arduino-esp32 (Arduino as component)

I could trace it back to 62deac6c488b02a2f97ba5cacf06f2683f811e78

If I change `#define ESP32_I2S_DEVICE            I2S_NUM_0` to `#define ESP32_I2S_DEVICE            I2S_NUM_1` (in esp32_i2s_parallel_mcu_def.h) I do not have any issues.

I am just not sure why it was no issue before that commit as it seems I2S_NUM_0 was used before. But with all later commits, including the mentioned one, I get this watchdog issue BUT only when I enable doublebuff (did not use it before so did not notice before).

E (11167) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (11167) task_wdt:  - IDLE (CPU 1)
E (11167) task_wdt: Tasks currently running:
E (11167) task_wdt: CPU 0: IDLE
E (11167) task_wdt: CPU 1: loopTask
E (11167) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400DFFCF:0x3FFB0BD00x40084109:0x3FFB0C00 0x400E6AF7:0x3FFB6B20 0x400D1F52:0x3FFB6B40 0x4008B250:0x3FFB6B60
0x400dffcf: task_wdt_isr at C:/Users/solhu/esp/esp-idf/components/esp_system/task_wdt.c:183 (discriminator 3)

0x40084109: _xt_lowint1 at C:/Users/solhu/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1111

0x400e6af7: cpu_ll_waiti at C:/Users/solhu/esp/esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at C:/Users/solhu/esp/esp-idf/components/esp_pm/pm_impl.c:837

0x400d1f52: esp_vApplicationIdleHook at C:/Users/solhu/esp/esp-idf/components/esp_system/freertos_hooks.c:63

0x4008b250: prvIdleTask at C:/Users/solhu/esp/esp-idf/components/freertos/tasks.c:3974

E (11167) task_wdt: Print CPU 1 backtrace

solhuebner, 2022-01-31:
I found the issue:

```
/* Compiler pre-processor check. Saves a few cycles, no need to cast void ptr to i2s_port_t and then check 120 times second... */
#if ESP32_I2S_DEVICE == I2S_NUM_0
      SET_PERI_REG_BITS(I2S_INT_CLR_REG(1), I2S_OUT_EOF_INT_CLR_V, 1, I2S_OUT_EOF_INT_CLR_S);
#else
      // For I2S0
      SET_PERI_REG_BITS(I2S_INT_CLR_REG(0), I2S_OUT_EOF_INT_CLR_V, 1, I2S_OUT_EOF_INT_CLR_S);    
#endif
```

If I change I2S_INT_CLR_REG(1) to 0 for I2S_NUM_0 and 1 for I2S_NUM_1 it works but the check `#if ESP32_I2S_DEVICE == I2S_NUM_0` is also not working for me for whatever reason.

vortigont, 2022-01-31:
@solhuebner, a bit of offtopic :)
mind to share your env setup for esp-idf + arduino as a component? I've been struggling days and nights trying to make it work IDF master and some recent arduino (2.0.2 is OK) as a component. Are you using idf.py or platformio?
We can move to discussions in [this repo](https://github.com/vortigont/ESP32-LightManager) to keep this issue clean :)

tsctrl, 2022-02-01:
same issue i mentioned here [issue](https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/issues/246#issuecomment-1015087445) 

i guess some settings change the i2s to 0 instead of 1 in idf. i havent check which config to set where the matrix lib codes expected esp32 to run i2s 1 by default. better to have the lib to refer to idf config i2s settings for current active i2s

solhuebner, 2022-02-01:
I think I have a rough idea what is happening:

`#if ESP32_I2S_DEVICE == I2S_NUM_0` is not working as it is an enum that is not available to the preprocessor.

We could change it to:

```
/* Saves a few cycles, no need to cast void ptr to i2s_port_t and then check 120 times second... */
if (ESP32_I2S_DEVICE == I2S_NUM_0)
      SET_PERI_REG_BITS(I2S_INT_CLR_REG(0), I2S_OUT_EOF_INT_CLR_V, 1, I2S_OUT_EOF_INT_CLR_S);
} else {
      // I2S_NUM_1
      SET_PERI_REG_BITS(I2S_INT_CLR_REG(1), I2S_OUT_EOF_INT_CLR_V, 1, I2S_OUT_EOF_INT_CLR_S);    
}
```

That includes both fixes: `I2S_NUM_0 -> I2S_INT_CLR_REG(0)` / `I2S_NUM_1 -> I2S_INT_CLR_REG(1)` plus the check is working BUT we get an 'additional' if statement during runtime.

solhuebner, 2022-02-01:
@tsctrl could you try replacing it in `esp32_i2s_parallel_dma.c`?

tsctrl, 2022-02-01:
but that will change I2S_NUM_0 -> I2S_INT_CLR_REG(0) where it suppose to I2S_NUM_0 -> I2S_INT_CLR_REG(1)? look like a bug but i not sure if that was the intended mapping.

solhuebner, 2022-02-01:
I tested it on my ESP32-WROVER-E and it crashes mine if it is not `I2S_NUM_0 -> I2S_INT_CLR_REG(0)` / `I2S_NUM_1 -> I2S_INT_CLR_REG(1)`

mrfaptastic, 2022-02-06:
Thanks for the fix @solhuebner

closed 2022-02-06 by mrfaptastic